EmixamPP/linux-enable-ir-emitter #110: Lenovo T14s (OpenRC)

NicolaSmaniotto, 2023-06-15:
**Output of `linux-enable-ir-emitter boot status`**
```
# linux-enable-ir-emitter boot status
Traceback (most recent call last):
  File "/usr/bin/linux-enable-ir-emitter", line 93, in <module>
    boot(args.boot_status)
  File "/usr/lib64/linux-enable-ir-emitter/command/boot.py", line 37, in boot
    if Systemd.status():
       ^^^^^^^^^^^^^^^^
  File "/usr/lib64/linux-enable-ir-emitter/systemd/Systemd.py", line 73, in status
    exec = subprocess.run(["systemctl", "status", SYSTEMD_NAME], capture_output=True)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/subprocess.py", line 548, in run
    with Popen(*popenargs, **kwargs) as process:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/subprocess.py", line 1024, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.11/subprocess.py", line 1917, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'systemctl'
```
I use Gentoo (OpenRC), so of course this fails.

**Output of `cat /etc/linux-enable-ir-emitter/*`**
```
$ cat /etc/linux-enable-ir-emitter/*
device=/dev/v4l/by-path/pci-0000:34:00.0-usb-0:1:1.2-video-index0
unit=7
selector=6
size=9
control0=1
control1=3
control2=2
control3=0
control4=0
control5=0
control6=0
control7=0
control8=0
4 3
```

**Give more information if you have**
The output of the configuration phase:
```
# linux-enable-ir-emitter configure
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
Is the ir emitter flashing (not just turn on) ? Yes/No ? No
Is the ir emitter flashing (not just turn on) ? Yes/No ? No
Is the ir emitter flashing (not just turn on) ? Yes/No ? No
Is the ir emitter flashing (not just turn on) ? Yes/No ? No
Is the ir emitter flashing (not just turn on) ? Yes/No ? No
Is the ir emitter flashing (not just turn on) ? Yes/No ? No
Is the ir emitter flashing (not just turn on) ? Yes/No ? No
Is the ir emitter flashing (not just turn on) ? Yes/No ? No
Is the ir emitter flashing (not just turn on) ? Yes/No ? No
Is the ir emitter flashing (not just turn on) ? Yes/No ? No
Is the ir emitter flashing (not just turn on) ? Yes/No ? No
Is the ir emitter flashing (not just turn on) ? Yes/No ? No
Is the ir emitter flashing (not just turn on) ? Yes/No ? No
Is the ir emitter flashing (not just turn on) ? Yes/No ? No
Is the ir emitter flashing (not just turn on) ? Yes/No ? No
Is the ir emitter flashing (not just turn on) ? Yes/No ? No
Is the ir emitter flashing (not just turn on) ? Yes/No ? No
Is the ir emitter flashing (not just turn on) ? Yes/No ? No
Is the ir emitter flashing (not just turn on) ? Yes/No ? No
Is the ir emitter flashing (not just turn on) ? Yes/No ? No
Is the ir emitter flashing (not just turn on) ? Yes/No ? No
Is the ir emitter flashing (not just turn on) ? Yes/No ? No
[... 175 lines cut ...]
                       contrast 0x00980901 (int)    : min=0 max=100 step=1 default=32 value=32
                     saturation 0x00980902 (int)    : min=0 max=100 step=1 default=64 value=64
                            hue 0x00980903 (int)    : min=-180 max=180 step=1 default=0 value=0
        white_balance_automatic 0x0098090c (bool)   : default=1 value=1
                          gamma 0x00980910 (int)    : min=90 max=150 step=1 default=120 value=120
           power_line_frequency 0x00980918 (menu)   : min=0 max=3 default=1 value=1 (50 Hz)
                                0: Disabled
                                1: 50 Hz
                                2: 60 Hz
                                3: Auto
      white_balance_temperature 0x0098091a (int)    : min=2800 max=6500 step=1 default=4600 value=4600 flags=inactive
                      sharpness 0x0098091b (int)    : min=0 max=7 step=1 default=3 value=3
         backlight_compensation 0x0098091c (int)    : min=0 max=2 step=1 default=1 value=1

Camera Controls

                  auto_exposure 0x009a0901 (menu)   : min=0 max=3 default=3 value=3 (Aperture Priority Mode)
                                1: Manual Mode
                                3: Aperture Priority Mode
         exposure_time_absolute 0x009a0902 (int)    : min=2 max=1250 step=1 default=156 value=156 flags=inactive
     exposure_dynamic_framerate 0x009a0903 (bool)   : default=0 value=1
                        privacy 0x009a0910 (bool)   : default=0 value=0
```

</details>

EmixamPP, 2023-06-15:
> I'm opening the issue because after running # linux-enable-ir-emitter run -v, the process immediately terminates with no output, 

`$ linux-enable-ir-emitter run` shows nothing if there is no error (also, the [exit code is 0 if success](https://github.com/EmixamPP/linux-enable-ir-emitter/wiki/Exit-code)).

> but the test command suggested [here on the arch wiki](https://wiki.archlinux.org/title/Howdy#Testing_your_IR_camera) outputs: ... Then returns with code 1

After executing `linux-enable-ir-emitter run`, you can check if the emitter is working properly (i.e. blinks), by executing `sudo howdy test` (if installed, think to configure the ir camera path first in `sudo howdy configure`) or using `ffplay /dev/video2` (usually provided by the `ffmpeg` package)

NicolaSmaniotto, 2023-06-15:
> `$ linux-enable-ir-emitter run`

Forgot to mention, the exit code is `0`

> ` $ ffplay /dev/video2`

This works, yay! It opens a window and the ir flashes. There must be some other problem with the gstreamer command.

I don't have howdy installed yet, since it is not packaged for gentoo. Working to package it myself, so I can test the whole setup.

Closing this, since it was actually an issue with how I tested the emitter, rather than it not being enabled.

closed 2023-06-15 by NicolaSmaniotto