EmixamPP/linux-enable-ir-emitter #129: Lenovo 510 FHD

intrapt, 2023-08-21:
**Ouput of `linux-enable-ir-emitter -v configure -l -1`**
<details><summary>configure command output</summary>

```
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
INFO: Configuring the camera: /dev/v4l/by-path/pci-0000:09:00.3-usb-0:4:1.2-video-index0.
Is the video flashing? Press Y or N
N pressed
DEBUG: unit: 2 selector: 6
DEBUG: current: 1 3 1 0 0 0 0 0 0
DEBUG: minimum: 0 0 0 0 0 0 0 0 0
DEBUG: maximum: 1 3 3 0 0 0 0 0 0
DEBUG: new current: 0 0 0 0 0 0 0 0 0
DEBUG: new current: 1 0 0 0 0 0 0 0 0
DEBUG: new current: 0 1 0 0 0 0 0 0 0
DEBUG: new current: 1 1 0 0 0 0 0 0 0
DEBUG: new current: 0 2 0 0 0 0 0 0 0
DEBUG: new current: 1 2 0 0 0 0 0 0 0
DEBUG: new current: 0 3 0 0 0 0 0 0 0
DEBUG: new current: 1 3 0 0 0 0 0 0 0
DEBUG: new current: 0 0 1 0 0 0 0 0 0
DEBUG: new current: 1 0 1 0 0 0 0 0 0
DEBUG: new current: 0 1 1 0 0 0 0 0 0
DEBUG: new current: 1 1 1 0 0 0 0 0 0
DEBUG: new current: 0 2 1 0 0 0 0 0 0
DEBUG: new current: 1 2 1 0 0 0 0 0 0
DEBUG: new current: 0 3 1 0 0 0 0 0 0
DEBUG: new current: 1 3 1 0 0 0 0 0 0
DEBUG: new current: 0 0 2 0 0 0 0 0 0
DEBUG: new current: 1 0 2 0 0 0 0 0 0
DEBUG: new current: 0 1 2 0 0 0 0 0 0
DEBUG: new current: 1 1 2 0 0 0 0 0 0
DEBUG: new current: 0 2 2 0 0 0 0 0 0
DEBUG: new current: 1 2 2 0 0 0 0 0 0
DEBUG: new current: 0 3 2 0 0 0 0 0 0
DEBUG: new current: 1 3 2 0 0 0 0 0 0
DEBUG: new current: 0 0 3 0 0 0 0 0 0
DEBUG: new current: 1 0 3 0 0 0 0 0 0
DEBUG: new current: 0 1 3 0 0 0 0 0 0
DEBUG: new current: 1 1 3 0 0 0 0 0 0
DEBUG: new current: 0 2 3 0 0 0 0 0 0
DEBUG: new current: 1 2 3 0 0 0 0 0 0
DEBUG: new current: 0 3 3 0 0 0 0 0 0
DEBUG: new current: 1 3 3 0 0 0 0 0 0
DEBUG: new current: 1 3 3 0 0 0 0 0 0
DEBUG:
ERROR: The configuration has failed.
INFO: Do not hesitate to visit the GitHub !
INFO: https://github.com/EmixamPP/linux-enable-ir-emitter/blob/master/docs/README.md
```
</details>

**Output of `v4l2-ctl --list-devices`**
```
Lenovo 510 Camera: Lenovo 510 R (usb-0000:09:00.3-4):
/dev/video0
/dev/video1
/dev/video2
/dev/video3
/dev/media0
/dev/media1
```

**Ouput of `v4l2-ctl -d /dev/videoX --list-formats-ext`**

<details><summary>v4l2 device output</summary>

```
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'GREY' (8-bit Greyscale)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
	[1]: 'YUYV' (YUYV 4:2:2)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.042s (24.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.133s (7.500 fps)
			Interval: Discrete 0.200s (5.000 fps)
```
</details>

**Output of `ls -l /dev/v4l/by-path`**
```
lrwxrwxrwx@ 12 root 21 Aug 10:30 pci-0000:09:00.3-usb-0:4:1.0-video-index0 -> ../../video0
lrwxrwxrwx@ 12 root 21 Aug 10:30 pci-0000:09:00.3-usb-0:4:1.0-video-index1 -> ../../video1
lrwxrwxrwx@ 12 root 21 Aug 10:30 pci-0000:09:00.3-usb-0:4:1.2-video-index0 -> ../../video2
lrwxrwxrwx@ 12 root 21 Aug 10:30 pci-0000:09:00.3-usb-0:4:1.2-video-index1 -> ../../video3
lrwxrwxrwx@ 12 root 21 Aug 10:30 pci-0000:09:00.3-usbv2-0:4:1.0-video-index0 -> ../../video0
lrwxrwxrwx@ 12 root 21 Aug 10:30 pci-0000:09:00.3-usbv2-0:4:1.0-video-index1 -> ../../video1
lrwxrwxrwx@ 12 root 21 Aug 10:30 pci-0000:09:00.3-usbv2-0:4:1.2-video-index0 -> ../../video2
lrwxrwxrwx@ 12 root 21 Aug 10:30 pci-0000:09:00.3-usbv2-0:4:1.2-video-index1 -> ../../video3
```

**Additional info**
 - Computer (or camera) model: Lenovo 510 FHD
 - Linux distribution: Arch Linux
 - Version of linux-enable-ir-emitter: 5.0.0

EmixamPP, 2023-08-21:
Did you try the manual mode (i.e. `-m`) ?

intrapt, 2023-08-21:
Yep, here's the output
```
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
INFO: Configuring the camera: /dev/v4l/by-path/pci-0000:09:00.3-usb-0:4:1.2-video-index0.
Is the video flashing? Press Y or N
N pressed
DEBUG: unit: 2 selector: 6
DEBUG: current: 1 3 1 0 0 0 0 0 0
DEBUG: minimum: 0 0 0 0 0 0 0 0 0
DEBUG: maximum: 1 3 3 0 0 0 0 0 0
DEBUG: new current: 0 0 0 0 0 0 0 0 0
Is the video flashing? Press Y or N
N pressed
DEBUG: new current: 1 0 0 0 0 0 0 0 0
Is the video flashing? Press Y or N
N pressed
DEBUG: new current: 0 1 0 0 0 0 0 0 0
Is the video flashing? Press Y or N
N pressed
DEBUG: new current: 1 1 0 0 0 0 0 0 0
Is the video flashing? Press Y or N
N pressed
DEBUG: new current: 1 3 3 0 0 0 0 0 0
DEBUG: new current: 1 3 3 0 0 0 0 0 0
DEBUG:
ERROR: The configuration has failed.
INFO: Do not hesitate to visit the GitHub !
INFO: https://github.com/EmixamPP/linux-enable-ir-emitter/blob/master/docs/README.md
```

EmixamPP, 2023-08-21:
Try this one: `linux-enable-ir-emitter -v configure -m -l -1`

NB: I see there is also a bug with this new version, I'll fix that, but I don't think it'll resolve your issue.

intrapt, 2023-08-21:
```
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
INFO: Configuring the camera: /dev/v4l/by-path/pci-0000:09:00.3-usb-0:4:1.2-video-index0.
Is the video flashing? Press Y or N
N pressed
DEBUG: unit: 2 selector: 6
DEBUG: current: 1 3 1 0 0 0 0 0 0
DEBUG: minimum: 0 0 0 0 0 0 0 0 0
DEBUG: maximum: 1 3 3 0 0 0 0 0 0
DEBUG: new current: 0 0 0 0 0 0 0 0 0
Is the video flashing? Press Y or N
N pressed
DEBUG: new current: 1 0 0 0 0 0 0 0 0
Is the video flashing? Press Y or N
N pressed
DEBUG: new current: 0 1 0 0 0 0 0 0 0
Is the video flashing? Press Y or N
N pressed
DEBUG: new current: 1 1 0 0 0 0 0 0 0
Is the video flashing? Press Y or N
N pressed
DEBUG: new current: 0 2 0 0 0 0 0 0 0
Is the video flashing? Press Y or N
N pressed
DEBUG: new current: 1 2 0 0 0 0 0 0 0
Is the video flashing? Press Y or N
N pressed
DEBUG: new current: 0 3 0 0 0 0 0 0 0
Is the video flashing? Press Y or N
N pressed
DEBUG: new current: 1 3 0 0 0 0 0 0 0
Is the video flashing? Press Y or N
N pressed
DEBUG: new current: 0 0 1 0 0 0 0 0 0
Is the video flashing? Press Y or N
N pressed
DEBUG: new current: 1 0 1 0 0 0 0 0 0
Is the video flashing? Press Y or N
N pressed
DEBUG: new current: 0 1 1 0 0 0 0 0 0
Is the video flashing? Press Y or N
N pressed
DEBUG: new current: 1 1 1 0 0 0 0 0 0
Is the video flashing? Press Y or N
N pressed
DEBUG: new current: 0 2 1 0 0 0 0 0 0
Is the video flashing? Press Y or N
N pressed
DEBUG: new current: 1 2 1 0 0 0 0 0 0
Is the video flashing? Press Y or N
N pressed
DEBUG: new current: 0 3 1 0 0 0 0 0 0
Is the video flashing? Press Y or N
N pressed
DEBUG: new current: 1 3 1 0 0 0 0 0 0
Is the video flashing? Press Y or N
N pressed
DEBUG: new current: 0 0 2 0 0 0 0 0 0
Is the video flashing? Press Y or N
N pressed
DEBUG: new current: 1 0 2 0 0 0 0 0 0
Is the video flashing? Press Y or N
N pressed
DEBUG: new current: 0 1 2 0 0 0 0 0 0
Is the video flashing? Press Y or N
N pressed
DEBUG: new current: 1 1 2 0 0 0 0 0 0
Is the video flashing? Press Y or N
N pressed
DEBUG: new current: 0 2 2 0 0 0 0 0 0
Is the video flashing? Press Y or N
N pressed
DEBUG: new current: 1 2 2 0 0 0 0 0 0
Is the video flashing? Press Y or N
N pressed
DEBUG: new current: 0 3 2 0 0 0 0 0 0
Is the video flashing? Press Y or N
N pressed
DEBUG: new current: 1 3 2 0 0 0 0 0 0
Is the video flashing? Press Y or N
N pressed
DEBUG: new current: 0 0 3 0 0 0 0 0 0
Is the video flashing? Press Y or N
N pressed
DEBUG: new current: 1 0 3 0 0 0 0 0 0
Is the video flashing? Press Y or N
N pressed
DEBUG: new current: 0 1 3 0 0 0 0 0 0
Is the video flashing? Press Y or N
N pressed
DEBUG: new current: 1 1 3 0 0 0 0 0 0
Is the video flashing? Press Y or N
N pressed
DEBUG: new current: 0 2 3 0 0 0 0 0 0
Is the video flashing? Press Y or N
N pressed
DEBUG: new current: 1 2 3 0 0 0 0 0 0
Is the video flashing? Press Y or N
N pressed
DEBUG: new current: 0 3 3 0 0 0 0 0 0
Is the video flashing? Press Y or N
N pressed
DEBUG: new current: 1 3 3 0 0 0 0 0 0
Is the video flashing? Press Y or N
N pressed
DEBUG: new current: 1 3 3 0 0 0 0 0 0
DEBUG:
ERROR: The configuration has failed.
INFO: Do not hesitate to visit the GitHub !
INFO: https://github.com/EmixamPP/linux-enable-ir-emitter/blob/master/docs/README.md
```

EmixamPP, 2023-10-09:
Hey, I'm coming back to you, with I hope, the solution.

In `/etc/linux-enable-ir-emitter/pci-0000:09:00.3-usb-0:4:1.2-video-index0_emitter0.driver` please write the following content:
```
device=/dev/v4l/by-path/pci-0000:09:00.3-usb-0:4:1.2-video-index0
unit=2
selector=6
control0=1
control1=3
control2=2
control3=0
control4=0
control5=0
control6=0
control7=0
control8=0
```
Then execute `sudo linux-enable-ir-emitter boot enable`, reboot, and check if the video is blinking (e.g. with `ffplay /dev/video2`)

intrapt, 2023-10-11:
Unfortunately the emitter still doesn't work

EmixamPP, 2023-10-11:
Maybe try this one: 
```
device=/dev/v4l/by-path/pci-0000:09:00.3-usb-0:4:1.2-video-index0
unit=2
selector=6
control0=1
control1=3
control2=3
control3=0
control4=0
control5=0
control6=0
control7=0
control8=0
```

EmixamPP, 2023-10-11:
And could you share with me the output of `sudo linux-enable-ir-emitter boot status` after rebooting?

intrapt, 2023-10-11:
Here's the output:
```
○ linux-enable-ir-emitter.service - enable the infrared emitter
     Loaded: loaded (/usr/lib/systemd/system/linux-enable-ir-emitter.service; enabled; preset: disabled)
     Active: inactive (dead) since Wed 2023-10-11 18:27:09 BST; 1min 0s ago
    Process: 732 ExecStartPre=/sbin/modprobe uvcvideo (code=exited, status=0/SUCCESS)
    Process: 773 ExecStart=/usr/bin/linux-enable-ir-emitter run (code=exited, status=0/SUCCESS)
   Main PID: 773 (code=exited, status=0/SUCCESS)
        CPU: 139ms

Oct 11 18:27:09 tyxnrzakt systemd[1]: Starting enable the infrared emitter...
Oct 11 18:27:09 tyxnrzakt systemd[1]: linux-enable-ir-emitter.service: Deactivated successfully.
Oct 11 18:27:09 tyxnrzakt systemd[1]: Finished enable the infrared emitter.
```
It's still not working unfortunately

intrapt, 2023-10-21:
It turns out that the emitter does work and always has, but only when the pixel format is set to GREY instead of YUYV. Unfortunately nothing I've done so far will make any program except [camset](https://github.com/azeam/camset/) use GREY, but at least it's not bug with this program.

Sorry for wasting your time, I really should've figured this out sooner

closed 2023-10-21 by intrapt