EmixamPP/linux-enable-ir-emitter #133: Lenovo Yoga 7i IR camera seemingly disappears from the system when it switches to the next set selector

NovaViper, 2023-09-17:
**Describe the bug**
I'm trying to configure my IR camera (testing a package for NixOS) and I managed to get it configured once over in ArchLinux but then accidentally formatted my computer's ssd so I lost the data for it. Ever since doing that now I can't get the program to finish configuring the camera without the camera just suddenly not responding to commands from the app; eventually failing with "Impossible to find your infared camera". This issue occurs on both NixOS and ArchLinux now and the issue happens as soon as it finishes one pattern and goes to move to the next selector and the IR light turns off and stays off until I restart the laptop. These were the last settings I had that had it working: unit: , selector 6, current: 1 3 2 0 0 0 0 0 0

I did check the camera over in Windows just to make sure that nothing was damaged and it worked just fine over in Windows so I'm completely unsure why now I can't get it to complete anymore

**How to reproduce**
Ran it several times til eventually it just randomly stops

**Additional info**
 - Computer (or camera) model: Yoga 7 16IRL8 - Type 82YN
 - Linux distribution: Nixos 23.05 and EndeavourOS
 - Version of linux-enable-ir-emitter: 5.0.2

EmixamPP, 2023-09-19:
I'm not sure to fully understand your case. 

Please use the correct issue template in order to provide all the necessary info: 
https://github.com/EmixamPP/linux-enable-ir-emitter/issues/new?assignees=&labels=&projects=&template=configuration-has-failed.md&title=

NovaViper, 2023-09-20:
Hey sorry for the delayed response! I did reboot each time before and after I ran the commands.

However, I actually can't replicate the issue anymore; looks like adding the values I saved on my phone made it work properly again. Even when letting it run again after deleting the driver off the computer completely, the camera doesn't do that weird behavior of it just not taking anymore commands like it was doing before. I've edited my the issue so if anyone else comes across it maybe they could find something out of it. But I'm going to close it since adding the values I saved fixed the issue.

closed 2023-09-20 by NovaViper