EmixamPP/linux-enable-ir-emitter #134: Fedora 38: Configure won't accept the answers

elegos, 2023-09-24:
**Describe the bug**
Trying to configure via ` sudo linux-enable-ir-emitter -v configure` the output asks me if I see the video flashing, but the script will not accept the response (I press "n" and "enter"). In the meanwhile a camera view is running and, if closed, will be reopened automatically.

[image]

**How to reproduce**
Just start the configure command on Fedora 38

**Additional info**
 - Computer (or camera) model:  NexiGo HelloCam N930W Camera
 - Linux distribution: Fedora 38
 - Version of linux-enable-ir-emitter: linux-enable-ir-emitter 5.0.2

EmixamPP, 2023-09-24:
You have to press "n" in the GUI (i.e. the camera feedback)

elegos, 2023-09-24:
Dumb, thanks! A little bit misleading, though.

Next error :( - ERROR: The configuration has failed.

**Ouput of `linux-enable-ir-emitter -v configure`**
<details><summary>configure command output</summary>

```
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
INFO: Configuring the camera: /dev/v4l/by-path/pci-0000:09:00.3-usb-0:3.2.1:1.2-video-index0.
ERROR: The configuration has failed.
INFO: Do not hesitate to visit the GitHub !
INFO: https://github.com/EmixamPP/linux-enable-ir-emitter/blob/master/docs/README.md
```
</details>

**Output of `v4l2-ctl --list-devices`**
```
NexiGo HelloCam N930W Camera: N (usb-0000:09:00.3-3.2.1):
        /dev/video0
        /dev/video1
        /dev/video2
        /dev/video3
        /dev/media0
        /dev/media1
```

**Ouput of `v4l2-ctl -d /dev/videoX --list-formats-ext`**

<details><summary>v4l2 device output</summary>

```
❯ v4l2-ctl -d /dev/video0 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'MJPG' (Motion-JPEG, compressed)
                Size: Discrete 1920x1080
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 1280x720
                        Interval: Discrete 0.017s (60.000 fps)
                Size: Discrete 960x540
                        Interval: Discrete 0.017s (60.000 fps)
                Size: Discrete 854x480
                        Interval: Discrete 0.017s (60.000 fps)
                Size: Discrete 640x480
                        Interval: Discrete 0.033s (30.000 fps)
        [1]: 'YUYV' (YUYV 4:2:2)
                Size: Discrete 1920x1080
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 1280x720
                        Interval: Discrete 0.100s (10.000 fps)
                Size: Discrete 960x540
                        Interval: Discrete 0.050s (20.000 fps)
                Size: Discrete 854x480
                        Interval: Discrete 0.050s (20.000 fps)
                Size: Discrete 640x480
                        Interval: Discrete 0.033s (30.000 fps)

❯ v4l2-ctl -d /dev/video1 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'GREY' (8-bit Greyscale)
                Size: Discrete 640x360
                        Interval: Discrete 0.033s (30.000 fps)

❯ v4l2-ctl -d /dev/video3 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

```
</details>

**Output of `ls -l /dev/v4l/by-path`**
```
lrwxrwxrwx. 1 root root 12 24 set 19.06 pci-0000:09:00.3-usb-0:3.2.1:1.0-video-index0 -> ../../video0
lrwxrwxrwx. 1 root root 12 24 set 19.06 pci-0000:09:00.3-usb-0:3.2.1:1.0-video-index1 -> ../../video1
lrwxrwxrwx. 1 root root 12 24 set 19.06 pci-0000:09:00.3-usb-0:3.2.1:1.2-video-index0 -> ../../video2
lrwxrwxrwx. 1 root root 12 24 set 19.06 pci-0000:09:00.3-usb-0:3.2.1:1.2-video-index1 -> ../../video3
```

**Give more information if you have**

**Additional info**
 - Computer (or camera) model: NexiGo HelloCam N930W Camera
 - Linux distribution: Fedora 38
 - Version of linux-enable-ir-emitter: 5.0.2

elegos, 2023-09-26:
With version 4.x it kind of work... when I run the configuration, the infrared starts blinking immediately, so I answer positively. After that the application exits and no driver is being installed (and thus enabled at startup). If I answer no, some settings change, fortunately it continues blinking and I can answer yes and it writes the driver. I don't have the log after my eyes, but I remember that version 4.x and 5.x act differently: the former tries the maximum setting (1 3 3 0 0 0...), and then lowers by 1 (i.e. 1 3 2 0 0 0), while 5.x starts from the bottom (1 3 1), which probably is not supported by the camera, failing without logging any error.

EmixamPP, 2023-09-26:
Thanks ! Indeed I see a regression in the outputs you given. I will manually create for you the configuration file, because at the moment, I do not have the time to fix the regression.

EmixamPP, 2023-09-26:
In `/etc/linux-enable-ir-emitter/pci-0000:09:00.3-usb-0:3.2.1:1.2-video-index0_emitter0.driver` please write the following content: 
```
device=pci-0000:09:00.3-usb-0:3.2.1:1.2-video-index0
unit=4
selector=6
control0=1
control1=3
control2=2
control3=0
control4=0
control5=0
control6=0
control7=0
control8=0
```
Then execute `sudo linux-enable-ir-emitter boot enable`

Please use the version 5.x

elegos, 2023-09-26:
```bash
❯ sudo linux-enable-ir-emitter boot enable       
ERROR: Error with the systemd boot service.
```

```bash
❯ journalctl -b -u linux-enable-ir-emitter.service
set 26 16:24:41 plungedcopper systemd[1]: Starting linux-enable-ir-emitter.service - enable the infrared emitter...
set 26 16:24:41 plungedcopper linux-enable-ir-emitter[6297]: terminate called after throwing an instance of 'std::runtime_error'
set 26 16:24:41 plungedcopper linux-enable-ir-emitter[6297]:   what():  CRITICAL: Unable to obtain the /dev/videoX path
set 26 16:24:41 plungedcopper systemd[1]: linux-enable-ir-emitter.service: Main process exited, code=exited, status=250/n/a
set 26 16:24:41 plungedcopper linux-enable-ir-emitter[6296]: ERROR: Impossible to execute the driver at /etc/linux-enable-ir-emitter/pci-0000:09:00.3-usb-0:3.2.1:1.2-video-index0_emitter0.driver.
set 26 16:24:41 plungedcopper systemd[1]: linux-enable-ir-emitter.service: Failed with result 'exit-code'.
set 26 16:24:41 plungedcopper systemd[1]: Failed to start linux-enable-ir-emitter.service - enable the infrared emitter.
```

EmixamPP, 2023-09-27:
Sorry, I made a mistake. Here is the correct file content: 
```
device=/dev/v4l/by-path/pci-0000:09:00.3-usb-0:3.2.1:1.2-video-index0
unit=4
selector=6
control0=1
control1=3
control2=2
control3=0
control4=0
control5=0
control6=0
control7=0
control8=0
```

elegos, 2023-09-28:
Yay! It works!

Thank you. What's the logic behind the regression? I'd try messing around the source.

EmixamPP, 2023-09-29:
Honestly I don't know. Actually, I don't think that there is a regression, but maybe something related to the automatic detection. The manual mode (i.e. `-m`) could resolve the issue you faced.

closed 2023-10-09 by EmixamPP