EmixamPP/linux-enable-ir-emitter #135: Cannot configure in Fedora of Lenovo Yoga Pro 7/7i

azvyae, 2023-09-27:
Last time i configure this package inside my Kubuntu distro (same device), it is working smoothly and it can prompt the y/n answer in certain period. But now, there is no prompt afterwards even though the emitter is still blinking, it just keep searching

**Ouput of `linux-enable-ir-emitter -v configure`**
<details><summary>configure command output</summary>

```
sudo linux-enable-ir-emitter -v -d /dev/video2 configure
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
INFO: Configuring the camera: /dev/v4l/by-path/pci-0000:00:14.0-usb-0:6:1.2-video-index0.
Is the video flashing? Press Y or N
N pressed
DEBUG: unit: 13 selector: 14
DEBUG: current: 0 2
DEBUG: minimum: 0 0
DEBUG: maximum: 2 100
DEBUG: new current: 0 0
DEBUG: new current: 1 0
DEBUG: new current: 2 0
DEBUG: new current: 0 1
DEBUG: new current: 1 1
DEBUG: new current: 2 1
DEBUG: new current: 0 2
DEBUG: new current: 1 2
DEBUG: new current: 2 2
DEBUG: new current: 0 3
DEBUG: new current: 1 3
DEBUG: new current: 2 3
DEBUG: new current: 0 4
DEBUG: new current: 1 4
DEBUG: new current: 2 4
DEBUG: new current: 0 5
DEBUG: new current: 1 5
DEBUG: new current: 2 5
DEBUG: new current: 0 6
DEBUG: new current: 1 6
DEBUG: new current: 2 6
DEBUG: new current: 0 7
DEBUG: new current: 1 7
DEBUG: new current: 2 7
DEBUG: new current: 0 8
DEBUG: new current: 1 8
DEBUG: new current: 2 8
DEBUG: new current: 0 9
DEBUG: new current: 1 9
DEBUG: new current: 2 9
DEBUG: new current: 0 10
DEBUG: new current: 1 10
DEBUG: new current: 2 10
DEBUG: new current: 0 11
DEBUG: new current: 1 11
DEBUG: new current: 2 11
DEBUG: new current: 0 12
DEBUG: new current: 1 12
DEBUG: new current: 2 12
DEBUG: new current: 0 13
DEBUG: new current: 1 13
DEBUG: new current: 2 100
DEBUG: 
DEBUG: unit: 13 selector: 16
DEBUG: current: 0
DEBUG: minimum: 0
DEBUG: maximum: 1
DEBUG: new current: 1
DEBUG: 
DEBUG: unit: 14 selector: 2
DEBUG: current: 1 0 0 0 0 0 0 16 39 0 0 0 0 0 0
DEBUG: minimum: 0 0 0 0 0 0 0 16 39 0 0 0 0 0 0
DEBUG: maximum: 7 0 0 0 0 0 0 200 20 5 0 0 0 0 0
DEBUG: new current: 0 0 0 0 0 0 0 16 39 0 0 0 0 0 0
DEBUG: new current: 1 0 0 0 0 0 0 16 39 0 0 0 0 0 0
DEBUG: new current: 2 0 0 0 0 0 0 16 39 0 0 0 0 0 0
DEBUG: new current: 3 0 0 0 0 0 0 16 39 0 0 0 0 0 0
DEBUG: new current: 4 0 0 0 0 0 0 16 39 0 0 0 0 0 0
DEBUG: new current: 5 0 0 0 0 0 0 16 39 0 0 0 0 0 0
DEBUG: new current: 6 0 0 0 0 0 0 16 39 0 0 0 0 0 0
DEBUG: new current: 7 0 0 0 0 0 0 16 39 0 0 0 0 0 0
DEBUG: new current: 0 0 0 0 0 0 0 17 39 0 0 0 0 0 0
DEBUG: new current: 1 0 0 0 0 0 0 17 39 0 0 0 0 0 0
DEBUG: new current: 2 0 0 0 0 0 0 17 39 0 0 0 0 0 0
DEBUG: new current: 3 0 0 0 0 0 0 17 39 0 0 0 0 0 0
DEBUG: new current: 4 0 0 0 0 0 0 17 39 0 0 0 0 0 0
DEBUG: new current: 5 0 0 0 0 0 0 17 39 0 0 0 0 0 0
DEBUG: new current: 6 0 0 0 0 0 0 17 39 0 0 0 0 0 0
DEBUG: new current: 7 0 0 0 0 0 0 17 39 0 0 0 0 0 0
DEBUG: new current: 0 0 0 0 0 0 0 18 39 0 0 0 0 0 0
DEBUG: new current: 1 0 0 0 0 0 0 18 39 0 0 0 0 0 0
DEBUG: new current: 2 0 0 0 0 0 0 18 39 0 0 0 0 0 0
DEBUG: new current: 3 0 0 0 0 0 0 18 39 0 0 0 0 0 0
DEBUG: new current: 4 0 0 0 0 0 0 18 39 0 0 0 0 0 0
DEBUG: new current: 5 0 0 0 0 0 0 18 39 0 0 0 0 0 0
DEBUG: new current: 6 0 0 0 0 0 0 18 39 0 0 0 0 0 0
DEBUG: new current: 7 0 0 0 0 0 0 18 39 0 0 0 0 0 0
DEBUG: new current: 0 0 0 0 0 0 0 19 39 0 0 0 0 0 0
DEBUG: new current: 1 0 0 0 0 0 0 19 39 0 0 0 0 0 0
DEBUG: new current: 2 0 0 0 0 0 0 19 39 0 0 0 0 0 0
DEBUG: new current: 3 0 0 0 0 0 0 19 39 0 0 0 0 0 0
DEBUG: new current: 4 0 0 0 0 0 0 19 39 0 0 0 0 0 0
DEBUG: new current: 5 0 0 0 0 0 0 19 39 0 0 0 0 0 0
DEBUG: new current: 6 0 0 0 0 0 0 19 39 0 0 0 0 0 0
DEBUG: new current: 7 0 0 0 0 0 0 19 39 0 0 0 0 0 0
DEBUG: new current: 0 0 0 0 0 0 0 20 39 0 0 0 0 0 0
DEBUG: new current: 1 0 0 0 0 0 0 20 39 0 0 0 0 0 0
DEBUG: new current: 2 0 0 0 0 0 0 20 39 0 0 0 0 0 0
DEBUG: new current: 3 0 0 0 0 0 0 20 39 0 0 0 0 0 0
DEBUG: new current: 4 0 0 0 0 0 0 20 39 0 0 0 0 0 0
DEBUG: new current: 5 0 0 0 0 0 0 20 39 0 0 0 0 0 0
DEBUG: new current: 6 0 0 0 0 0 0 20 39 0 0 0 0 0 0
DEBUG: new current: 7 0 0 0 0 0 0 20 39 0 0 0 0 0 0
DEBUG: new current: 0 0 0 0 0 0 0 21 39 0 0 0 0 0 0
DEBUG: new current: 7 0 0 0 0 0 0 200 20 5 0 0 0 0 0
DEBUG: 
DEBUG: unit: 14 selector: 3
DEBUG: current: 16 0 0 0 0 0 0 0 0 0 0
DEBUG: minimum: 16 0 0 0 0 0 0 0 0 0 0
DEBUG: maximum: 16 0 0 0 0 0 0 232 3 0 0
DEBUG: new current: 16 0 0 0 0 0 0 1 0 0 0
DEBUG: new current: 16 0 0 0 0 0 0 2 0 0 0
DEBUG: new current: 16 0 0 0 0 0 0 3 0 0 0
DEBUG: new current: 16 0 0 0 0 0 0 4 0 0 0
DEBUG: new current: 16 0 0 0 0 0 0 5 0 0 0
DEBUG: new current: 16 0 0 0 0 0 0 6 0 0 0
DEBUG: new current: 16 0 0 0 0 0 0 7 0 0 0
DEBUG: new current: 16 0 0 0 0 0 0 8 0 0 0
DEBUG: new current: 16 0 0 0 0 0 0 9 0 0 0
DEBUG: new current: 16 0 0 0 0 0 0 10 0 0 0
DEBUG: new current: 16 0 0 0 0 0 0 11 0 0 0
DEBUG: new current: 16 0 0 0 0 0 0 12 0 0 0
DEBUG: new current: 16 0 0 0 0 0 0 13 0 0 0
DEBUG: new current: 16 0 0 0 0 0 0 14 0 0 0
DEBUG: new current: 16 0 0 0 0 0 0 15 0 0 0
DEBUG: new current: 16 0 0 0 0 0 0 16 0 0 0
DEBUG: new current: 16 0 0 0 0 0 0 17 0 0 0
DEBUG: new current: 16 0 0 0 0 0 0 18 0 0 0
DEBUG: new current: 16 0 0 0 0 0 0 19 0 0 0
DEBUG: new current: 16 0 0 0 0 0 0 20 0 0 0
DEBUG: new current: 16 0 0 0 0 0 0 21 0 0 0
DEBUG: new current: 16 0 0 0 0 0 0 22 0 0 0
DEBUG: new current: 16 0 0 0 0 0 0 23 0 0 0
DEBUG: new current: 16 0 0 0 0 0 0 24 0 0 0
DEBUG: new current: 16 0 0 0 0 0 0 25 0 0 0
DEBUG: new current: 16 0 0 0 0 0 0 26 0 0 0
DEBUG: new current: 16 0 0 0 0 0 0 27 0 0 0
DEBUG: new current: 16 0 0 0 0 0 0 28 0 0 0
DEBUG: new current: 16 0 0 0 0 0 0 29 0 0 0
DEBUG: new current: 16 0 0 0 0 0 0 30 0 0 0
DEBUG: new current: 16 0 0 0 0 0 0 31 0 0 0
DEBUG: new current: 16 0 0 0 0 0 0 32 0 0 0
DEBUG: new current: 16 0 0 0 0 0 0 33 0 0 0
DEBUG: new current: 16 0 0 0 0 0 0 34 0 0 0
DEBUG: new current: 16 0 0 0 0 0 0 35 0 0 0
DEBUG: new current: 16 0 0 0 0 0 0 36 0 0 0
DEBUG: new current: 16 0 0 0 0 0 0 37 0 0 0
DEBUG: new current: 16 0 0 0 0 0 0 38 0 0 0
DEBUG: new current: 16 0 0 0 0 0 0 39 0 0 0
DEBUG: new current: 16 0 0 0 0 0 0 40 0 0 0
DEBUG: new current: 16 0 0 0 0 0 0 232 3 0 0
DEBUG: 
DEBUG: Using current as minimum control.
DEBUG: unit: 14 selector: 6
DEBUG: current: 1 3 1 0 0 0 0 0 0
DEBUG: minimum: 1 3 1 0 0 0 0 0 0
DEBUG: maximum: 1 3 3 0 0 0 0 0 0
DEBUG: new current: 1 3 2 0 0 0 0 0 0
DEBUG: new current: 1 3 3 0 0 0 0 0 0
DEBUG: 
DEBUG: unit: 14 selector: 9
DEBUG: current: 1 0 0 0
DEBUG: minimum: 0 0 0 0
DEBUG: maximum: 1 0 0 0
DEBUG: new current: 0 0 0 0
DEBUG: new current: 1 0 0 0
DEBUG: 
ERROR: The configuration has failed.
INFO: Do not hesitate to visit the GitHub !
INFO: https://github.com/EmixamPP/linux-enable-ir-emitter/blob/master/docs/README.md
```
</details>

**Output of `v4l2-ctl --list-devices`**
```
v4l2-ctl --list-devices
Integrated Camera: Integrated C (usb-0000:00:14.0-6):
	/dev/video0
	/dev/video1
	/dev/video2
	/dev/video3
	/dev/media0
	/dev/media1
```

**Ouput of `v4l2-ctl -d /dev/video2 --list-formats-ext`**

<details><summary>v4l2 device output</summary>

```
v4l2-ctl -d /dev/video2 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'GREY' (8-bit Greyscale)
		Size: Discrete 640x360
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
```
</details>

**Output of `ls -l /dev/v4l/by-path`**
```
ls -l /dev/v4l/by-path
total 0
lrwxrwxrwx. 1 root root 12 Sep 27 12:17 pci-0000:00:14.0-usb-0:6:1.0-video-index0 -> ../../video0
lrwxrwxrwx. 1 root root 12 Sep 27 12:17 pci-0000:00:14.0-usb-0:6:1.0-video-index1 -> ../../video1
lrwxrwxrwx. 1 root root 12 Sep 27 12:17 pci-0000:00:14.0-usb-0:6:1.2-video-index0 -> ../../video2
lrwxrwxrwx. 1 root root 12 Sep 27 12:17 pci-0000:00:14.0-usb-0:6:1.2-video-index1 -> ../../video3

```

**Give more information if you have**

**Additional info**
 - Computer (or camera) model: Lenovo Yoga Pro 7/7i (2023 model) - 14IRH8
 - Linux distribution: Fedora Linux 38 (Workstation Edition), Gnome 44.5 on Wayland, Linux 6.4.15-200.fc38.x86_64
 - Version of linux-enable-ir-emitter:

azvyae, 2023-09-27:
You have to do it manually by providing -m

closed 2023-09-27 by azvyae

EmixamPP, 2023-09-27:
I will add a message in the terminal that ask to retry with -m in case of failure.