EmixamPP/linux-enable-ir-emitter #138: Lenovo ThinkPad E14 Gen 3

Haorelian, 2023-10-07:
**Ouput of `linux-enable-ir-emitter -v configure`**
<details><summary>configure command output</summary>

```
haorelian@localhost:~> sudo linux-enable-ir-emitter -v configure
[sudo] password for root: 
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
INFO: Configuring the camera: /dev/v4l/by-path/pci-0000:05:00.3-usb-0:3:1.2-video-index0.
Is the video flashing? Press Y or N
terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.8.0) /home/runner/work/linux-enable-ir-emitter/linux-enable-ir-emitter/opencv/opencv-4.8.0/modules/highgui/src/window_gtk.cpp:638: error: (-2:Unspecified error) Can't initialize GTK backend in function 'cvInitSystem'

ERROR: The configuration has failed.
INFO: Do not hesitate to visit the GitHub !
INFO: https://github.com/EmixamPP/linux-enable-ir-emitter/blob/master/docs/README.md
haorelian@localhost:~> 

```
</details>

**Output of `v4l2-ctl --list-devices`**
```
haorelian@localhost:~> v4l2-ctl --list-devices
Integrated Camera: Integrated C (usb-0000:05:00.3-3):
        /dev/video0
        /dev/video1
        /dev/video2
        /dev/video3
        /dev/media0
        /dev/media1

```

**Ouput of `v4l2-ctl -d /dev/videoX --list-formats-ext`**

<details><summary>v4l2 device output</summary>

```
haorelian@localhost:~> v4l2-ctl -d /dev/video0 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'MJPG' (Motion-JPEG, compressed)
                Size: Discrete 1280x720
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 320x180
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 320x240
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 352x288
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 424x240
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 640x360
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 640x480
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 848x480
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 960x540
                        Interval: Discrete 0.033s (30.000 fps)
        [1]: 'YUYV' (YUYV 4:2:2)
                Size: Discrete 640x480
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 320x180
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 320x240
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 352x288
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 424x240
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 640x360
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 848x480
                        Interval: Discrete 0.050s (20.000 fps)
                Size: Discrete 960x540
                        Interval: Discrete 0.067s (15.000 fps)
                Size: Discrete 1280x720
                        Interval: Discrete 0.100s (10.000 fps)
haorelian@localhost:~> v4l2-ctl -d /dev/video1 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

haorelian@localhost:~> v4l2-ctl -d /dev/video2 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'GREY' (8-bit Greyscale)
                Size: Discrete 640x360
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.033s (30.000 fps)
haorelian@localhost:~> v4l2-ctl -d /dev/video3 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

```
</details>

**Output of `ls -l /dev/v4l/by-path`**
```
haorelian@localhost:~> ls -l /dev/v4l/by-path
total 0
lrwxrwxrwx 1 root root 12 Oct  7 10:50 pci-0000:05:00.3-usb-0:3:1.0-video-index0 -> ../../video0
lrwxrwxrwx 1 root root 12 Oct  7 10:50 pci-0000:05:00.3-usb-0:3:1.0-video-index1 -> ../../video1
lrwxrwxrwx 1 root root 12 Oct  7 10:50 pci-0000:05:00.3-usb-0:3:1.2-video-index0 -> ../../video2
lrwxrwxrwx 1 root root 12 Oct  7 10:50 pci-0000:05:00.3-usb-0:3:1.2-video-index1 -> ../../video3
lrwxrwxrwx 1 root root 12 Oct  7 10:50 pci-0000:05:00.3-usbv2-0:3:1.0-video-index0 -> ../../video0
lrwxrwxrwx 1 root root 12 Oct  7 10:50 pci-0000:05:00.3-usbv2-0:3:1.0-video-index1 -> ../../video1
lrwxrwxrwx 1 root root 12 Oct  7 10:50 pci-0000:05:00.3-usbv2-0:3:1.2-video-index0 -> ../../video2
lrwxrwxrwx 1 root root 12 Oct  7 10:50 pci-0000:05:00.3-usbv2-0:3:1.2-video-index1 -> ../../video3

```

**Give more information if you have**
I'm on openSUSE Tumbleweed, installed Howdy and the IR Camera didn't work. I've checked with "ffplay /dev/video2" and IR Camera is too dark and the red light in IR Camera didn't light up. I don't really know what information would be useful, so specify it while asking, thanks.

**Additional info**
 - Computer (or camera) model: Lenovo ThinkPad E14 Gen 3
 - Linux distribution: openSUSE Tumbleweed
 - Version of linux-enable-ir-emitter: linux-enable-ir-emitter 5.0.2

Haorelian, 2023-10-07:
Addition: I've tried to run it with "sudo DISPLAY=$DISPLAY linux-enable-ir-emitter configure" and switched back to X11 it throws this: Authorization required, but no authorization protocol specified

Authorization required, but no authorization protocol specified

Authorization required, but no authorization protocol specified

Authorization required, but no authorization protocol specified

^CTraceback (most recent call last):
  File "/usr/bin/linux-enable-ir-emitter", line 118, in <module>
    configure(device, args.manual, args.emitters[0], args.limit[0])
  File "/usr/lib64/linux-enable-ir-emitter/command/configure.py", line 49, in configure
    exit_code = subprocess.call(
                ^^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.11/subprocess.py", line 391, in call
    return p.wait(timeout=timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.11/subprocess.py", line 1264, in wait
    return self._wait(timeout=timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.11/subprocess.py", line 2046, in _wait
    (pid, sts) = self._try_wait(0)
                 ^^^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.11/subprocess.py", line 2004, in _try_wait
    (pid, sts) = os.waitpid(self.pid, wait_flags)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
KeyboardInterrupt

EmixamPP, 2023-10-07:
How did you installed linux-enable-ir-emitter? Using the binary provided by this repo?

Haorelian, 2023-10-08:
> How did you installed linux-enable-ir-emitter? Using the binary provided by this repo?

Like this: Download the latest linux-enable-ir-emitter-x.x.x.systemd.x86-64.tar.gz [here](https://github.com/EmixamPP/linux-enable-ir-emitter/releases/latest), then execute:

sudo tar -C / --no-same-owner -h -xzf linux-enable-ir-emitter*.tar.gz

EmixamPP, 2023-10-09:
Could you try manual building and installation? The instructions are in the README, but if you need help, please don't hesitate.

EmixamPP, 2023-10-09:
I've just released a new binary that contains a new version of OpenCV, I don't know if it contains a fix for your problem

Haorelian, 2023-10-23:
Sorry for the very late response. I've tried a bunch of things to make it work, but when it didn't, I just gave up and switched to the Fedora 38 Workstation. Now it works as intended. Once again, sorry for taking your time.

closed 2023-10-23 by EmixamPP

ksajan, 2025-03-15:
I am also recieving the same exact error.

EmixamPP, 2025-03-15:
@ksajan please use the binary I am providing here on github