EmixamPP/linux-enable-ir-emitter #143: Can't configure on T480s

Alonely0, 2023-10-13:
**Ouput of `linux-enable-ir-emitter -v configure`**
<details><summary>configure command output</summary>

```
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
INFO: Configuring the camera: /dev/v4l/by-path/pci-0000:00:14.0-usbv2-0:8:1.0-video-index0
/dev/v4l/by-path/pci-0000:00:14.0-usb-0:8:1.0-video-index0.
terminate called after throwing an instance of 'std::runtime_error'
  what():  CRITICAL: Unable to obtain the /dev/videoX path
ERROR: The configuration has failed.
ERROR: Please retry in manual mode by adding the '-m' option.
INFO: Do not hesitate to visit the GitHub !
INFO: https://github.com/EmixamPP/linux-enable-ir-emitter/blob/master/docs/README.md
```
</details>

**Output of `v4l2-ctl --list-devices`**
```
Integrated IR Camera: Integrate (usb-0000:00:14.0-5):
        /dev/video0
        /dev/video1
        /dev/media0

Integrated Camera: Integrated C (usb-0000:00:14.0-8):
        /dev/video2
        /dev/video3
        /dev/media1
```

**Ouput of `v4l2-ctl -d /dev/videoX --list-formats-ext`**

<details><summary>v4l2 device output</summary>

```
# 0
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'MJPG' (Motion-JPEG, compressed)
                Size: Discrete 640x480
                        Interval: Discrete 0.017s (60.000 fps)
                Size: Discrete 320x240
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 160x120
                        Interval: Discrete 0.033s (30.000 fps)
        [1]: 'YUYV' (YUYV 4:2:2)
                Size: Discrete 400x480
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 400x400
                        Interval: Discrete 0.033s (30.000 fps)
# 1
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture
# 2
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'MJPG' (Motion-JPEG, compressed)
                Size: Discrete 1280x720
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 320x180
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 320x240
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 352x288
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 424x240
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 640x360
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 640x480
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 848x480
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 960x540
                        Interval: Discrete 0.033s (30.000 fps)
        [1]: 'YUYV' (YUYV 4:2:2)
                Size: Discrete 1280x720
                        Interval: Discrete 0.100s (10.000 fps)
                Size: Discrete 320x180
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 320x240
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 352x288
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 424x240
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 640x360
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 640x480
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 848x480
                        Interval: Discrete 0.050s (20.000 fps)
                Size: Discrete 960x540
                        Interval: Discrete 0.067s (15.000 fps)
# 3
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture
```
</details>

**Output of `ls -l /dev/v4l/by-path`**
```
lrwxrwxrwx 1 root root 12 Oct 13 13:05 pci-0000:00:14.0-usb-0:5:1.0-video-index0 -> ../../video0
lrwxrwxrwx 1 root root 12 Oct 13 13:05 pci-0000:00:14.0-usb-0:5:1.0-video-index1 -> ../../video1
lrwxrwxrwx 1 root root 12 Oct 13 13:05 pci-0000:00:14.0-usb-0:8:1.0-video-index0 -> ../../video2
lrwxrwxrwx 1 root root 12 Oct 13 13:05 pci-0000:00:14.0-usb-0:8:1.0-video-index1 -> ../../video3
lrwxrwxrwx 1 root root 12 Oct 13 13:05 pci-0000:00:14.0-usbv2-0:5:1.0-video-index0 -> ../../video0
lrwxrwxrwx 1 root root 12 Oct 13 13:05 pci-0000:00:14.0-usbv2-0:5:1.0-video-index1 -> ../../video1
lrwxrwxrwx 1 root root 12 Oct 13 13:05 pci-0000:00:14.0-usbv2-0:8:1.0-video-index0 -> ../../video2
lrwxrwxrwx 1 root root 12 Oct 13 13:05 pci-0000:00:14.0-usbv2-0:8:1.0-video-index1 -> ../../video3
```

**Give more information if you have**

I also have tried the manual mode and it's the same. The T480s has two emitters, so I did try `-e 2`, but with no success.

**Additional info**
 - Computer (or camera) model: Lenovo Thinkpad T480s
 - Linux distribution: OpenSUSE Tumbleweed
 - Version of linux-enable-ir-emitter:  This one is weird, because I downloaded the 5.0.4-systemd, but the command outputs 5.0.2

EmixamPP, 2023-10-14:
Please add the argument `-d /dev/video0`

Alonely0, 2023-10-14:
I actually tried it with all video devices because it couldn't auto-detect it, but I'll attach the output:
```
guillem@localhost:~/Downloads> for i in /dev/video*; do printf "\n\n$i\n" && sudo linux-enable-ir-emitter -d $i -v configure; done

/dev/video0
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
INFO: Configuring the camera: /dev/v4l/by-path/pci-0000:00:14.0-usbv2-0:5:1.0-video-index0
/dev/v4l/by-path/pci-0000:00:14.0-usb-0:5:1.0-video-index0.
terminate called after throwing an instance of 'std::runtime_error'
  what():  CRITICAL: Unable to obtain the /dev/videoX path
ERROR: The configuration has failed.
ERROR: Please retry in manual mode by adding the '-m' option.
INFO: Do not hesitate to visit the GitHub !
INFO: https://github.com/EmixamPP/linux-enable-ir-emitter/blob/master/docs/README.md

/dev/video1
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
INFO: Configuring the camera: /dev/v4l/by-path/pci-0000:00:14.0-usbv2-0:5:1.0-video-index1
/dev/v4l/by-path/pci-0000:00:14.0-usb-0:5:1.0-video-index1.
terminate called after throwing an instance of 'std::runtime_error'
  what():  CRITICAL: Unable to obtain the /dev/videoX path
ERROR: The configuration has failed.
ERROR: Please retry in manual mode by adding the '-m' option.
INFO: Do not hesitate to visit the GitHub !
INFO: https://github.com/EmixamPP/linux-enable-ir-emitter/blob/master/docs/README.md

/dev/video2
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
INFO: Configuring the camera: /dev/v4l/by-path/pci-0000:00:14.0-usbv2-0:8:1.0-video-index0
/dev/v4l/by-path/pci-0000:00:14.0-usb-0:8:1.0-video-index0.
terminate called after throwing an instance of 'std::runtime_error'
  what():  CRITICAL: Unable to obtain the /dev/videoX path
ERROR: The configuration has failed.
ERROR: Please retry in manual mode by adding the '-m' option.
INFO: Do not hesitate to visit the GitHub !
INFO: https://github.com/EmixamPP/linux-enable-ir-emitter/blob/master/docs/README.md

/dev/video3
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
INFO: Configuring the camera: /dev/v4l/by-path/pci-0000:00:14.0-usb-0:8:1.0-video-index1
/dev/v4l/by-path/pci-0000:00:14.0-usbv2-0:8:1.0-video-index1.
terminate called after throwing an instance of 'std::runtime_error'
  what():  CRITICAL: Unable to obtain the /dev/videoX path
ERROR: The configuration has failed.
ERROR: Please retry in manual mode by adding the '-m' option.
INFO: Do not hesitate to visit the GitHub !
INFO: https://github.com/EmixamPP/linux-enable-ir-emitter/blob/master/docs/README.md
```

Alonely0, 2023-10-14:
For what it's worth, on Firefox when connecting to a Google Meet session, if I select the IR camera, the emitters trigger. That doesn't happen on any camera app, nor in howdy.

EmixamPP, 2023-10-14:
In howdy did you configured it to use /dev/video0 or pci-0000:00:14.0-usb-0:5:1.0-video-index0 ?

Alonely0, 2023-10-14:
> In howdy did you configured it to use /dev/video0 or pci-0000:00:14.0-usb-0:5:1.0-video-index0 ?

/dev/video0, and I also tried /dev/video2

EmixamPP, 2023-10-14:
Your infrared camera is on /dev/video0.

Could you try with `-d /dev/v4l/by-path/pci-0000:00:14.0-usb-0:5:1.0-video-index0` ?

Alonely0, 2023-10-14:
Wait. IR emitters are now triggering on howdy when using /dev/video0 (very, very strange), but linux-enable-ir-emitters still doesn't work,

EmixamPP, 2023-10-14:
Anyway, your infrared emitter is working, you don't need leie :smile:

EmixamPP, 2023-10-18:
I'm closing the issue, but feel free to comment if you need help

closed 2023-10-18 by EmixamPP