EmixamPP/linux-enable-ir-emitter #148: Segmentation Fault during `configure`

http403, 2023-10-26:
**Describe the bug**
Segmentation fault during the `configure` process. Unknown reason.

**How to reproduce**
Fresh install OS with all packages up to date and [Howdy](https://github.com/boltgolt/howdy) installed via official PPA. Given how new the system is I have no idea how to reproduce.

**Output of `linux-enable-ir-emitter -d /dev/video0 configure`**
<details><summary>configure command output</summary>

```
INFO: Stand in front of and close to the camera and make sure the room is well lit.
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
INFO: Configuring the camera: /dev/v4l/by-path/pci-0000:00:14.0-usb-0:5:1.0-video-index0 .
Is the video flashing? Press Y or N in the window
N pressed
Segmentation fault
```
</details>

**Output of `linux-enable-ir-emitter -v -d /dev/video0 configure`**
<details><summary>first run</summary>

```
INFO: Stand in front of and close to the camera and make sure the room is well lit.
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
Segmentation fault
```
</details>

<details><summary>second run (after reboot)</summary>

```
INFO: Stand in front of and close to the camera and make sure the room is well lit.
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
INFO: Configuring the camera: /dev/v4l/by-path/pci-0000:00:14.0-usb-0:5:1.0-video-index0 .
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: unit:  3  selector:  2
DEBUG: current: 2
DEBUG: minimum: 3
DEBUG: maximum: 3
DEBUG: new current: 3
DEBUG: 
DEBUG: unit:  4  selector:  2
DEBUG: current: 0 32
DEBUG: minimum: 0 0
DEBUG: maximum: 255 255
DEBUG: new current: 0 0
DEBUG: new current: 1 0
DEBUG: new current: 2 0
DEBUG: new current: 3 0
DEBUG: new current: 4 0
DEBUG: new current: 5 0
DEBUG: new current: 6 0
DEBUG: new current: 7 0
DEBUG: new current: 8 0
DEBUG: new current: 9 0
DEBUG: new current: 10 0
DEBUG: new current: 11 0
DEBUG: new current: 12 0
DEBUG: new current: 13 0
DEBUG: new current: 14 0
DEBUG: new current: 15 0
DEBUG: new current: 16 0
DEBUG: new current: 17 0
DEBUG: new current: 18 0
DEBUG: new current: 19 0
DEBUG: new current: 20 0
DEBUG: new current: 21 0
DEBUG: new current: 22 0
DEBUG: new current: 23 0
DEBUG: new current: 24 0
DEBUG: new current: 25 0
DEBUG: new current: 26 0
DEBUG: new current: 27 0
DEBUG: new current: 28 0
DEBUG: new current: 29 0
DEBUG: new current: 30 0
DEBUG: new current: 31 0
DEBUG: new current: 32 0
DEBUG: new current: 33 0
DEBUG: new current: 34 0
DEBUG: new current: 35 0
DEBUG: new current: 36 0
DEBUG: new current: 37 0
DEBUG: new current: 38 0
DEBUG: new current: 39 0
DEBUG: new current: 40 0
DEBUG: new current: 255 255
DEBUG: 
DEBUG: unit:  4  selector:  3
DEBUG: current: 0
DEBUG: minimum: 0
DEBUG: maximum: 255
DEBUG: new current: 1
Segmentation fault
```
</details>

**Output of `linux-enable-ir-emitter -v -d /dev/video0 configure -m`**
<details><summary>configure command output</summary>

```
INFO: Stand in front of and close to the camera and make sure the room is well lit.
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
INFO: Configuring the camera: /dev/v4l/by-path/pci-0000:00:14.0-usb-0:5:1.0-video-index0 .
Is the video flashing? Press Y or N in the window
terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.8.1) /home/runner/work/linux-enable-ir-emitter/linux-enable-ir-emitter/opencv/opencv-4.8.1/modules/highgui/src/window.cpp:971: error: (-215:Assertion failed) size.width>0 && size.height>0 in function 'imshow'

Aborted
```
</details>

**Output of `v4l2-ctl --list-devices`**
```
Integrated IR Camera: Integrate (usb-0000:00:14.0-5):
	/dev/video0
	/dev/video1
	/dev/media0

Integrated Camera: Integrated C (usb-0000:00:14.0-8):
	/dev/video2
	/dev/video3
	/dev/media1
```

**Output of `v4l2-ctl -d /dev/videoX --list-formats-ext`**

<details><summary>/dev/video0</summary>

```
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'MJPG' (Motion-JPEG, compressed)
		Size: Discrete 640x480
			Interval: Discrete 0.017s (60.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 160x120
			Interval: Discrete 0.033s (30.000 fps)
	[1]: 'YUYV' (YUYV 4:2:2)
		Size: Discrete 400x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 400x400
			Interval: Discrete 0.033s (30.000 fps)
```
</details>
<details><summary>/dev/video1</summary>

```
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture
```
</details>
<details><summary>/dev/video2</summary>

```
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'MJPG' (Motion-JPEG, compressed)
		Size: Discrete 1280x720
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x180
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 352x288
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 424x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 848x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 960x540
			Interval: Discrete 0.033s (30.000 fps)
	[1]: 'YUYV' (YUYV 4:2:2)
		Size: Discrete 1280x720
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 320x180
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 352x288
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 424x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 848x480
			Interval: Discrete 0.050s (20.000 fps)
		Size: Discrete 960x540
			Interval: Discrete 0.067s (15.000 fps)
```
</details>
<details><summary>/dev/video3</summary>

```
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture
```
</details>

**Output of `ls -l /dev/v4l/by-path`**
```
total 0
lrwxrwxrwx 1 root root 12 Oct 26 03:01 pci-0000:00:14.0-usb-0:5:1.0-video-index0 -> ../../video0
lrwxrwxrwx 1 root root 12 Oct 26 03:01 pci-0000:00:14.0-usb-0:5:1.0-video-index1 -> ../../video1
lrwxrwxrwx 1 root root 12 Oct 26 03:01 pci-0000:00:14.0-usb-0:8:1.0-video-index0 -> ../../video2
lrwxrwxrwx 1 root root 12 Oct 26 03:01 pci-0000:00:14.0-usb-0:8:1.0-video-index1 -> ../../video3
```

**Output of `howdy test`**
<details><summary>test command ouput</summary>

```
[ WARN:0@10.244] global cap_v4l.cpp:1119 tryIoctl VIDEOIO(V4L2:/dev/video0): select() timeout.

Opening a window with a test feed

Press ctrl+C in this terminal to quit
Click on the image to enable or disable slow mode

[ WARN:0@21.438] global cap_v4l.cpp:1119 tryIoctl VIDEOIO(V4L2:/dev/video0): select() timeout.
Failed to read camera specified in your 'device_path', aborting
```
</details>

**Give more information if you have**
If I run `configure` twice in a row it will seg fault immediately. I ran `mpv /dev/video0` after, mpv hangs and unable to output a video stream. I ran `configure` again with `-m` and OpenCV throws an exception. I had tried without `-d` but it unable to find any IR camera.

**Additional info**
 - Computer (or camera) model: Lenovo T480
 - Linux distribution: Linux Mint Xfce 21.2
 - Version of linux-enable-ir-emitter: 5.2.0 (but I did download the 5.2.1)

EmixamPP, 2023-10-26:
Hum, I think this is an issue related to OpenCV.

Could you past the output of `ldd /usr/lib64/linux-enable-ir-emitter/libcommands.so` please?

http403, 2023-10-26:
<details><summary>ldd libcommands.so output</summary>

```
	linux-vdso.so.1 (0x00007ffec1dd3000)
	libgtk-3.so.0 => /lib/x86_64-linux-gnu/libgtk-3.so.0 (0x00007fd03adee000)
	libgdk-3.so.0 => /lib/x86_64-linux-gnu/libgdk-3.so.0 (0x00007fd03ace7000)
	libcairo.so.2 => /lib/x86_64-linux-gnu/libcairo.so.2 (0x00007fd03abbf000)
	libgdk_pixbuf-2.0.so.0 => /lib/x86_64-linux-gnu/libgdk_pixbuf-2.0.so.0 (0x00007fd03ab8f000)
	libgobject-2.0.so.0 => /lib/x86_64-linux-gnu/libgobject-2.0.so.0 (0x00007fd03ab2f000)
	libglib-2.0.so.0 => /lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007fd03a9f3000)
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fd03a7c7000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fd03a6e0000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fd03a6c0000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fd03a498000)
	libgmodule-2.0.so.0 => /lib/x86_64-linux-gnu/libgmodule-2.0.so.0 (0x00007fd03a491000)
	libpangocairo-1.0.so.0 => /lib/x86_64-linux-gnu/libpangocairo-1.0.so.0 (0x00007fd03a47f000)
	libX11.so.6 => /lib/x86_64-linux-gnu/libX11.so.6 (0x00007fd03a33d000)
	libXi.so.6 => /lib/x86_64-linux-gnu/libXi.so.6 (0x00007fd03a329000)
	libXfixes.so.3 => /lib/x86_64-linux-gnu/libXfixes.so.3 (0x00007fd03a321000)
	libcairo-gobject.so.2 => /lib/x86_64-linux-gnu/libcairo-gobject.so.2 (0x00007fd03a315000)
	libatk-1.0.so.0 => /lib/x86_64-linux-gnu/libatk-1.0.so.0 (0x00007fd03a2eb000)
	libatk-bridge-2.0.so.0 => /lib/x86_64-linux-gnu/libatk-bridge-2.0.so.0 (0x00007fd03a2b3000)
	libepoxy.so.0 => /lib/x86_64-linux-gnu/libepoxy.so.0 (0x00007fd03a17c000)
	libfribidi.so.0 => /lib/x86_64-linux-gnu/libfribidi.so.0 (0x00007fd03a160000)
	libgio-2.0.so.0 => /lib/x86_64-linux-gnu/libgio-2.0.so.0 (0x00007fd039f87000)
	libpangoft2-1.0.so.0 => /lib/x86_64-linux-gnu/libpangoft2-1.0.so.0 (0x00007fd039f6c000)
	libpango-1.0.so.0 => /lib/x86_64-linux-gnu/libpango-1.0.so.0 (0x00007fd039f05000)
	libharfbuzz.so.0 => /lib/x86_64-linux-gnu/libharfbuzz.so.0 (0x00007fd039e36000)
	libfontconfig.so.1 => /lib/x86_64-linux-gnu/libfontconfig.so.1 (0x00007fd039dea000)
	libXinerama.so.1 => /lib/x86_64-linux-gnu/libXinerama.so.1 (0x00007fd039de5000)
	libXrandr.so.2 => /lib/x86_64-linux-gnu/libXrandr.so.2 (0x00007fd039dd8000)
	libXcursor.so.1 => /lib/x86_64-linux-gnu/libXcursor.so.1 (0x00007fd039dcc000)
	libXcomposite.so.1 => /lib/x86_64-linux-gnu/libXcomposite.so.1 (0x00007fd039dc7000)
	libXdamage.so.1 => /lib/x86_64-linux-gnu/libXdamage.so.1 (0x00007fd039dc0000)
	libxkbcommon.so.0 => /lib/x86_64-linux-gnu/libxkbcommon.so.0 (0x00007fd039d79000)
	libwayland-cursor.so.0 => /lib/x86_64-linux-gnu/libwayland-cursor.so.0 (0x00007fd039d6f000)
	libwayland-egl.so.1 => /lib/x86_64-linux-gnu/libwayland-egl.so.1 (0x00007fd039d6a000)
	libwayland-client.so.0 => /lib/x86_64-linux-gnu/libwayland-client.so.0 (0x00007fd039d59000)
	libXext.so.6 => /lib/x86_64-linux-gnu/libXext.so.6 (0x00007fd039d44000)
	libpixman-1.so.0 => /lib/x86_64-linux-gnu/libpixman-1.so.0 (0x00007fd039c97000)
	libfreetype.so.6 => /lib/x86_64-linux-gnu/libfreetype.so.6 (0x00007fd039bcf000)
	libpng16.so.16 => /lib/x86_64-linux-gnu/libpng16.so.16 (0x00007fd039b94000)
	libxcb-shm.so.0 => /lib/x86_64-linux-gnu/libxcb-shm.so.0 (0x00007fd039b8f000)
	libxcb.so.1 => /lib/x86_64-linux-gnu/libxcb.so.1 (0x00007fd039b65000)
	libxcb-render.so.0 => /lib/x86_64-linux-gnu/libxcb-render.so.0 (0x00007fd039b56000)
	libXrender.so.1 => /lib/x86_64-linux-gnu/libXrender.so.1 (0x00007fd039b47000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007fd039b2b000)
	libjpeg.so.8 => /lib/x86_64-linux-gnu/libjpeg.so.8 (0x00007fd039aaa000)
	libffi.so.8 => /lib/x86_64-linux-gnu/libffi.so.8 (0x00007fd039a9d000)
	libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007fd039a27000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fd03bbb7000)
	libdbus-1.so.3 => /lib/x86_64-linux-gnu/libdbus-1.so.3 (0x00007fd0399d7000)
	libatspi.so.0 => /lib/x86_64-linux-gnu/libatspi.so.0 (0x00007fd03999d000)
	libmount.so.1 => /lib/x86_64-linux-gnu/libmount.so.1 (0x00007fd039959000)
	libselinux.so.1 => /lib/x86_64-linux-gnu/libselinux.so.1 (0x00007fd03992d000)
	libthai.so.0 => /lib/x86_64-linux-gnu/libthai.so.0 (0x00007fd039922000)
	libgraphite2.so.3 => /lib/x86_64-linux-gnu/libgraphite2.so.3 (0x00007fd0398f9000)
	libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x00007fd0398c8000)
	libuuid.so.1 => /lib/x86_64-linux-gnu/libuuid.so.1 (0x00007fd0398bf000)
	libbrotlidec.so.1 => /lib/x86_64-linux-gnu/libbrotlidec.so.1 (0x00007fd0398b1000)
	libXau.so.6 => /lib/x86_64-linux-gnu/libXau.so.6 (0x00007fd0398ab000)
	libXdmcp.so.6 => /lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007fd0398a1000)
	libsystemd.so.0 => /lib/x86_64-linux-gnu/libsystemd.so.0 (0x00007fd0397da000)
	libblkid.so.1 => /lib/x86_64-linux-gnu/libblkid.so.1 (0x00007fd0397a3000)
	libpcre2-8.so.0 => /lib/x86_64-linux-gnu/libpcre2-8.so.0 (0x00007fd03970c000)
	libdatrie.so.1 => /lib/x86_64-linux-gnu/libdatrie.so.1 (0x00007fd039703000)
	libbrotlicommon.so.1 => /lib/x86_64-linux-gnu/libbrotlicommon.so.1 (0x00007fd0396de000)
	libbsd.so.0 => /lib/x86_64-linux-gnu/libbsd.so.0 (0x00007fd0396c6000)
	liblzma.so.5 => /lib/x86_64-linux-gnu/liblzma.so.5 (0x00007fd03969b000)
	libzstd.so.1 => /lib/x86_64-linux-gnu/libzstd.so.1 (0x00007fd0395cc000)
	liblz4.so.1 => /lib/x86_64-linux-gnu/liblz4.so.1 (0x00007fd0395ac000)
	libcap.so.2 => /lib/x86_64-linux-gnu/libcap.so.2 (0x00007fd0395a1000)
	libgcrypt.so.20 => /lib/x86_64-linux-gnu/libgcrypt.so.20 (0x00007fd039461000)
	libmd.so.0 => /lib/x86_64-linux-gnu/libmd.so.0 (0x00007fd039454000)
	libgpg-error.so.0 => /lib/x86_64-linux-gnu/libgpg-error.so.0 (0x00007fd03942e000)
```
</details>

EmixamPP, 2023-10-26:
Hum, I uploaded the wrong package on the release; the v5.2.0 again instead of the v5.2.1
Please reinstall the one attached and try again.

http403, 2023-10-26:
Unfortunately, the newer version doesn't solve the problem. It still seg fault when running `configure` and behave just like the before. After the fault no other application seems able to access the device after until a reboot.

However, after a reboot, `-m` does get pass unit 4 selector 3. It doesn't seg fault til the end but an seemingly unrelated exception from OpenCV instead.
<details><summary>configure command output</summary>

```
INFO: Stand in front of and close to the camera and make sure the room is well lit.
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
INFO: Configuring the camera: /dev/v4l/by-path/pci-0000:00:14.0-usb-0:5:1.0-video-index0 .
Is the video flashing? Press Y or N in the window
n
N pressed
DEBUG: unit:  3  selector:  2
DEBUG: current: 2
DEBUG: minimum: 3
DEBUG: maximum: 3
DEBUG: new current: 3
DEBUG: 
DEBUG: unit:  4  selector:  2
DEBUG: current: 1 0
DEBUG: minimum: 0 0
DEBUG: maximum: 255 255
DEBUG: new current: 0 0
DEBUG: new current: 1 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 2 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 3 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 4 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 5 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 6 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 7 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 8 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 9 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 10 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 11 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 12 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 13 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 14 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 15 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 16 0
[... 501 lines cut ...]
DEBUG: new current: 13
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: 
DEBUG: unit:  4  selector:  8
DEBUG: current: 3
DEBUG: minimum: 0
DEBUG: maximum: 255
DEBUG: new current: 0
DEBUG: new current: 1
Is the video flashing? Press Y or N in the window
terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.8.1) /home/runner/work/linux-enable-ir-emitter/linux-enable-ir-emitter/opencv/opencv-4.8.1/modules/highgui/src/window.cpp:971: error: (-215:Assertion failed) size.width>0 && size.height>0 in function 'imshow'

Aborted
```
</details>

Still, after this exception, no other application seemingly able to open the video stream again. Here an capture from Hodwy after a reboot.
[image]

---
I successfully setup Windows Hello on the same machine in a very dark room (it is a dual boot laptop) which I saw the IR emitters (2 total) are being triggered. This proofs the hardware isn't the problem, although I need to install a separate driver for it to work. Here's the device details. Hope this helps.
[image]
[image]

http403, 2023-10-26:
I just realized I never use `-e 2` to configure as it have two emitters. Still, specifying it doesn't change the situation.

**Output of `linux-enable-ir-emitter -v -d /dev/video0 configure -e 2 -m`**
<details><summary>configure command output</summary>

```
INFO: Stand in front of and close to the camera and make sure the room is well lit.
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
INFO: Configuring the camera: /dev/v4l/by-path/pci-0000:00:14.0-usb-0:5:1.0-video-index0 .
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: unit:  3  selector:  2
DEBUG: current: 2
DEBUG: minimum: 3
DEBUG: maximum: 3
DEBUG: new current: 3
DEBUG: 
DEBUG: unit:  4  selector:  2
DEBUG: current: 0 32
DEBUG: minimum: 0 0
DEBUG: maximum: 255 255
DEBUG: new current: 0 0
DEBUG: new current: 1 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 2 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 3 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 4 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 5 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 6 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 7 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 8 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 9 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 10 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 11 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 12 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 13 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 14 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 15 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 16 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 17 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 18 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 19 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 20 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 21 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 22 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 23 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 24 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 25 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 26 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 27 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 28 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 29 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 30 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 31 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 32 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 33 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 34 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 35 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 36 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 37 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 38 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 39 0
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: new current: 40 0
DEBUG: new current: 255 255
Is the video flashing? Press Y or N in the window
N pressed
DEBUG: 
DEBUG: unit:  4  selector:  3
DEBUG: current: 0
DEBUG: minimum: 0
DEBUG: maximum: 255
DEBUG: new current: 1
Is the video flashing? Press Y or N in the window
terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.8.1) /home/runner/work/linux-enable-ir-emitter/linux-enable-ir-emitter/opencv/opencv-4.8.1/modules/highgui/src/window.cpp:971: error: (-215:Assertion failed) size.width>0 && size.height>0 in function 'imshow'

Aborted
```
</details>

EmixamPP, 2023-10-28:
I will try to provide you a solution today

EmixamPP, 2023-10-29:
In pr #150 there's a bot comment that contains a link to download a version where I've added a `--no-gui` option that should solve your problem.

Please try again using the command `sudo linux-enable-ir-emitter -v configure -e 2 --no-gui`.

http403, 2023-10-29:
<details><summary>configure command output</summary>

```
> sudo linux-enable-ir-emitter -v -d /dev/video0 configure -e 2 --no-gui
INFO: Stand in front of and close to the camera and make sure the room is well lit.
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
INFO: Configuring the camera: /dev/v4l/by-path/pci-0000:00:14.0-usb-0:5:1.0-video-index0 .
Is the ir emitter flashing (not just turn on) ? Yes/No ? no
DEBUG: unit:  3  selector:  2
DEBUG: current: 2
DEBUG: minimum: 3
DEBUG: maximum: 3
DEBUG: new current: 3
DEBUG: 
DEBUG: unit:  4  selector:  2
DEBUG: current: 0 32
DEBUG: minimum: 0 0
DEBUG: maximum: 255 255
DEBUG: new current: 0 0
DEBUG: new current: 1 0
DEBUG: new current: 2 0
DEBUG: new current: 3 0
DEBUG: new current: 4 0
DEBUG: new current: 5 0
DEBUG: new current: 6 0
DEBUG: new current: 7 0
DEBUG: new current: 8 0
DEBUG: new current: 9 0
DEBUG: new current: 10 0
DEBUG: new current: 11 0
DEBUG: new current: 12 0
DEBUG: new current: 13 0
DEBUG: new current: 14 0
DEBUG: new current: 15 0
DEBUG: new current: 16 0
DEBUG: new current: 17 0
DEBUG: new current: 18 0
DEBUG: new current: 19 0
DEBUG: new current: 20 0
DEBUG: new current: 21 0
DEBUG: new current: 22 0
DEBUG: new current: 23 0
DEBUG: new current: 24 0
DEBUG: new current: 25 0
DEBUG: new current: 26 0
DEBUG: new current: 27 0
DEBUG: new current: 28 0
DEBUG: new current: 29 0
DEBUG: new current: 30 0
DEBUG: new current: 31 0
DEBUG: new current: 32 0
DEBUG: new current: 33 0
DEBUG: new current: 34 0
DEBUG: new current: 35 0
DEBUG: new current: 36 0
DEBUG: new current: 37 0
DEBUG: new current: 38 0
DEBUG: new current: 39 0
DEBUG: new current: 40 0
DEBUG: new current: 255 255
DEBUG: 
DEBUG: unit:  4  selector:  3
DEBUG: current: 0
DEBUG: minimum: 0
DEBUG: maximum: 255
DEBUG: new current: 1
Segmentation fault
```
<details>

It seg fault again on unit 4 selector 3.

EmixamPP, 2023-11-01:
I have no clue about what could be the issue, I tried using Valgrind to find a reason but there doesn't seem to be any problem on my side. 

Could you try building the project yourself? All the instructions are in the readme, but if you need any help, please don't hesitate.

http403, 2023-11-05:
I had compiled the `opencv_err` branch with opencv 4.5.4 accroding to the gh action, then run with `-e 2 -l -1 --no-gui`. I'm about 1/3 (180 86) into the configure which doesn't seg fault yet. Hopefully it split out anything other than seg fault and exceptions.

EmixamPP, 2023-11-05:
Ho no, you should not use `-m -l -1` together , this will takes so much time and maybe you do not need the fully exhaustive search.

You could kill the process and reboot, then retry with the default parameters (just add `-e 2` and also `-v`)

If the configuration fails, show me the output I will guide you.

EmixamPP, 2023-11-05:
Did you tried without `-l -1`? This will takes so much time and maybe you do not need the fully exhaustive search.

You could kill the process and reboot, then retry with the default parameters (just add `-e 2` and also `-v`)

If the configuration fails, show me the output I will guide you.

http403, 2023-11-05:
It's on 223 so not far from completing. If I quit now, its kind of wasteful.

http403, 2023-11-05:
It is wasteful. I thought is went passed unit 4 selector 3, but it is not. Is just segfault on that spot.

http403, 2023-11-05:
I rebooted and run with `-m -e 2 -g`, went pass unit 4 selector 3. However, when got into it, the current is different from pair to single, and significantly slower. I'm talking 2-3 seconds before the Y/N prompt show up. However, there is good news, if I do it slow enough, it does NOT segfault.

<details><summary>configure command output</summary>

```
INFO: Stand in front of and close to the camera and make sure the room is well lit.
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
INFO: Configuring the camera: /dev/v4l/by-path/pci-0000:00:14.0-usb-0:5:1.0-video-index0 .
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: unit:  3  selector:  2
DEBUG: current: 2
DEBUG: minimum: 3
DEBUG: maximum: 3
DEBUG: new current: 3
DEBUG: 
DEBUG: unit:  4  selector:  2
DEBUG: current: 0 32
DEBUG: minimum: 0 0
DEBUG: maximum: 255 255
DEBUG: new current: 0 0
DEBUG: new current: 1 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 2 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 3 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 4 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 5 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 6 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 7 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 8 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 9 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 10 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 11 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 12 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 13 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 14 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 15 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 16 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 17 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 18 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 19 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 20 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 21 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 22 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 23 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 24 0
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 25 0
[... 91 lines cut ...]
DEBUG: new current: 28
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 29
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 30
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 31
Is the ir emitter flashing (not just turn on) ? Yes/No ? n
DEBUG: new current: 32
DEBUG: new current: 33
DEBUG: new current: 34
DEBUG: new current: 35
DEBUG: new current: 36
DEBUG: new current: 37
DEBUG: new current: 38
DEBUG: new current: 39
DEBUG: new current: 40
DEBUG: new current: 255
DEBUG: 
ERROR: The configuration has failed.
ERROR: Please retry in manual mode by adding the '-m' option.
INFO: Do not hesitate to visit the GitHub !
INFO: https://github.com/EmixamPP/linux-enable-ir-emitter/blob/master/docs/README.md
```
</details>

EmixamPP, 2023-11-08:
Hum, sadly I don't think your camera gives enough information to let my tool find an instruction which enable your emitter

http403, 2023-11-09:
Seems like a dead end for now, and I'm not very expert in low level hardware things. Do you wish to keep this issue up?

EmixamPP, 2023-11-09:
I will keep issue in mind. I hope find a way to enable your emitter in the coming releases!

EmixamPP, 2023-12-30:
Hey, please try the v6 https://github.com/EmixamPP/linux-enable-ir-emitter/pull/154. There is a bot comment with a link to the tarball. Please uninstall any existing version first:
```
sudo rm -rf /usr/lib64/linux-enable-ir-emitter \
/usr/bin/linux-enable-ir-emitter \
/usr/lib/systemd/system/linux-enable-ir-emitter.service \
/etc/udev/rules.d/99-linux-enable-ir-emitter.rules \
/usr/share/bash-completition/completitions/linux-enable-ir-emitter \
/etc/linux-enable-ir-emitter
```

And past here the content of the file inside `/etc/linux-enable-ir-emitter`

http403, 2023-12-30:
Sorry I couldn't. For some reason that the IR cam isn't recognizable via `v4l2-ctl`. Not working in Windows either with a 3 minutes check. It does shown up in `lsusb` and Windows' Device Mgr. I have quite a long time not booted into Windows thus not using the IR camera. I don't know at what point can cause a problem.

EmixamPP, 2024-02-02:
I think I've solved the problem of segfault #155 you had. You can give a new try to the v6 if you want to. 
But if you no longer want to use this tool (at least for now), no problem of course!

closed 2024-02-02 by EmixamPP