EmixamPP/linux-enable-ir-emitter #149: Driver generated but not configured.

Ollix27, 2023-10-26:
**Ouput of `linux-enable-ir-emitter -v configure`**
<details><summary>configure command output</summary>

```

```
</details>

**Output of `v4l2-ctl --list-devices`**
```
INFO: The driver has been successfully generated.
CRITICAL: No driver have been configured.
```

**Ouput of `v4l2-ctl -d /dev/videoX --list-formats-ext`**

<details><summary>v4l2 device output</summary>

```
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'GREY' (8-bit Greyscale)
		Size: Discrete 640x360
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.025s (40.000 fps)

```
</details>

**Output of `ls -l /dev/v4l/by-path`**
```
lrwxrwxrwx 1 root root 12 10-26 12:19 pci-0000:00:14.0-usb-0:8:1.0-video-index0 -> ../../video0
lrwxrwxrwx 1 root root 12 10-26 12:19 pci-0000:00:14.0-usb-0:8:1.0-video-index1 -> ../../video1
lrwxrwxrwx 1 root root 12 10-26 12:19 pci-0000:00:14.0-usb-0:8:1.2-video-index0 -> ../../video2
lrwxrwxrwx 1 root root 12 10-26 12:19 pci-0000:00:14.0-usb-0:8:1.2-video-index1 -> ../../video3
lrwxrwxrwx 1 root root 12 10-26 12:19 pci-0000:00:14.0-usbv2-0:8:1.0-video-index0 -> ../../video0
lrwxrwxrwx 1 root root 12 10-26 12:19 pci-0000:00:14.0-usbv2-0:8:1.0-video-index1 -> ../../video1
lrwxrwxrwx 1 root root 12 10-26 12:19 pci-0000:00:14.0-usbv2-0:8:1.2-video-index0 -> ../../video2
lrwxrwxrwx 1 root root 12 10-26 12:19 pci-0000:00:14.0-usbv2-0:8:1.2-video-index1 -> ../../video3

```

**Give more information if you have**

**Additional info**
 - Computer (or camera) model: Lenovo Yoga Slim 7 Pro-14 14IAH7
 - Linux distribution: EndeavourOS
 - Version of linux-enable-ir-emitter:  linux-enable-ir-emitter 5.2.1

EmixamPP, 2023-10-26:
Could you past the output of `ls /etc/linux-enable-ir-emitter` please?

EmixamPP, 2023-10-26:
Hum, I uploaded the wrong package on the release; the v5.2.0 again instead of the v5.2.1
Please reinstall the one attached and try again.
[linux-enable-ir-emitter-5.2.1.systemd.x86-64.tar.gz](https://github.com/EmixamPP/linux-enable-ir-emitter/files/13182891/linux-enable-ir-emitter-5.2.1.systemd.x86-64.tar.gz)

Ollix27, 2023-10-31:
Sorry, but how can i reinstall it?

EmixamPP, 2023-10-31:
Just by executing the install command specified in the readme or in the bot comment

Ollix27, 2023-10-31:
Okay, trying it now.

Ollix27, 2023-10-31:
Oh, and also it does this thing if i stop the command, it makes it so that the root password always errors out and i need to reboot.

Ollix27, 2023-10-31:
Still the same error, and i still have the 5.2.1v

EmixamPP, 2023-10-31:
Could you past the output of ls /etc/linux-enable-ir-emitter please?

Ollix27, 2023-10-31:
There is nothing in the directory

Ollix27, 2023-10-31:
And i also had the 5.2.1v when i downoladed it using the yay command

apodl1, 2023-10-31:
I have a same kind of problem on Yoga Pro 7 14APH8 (European version of the same laptop) and Fedora.

**Ouput of `linux-enable-ir-emitter -v configure`**
<details><summary>configure command output</summary>

```
Is the video flashing? Press Y or N in the window

** (python3:69030): WARNING **: 18:53:57.119: AT-SPI: Could not obtain desktop path or name

** (python3:69030): WARNING **: 18:53:57.186: atk-bridge: GetRegisteredEvents returned message with unknown signature

** (python3:69030): WARNING **: 18:53:57.186: atk-bridge: get_device_events_reply: unknown signature
N pressed
DEBUG: unit:  13  selector:  14
DEBUG: current: 0 0
DEBUG: minimum: 0 0
DEBUG: maximum: 2 100
DEBUG: new current: 1 0
DEBUG: new current: 2 0
DEBUG: new current: 0 1
DEBUG: new current: 1 1
DEBUG: new current: 2 1
DEBUG: new current: 0 2
DEBUG: new current: 1 2
DEBUG: new current: 2 2
DEBUG: new current: 0 3
DEBUG: new current: 1 3
DEBUG: new current: 2 3
DEBUG: new current: 0 4
DEBUG: new current: 1 4
DEBUG: new current: 2 4
DEBUG: new current: 0 5
DEBUG: new current: 1 5
DEBUG: new current: 2 5
DEBUG: new current: 0 6
DEBUG: new current: 1 6
DEBUG: new current: 2 6
DEBUG: new current: 0 7
DEBUG: new current: 1 7
DEBUG: new current: 2 7
DEBUG: new current: 0 8
DEBUG: new current: 1 8
DEBUG: new current: 2 8
DEBUG: new current: 0 9
DEBUG: new current: 1 9
DEBUG: new current: 2 9
DEBUG: new current: 0 10
DEBUG: new current: 1 10
DEBUG: new current: 2 10
DEBUG: new current: 0 11
DEBUG: new current: 1 11
DEBUG: new current: 2 11
DEBUG: new current: 0 12
DEBUG: new current: 1 12
DEBUG: new current: 2 12
DEBUG: new current: 0 13
DEBUG: new current: 1 13
DEBUG: new current: 2 100
DEBUG: 
DEBUG: unit:  13  selector:  16
DEBUG: current: 0
DEBUG: minimum: 0
DEBUG: maximum: 1
DEBUG: new current: 1
DEBUG: 
DEBUG: unit:  14  selector:  2
DEBUG: current: 1 0 0 0 0 0 0 16 39 0 0 0 0 0 0
DEBUG: minimum: 0 0 0 0 0 0 0 16 39 0 0 0 0 0 0
DEBUG: maximum: 7 0 0 0 0 0 0 200 20 5 0 0 0 0 0
[... 186 lines cut ...]
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

```
</details>

**Output of `ls -l /dev/v4l/by-path`**
```
total 0
lrwxrwxrwx. 1 root root 12 Oct 30 19:34 pci-0000:63:00.4-usb-0:1:1.0-video-index0 -> ../../video0
lrwxrwxrwx. 1 root root 12 Oct 30 19:34 pci-0000:63:00.4-usb-0:1:1.0-video-index1 -> ../../video1
lrwxrwxrwx. 1 root root 12 Oct 30 19:34 pci-0000:63:00.4-usb-0:1:1.2-video-index0 -> ../../video2
lrwxrwxrwx. 1 root root 12 Oct 30 19:34 pci-0000:63:00.4-usb-0:1:1.2-video-index1 -> ../../video3

```

**Give more information if you have**
The "INFO: The driver has been successfully generated.
CRITICAL: No driver have been configured."-message shows up when I configure the camera in manual mode.  I am able to find the flashing image around the last options, and the message prints after pressing "y". After that the ir-camera works (with the test-option and howdy), until a reboot. After rebooting I have to manually configure it again.
I also can't set "boot enable" due to "CRITICAL: No driver have been configured."-error.

**Additional info**
 - Computer (or camera) model: Lenovo Yoga Pro 7 14APH8
 - Linux distribution: Fedora 38
 - Version of linux-enable-ir-emitter: linux-enable-ir-emitter 5.2.1

EmixamPP, 2023-11-01:
In pr #150 there is a bot comment that contains a link to download a tarball, could you please try again with this version and make sure this issue is fixed? Thank you!

apodl1, 2023-11-03:
Yes, the latest version worked. Thank you!

EmixamPP, 2023-11-03:
Thank you very much for reporting this problem and testing the fix.

This problem was really ridiculous. Unfortunately, I don't have the possibility of automating tests, so this kind of issues happens a little too often :/

If you like the project, do not hesitate to star the repository to support me, thank you!

closed 2023-11-03 by EmixamPP