EmixamPP/linux-enable-ir-emitter #151: Fails to build with Clang

dthelegend, 2023-10-31:
**Describe the bug**
I found that Clang finds many issues with the code and there are too many errors to include here. Most of them seem to be generated by opencv, but the reason for it not compiling is not clear to me. I have left my build log attached [here](https://github.com/EmixamPP/linux-enable-ir-emitter/files/13210712/build.log). Compiles fine with GCC.

```
1620 errors generated.
ninja: build stopped: subcommand failed.
```

**How to reproduce**
Compile with the official Clang package
```
CC=clang CXX=clang CFLAGS="-ferror-limit=0" CXXFLAGS="-ferror-limit=0" meson setup --wipe build
```

Also not known if related, but this does not compile with opencv-cuda.

EmixamPP, 2023-11-01:
Have you compiled using meson?

EmixamPP, 2023-11-01:
Please compile using meson. The errors you have are only related to coding style.