EmixamPP/linux-enable-ir-emitter #152: Systemd service fails to start

iyanmv, 2023-11-05:
**Output of `linux-enable-ir-emitter boot status`**
```
sudo linux-enable-ir-emitter boot status
× linux-enable-ir-emitter.service - enable the infrared emitter
     Loaded: loaded (/usr/lib/systemd/system/linux-enable-ir-emitter.service; enabled; preset: disabled)
     Active: failed (Result: exit-code) since Sun 2023-11-05 11:53:18 CET; 1min 16s ago
    Process: 119264 ExecStartPre=/sbin/modprobe uvcvideo (code=exited, status=0/SUCCESS)
    Process: 119265 ExecStart=/usr/bin/linux-enable-ir-emitter run (code=exited, status=1/FAILURE)
   Main PID: 119265 (code=exited, status=1/FAILURE)
        CPU: 92ms

Nov 05 11:53:18 thinkpad systemd[1]: Starting enable the infrared emitter...
Nov 05 11:53:18 thinkpad linux-enable-ir-emitter[119265]: CRITICAL: No driver for  has been configured.
Nov 05 11:53:18 thinkpad systemd[1]: linux-enable-ir-emitter.service: Main process exited, code=exited, status=1/FAILURE
Nov 05 11:53:18 thinkpad systemd[1]: linux-enable-ir-emitter.service: Failed with result 'exit-code'.
Nov 05 11:53:18 thinkpad systemd[1]: Failed to start enable the infrared emitter.
```

**Output of `cat /etc/linux-enable-ir-emitter/*`**
```
(no files here)
```

**Give more information if you have**
I can rerun `sudo linux-enable-ir-emitter configure` and it says that it is already configured.

```
INFO: Stand in front of and close to the camera and make sure the room is well lit.
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
INFO: Configuring the camera: /dev/v4l/by-path/pci-0000:00:14.0-usbv2-0:4:1.2-video-index0 .
Is the video flashing? Press Y or N in the window
Y pressed
ERROR: Your emiter is already working, skipping the configuration.
```

**Additional info**
 - Computer (or camera) model: ThinkPad X1 Yoga Gen 6
 - Linux distribution: Arch Linux
 - Version of linux-enable-ir-emitter: 5.2.3 (actually from 5.2.4 tarball, but wrong version in meson)

It used to work in the past, but I don't know exactly with which version it stopped working.

EmixamPP, 2023-11-05:
I see that your configuration file has been removed from the etc folder. Reboot and re-configure it again

iyanmv, 2023-11-05:
I have done that before opening the issue. In any case, I tried again, and nothing is saved to `/etc/linux-enable-ir-emitter/`

```shell
$ sudo linux-enable-ir-emitter configure
[sudo] password for iyan: 
INFO: Stand in front of and close to the camera and make sure the room is well lit.
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
INFO: Configuring the camera: /dev/v4l/by-path/pci-0000:00:14.0-usb-0:4:1.2-video-index0 .
Is the video flashing? Press Y or N in the window
Y pressed
ERROR: Your emiter is already working, skipping the configuration.
$ ls -al /etc/linux-enable-ir-emitter 
total 16
drwxr-xr-x   2 root root  4096 Nov  5 11:52 .
drwxr-xr-x 159 root root 12288 Nov  5 12:31 ..
```

EmixamPP, 2023-11-05:
The problem is that currently your infrared emitter is working, so the tool cannot find which instruction is used to enable it. 

You could try to say "no" to the first question, let the process fail; may be the camera will be reset, and retry.

iyanmv, 2023-11-05:
Alright, that worked. I guess it would be nice that `linux-enable-ir-emitter configure` leaves the system in a state that would make the systemd not fail. For example, even if it had to do nothing to make the infrared emitter work, it could rewrite the config files in `/etc/linux-enable-ir-emitter`. What do you think?

closed 2023-11-05 by iyanmv

EmixamPP, 2023-11-05:
The configure command already let the system in a state where nothing is modified. The configuration files in /etc/linux-enable-ir-emitter are only delete by the delete command, or overwritted by a successfull configure commmand.

EmixamPP, 2023-11-05:
In case of failure, the configure command lets already the system in a state where nothing is modified. The configuration files in /etc/linux-enable-ir-emitter are only delete by the delete command, or overwritten by a successful configure command.