EmixamPP/linux-enable-ir-emitter #153: When the ir light blinks, it failed to configure with an exception Assertion '__n < this->size()' failed.

lwzhenglittle, 2023-11-06:
**Ouput of `linux-enable-ir-emitter -v configure`**
<details><summary>configure command output</summary>

```
sudo linux-enable-ir-emitter configure
INFO: Stand in front of and close to the camera and make sure the room is well lit.
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
INFO: Configuring the camera: /dev/v4l/by-path/pci-0000:00:14.0-usbv2-0:5:1.2-video-index0 .
Is the video flashing? Press Y or N in the window
qt.qpa.theme.dbus: Session DBus not running.
qt.qpa.theme.dbus: Application will not react to setting changes.
 Check your DBus installation.
N pressed
/usr/include/c++/13.2.1/bits/stl_vector.h:1125: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = std::unique_ptr<std::vector<int> >; _Alloc = std::allocator<std::unique_ptr<std::vector<int> > >; reference = std::unique_ptr<std::vector<int> >&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
[1]    39612 IOT instruction  sudo linux-enable-ir-emitter configure
```
</details>

**Output of `v4l2-ctl --list-devices`**
```
USB Camera: USB Camera (usb-0000:00:14.0-3):
	/dev/video0
	/dev/video1
	/dev/media0

Integrated Camera: Integrated C (usb-0000:00:14.0-5):
	/dev/video2
	/dev/video3
	/dev/video4
	/dev/video5
	/dev/media1
	/dev/media2

```

**Ouput of `v4l2-ctl -d /dev/videoX --list-formats-ext`**

<details><summary>v4l2 device output</summary>
Integrated Camera: Integrated C (usb-0000:00:14.0-5) is my IR Camera, so only list /dev/video[2,4] (3, 5 no output)
```
v4l2-ctl -d /dev/video2 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'MJPG' (Motion-JPEG, compressed)
		Size: Discrete 1280x720
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x180
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 352x288
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 424x240
			Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 848x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 960x540
			Interval: Discrete 0.033s (30.000 fps)
	[1]: 'YUYV' (YUYV 4:2:2)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x180
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 352x288
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 424x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 848x480
			Interval: Discrete 0.050s (20.000 fps)
		Size: Discrete 960x540
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.100s (10.000 fps)
v4l2-ctl -d /dev/video4 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'GREY' (8-bit Greyscale)
		Size: Discrete 360x360
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
```
</details>

**Output of `ls -l /dev/v4l/by-path`**
```
lrwxrwxrwx 1 root root 12 11月 6日 10:14 pci-0000:00:14.0-usb-0:3:1.0-video-index0 -> ../../video0
lrwxrwxrwx 1 root root 12 11月 6日 10:14 pci-0000:00:14.0-usb-0:3:1.0-video-index1 -> ../../video1
lrwxrwxrwx 1 root root 12 11月 6日 10:14 pci-0000:00:14.0-usb-0:5:1.0-video-index0 -> ../../video2
lrwxrwxrwx 1 root root 12 11月 6日 10:14 pci-0000:00:14.0-usb-0:5:1.0-video-index1 -> ../../video3
lrwxrwxrwx 1 root root 12 11月 6日 10:14 pci-0000:00:14.0-usb-0:5:1.2-video-index0 -> ../../video4
lrwxrwxrwx 1 root root 12 11月 6日 10:14 pci-0000:00:14.0-usb-0:5:1.2-video-index1 -> ../../video5
lrwxrwxrwx 1 root root 12 11月 6日 10:14 pci-0000:00:14.0-usbv2-0:3:1.0-video-index0 -> ../../video0
lrwxrwxrwx 1 root root 12 11月 6日 10:14 pci-0000:00:14.0-usbv2-0:3:1.0-video-index1 -> ../../video1
lrwxrwxrwx 1 root root 12 11月 6日 10:14 pci-0000:00:14.0-usbv2-0:5:1.0-video-index0 -> ../../video2
lrwxrwxrwx 1 root root 12 11月 6日 10:14 pci-0000:00:14.0-usbv2-0:5:1.0-video-index1 -> ../../video3
lrwxrwxrwx 1 root root 12 11月 6日 10:14 pci-0000:00:14.0-usbv2-0:5:1.2-video-index0 -> ../../video4
lrwxrwxrwx 1 root root 12 11月 6日 10:14 pci-0000:00:14.0-usbv2-0:5:1.2-video-index1 -> ../../video5

```

**Give more information if you have**

When the ir light starts to blink, linux-enable-ir-emitter will stuck (not popping up the preview window and no output) and throw an exception

 "/usr/include/c++/13.2.1/bits/stl_vector.h:1125: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = std::unique_ptr<std::vector<int> >; _Alloc = std::allocator<std::unique_ptr<std::vector<int> > >; reference = std::unique_ptr<std::vector<int> >&; size_type = long unsigned int]: Assertion '__n < this->size()' failed."

and it will die.

**Additional info**
 - Computer (or camera) model: X-STATION V2
 - Linux distribution: Arch Linux with linux 6.5.7-1-ck kernel
 - Version of linux-enable-ir-emitter: linux-enable-ir-emitter 5.2.3

EmixamPP, 2023-11-08:
Could you try using the `--no-gui` option?

lwzhenglittle, 2023-11-21:
I am so sorry to see your response so late.

Tried `sudo linux-enable-ir-emitter -v configure --no-gui`.

Output:

```
sudo linux-enable-ir-emitter -v configure --no-gui
INFO: Stand in front of and close to the camera and make sure the room is well lit.
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
INFO: Configuring the camera: /dev/v4l/by-path/pci-0000:00:14.0-usbv2-0:5:1.2-video-index0 .
Is the ir emitter flashing (not just turn on) ? Yes/No ? No
DEBUG: unit:  9  selector:  10
DEBUG: current: 0 8 0 187 187 187 187 187
DEBUG: minimum: 0 0 0 0 0 0 0 0
DEBUG: maximum: 255 255 255 255 255 255 255 255
DEBUG: new current: 0 0 0 0 0 0 0 0
DEBUG: new current: 1 0 0 0 0 0 0 0
DEBUG: new current: 2 0 0 0 0 0 0 0
DEBUG: new current: 3 0 0 0 0 0 0 0
DEBUG: new current: 4 0 0 0 0 0 0 0
DEBUG: new current: 5 0 0 0 0 0 0 0
DEBUG: new current: 6 0 0 0 0 0 0 0
DEBUG: new current: 7 0 0 0 0 0 0 0
DEBUG: new current: 8 0 0 0 0 0 0 0
DEBUG: new current: 9 0 0 0 0 0 0 0
DEBUG: new current: 10 0 0 0 0 0 0 0
DEBUG: new current: 11 0 0 0 0 0 0 0
DEBUG: new current: 12 0 0 0 0 0 0 0
DEBUG: new current: 13 0 0 0 0 0 0 0
DEBUG: new current: 14 0 0 0 0 0 0 0
DEBUG: new current: 15 0 0 0 0 0 0 0
DEBUG: new current: 16 0 0 0 0 0 0 0
DEBUG: new current: 17 0 0 0 0 0 0 0
DEBUG: new current: 18 0 0 0 0 0 0 0
DEBUG: new current: 19 0 0 0 0 0 0 0
DEBUG: new current: 20 0 0 0 0 0 0 0
DEBUG: new current: 21 0 0 0 0 0 0 0
DEBUG: new current: 22 0 0 0 0 0 0 0
DEBUG: new current: 23 0 0 0 0 0 0 0
DEBUG: new current: 24 0 0 0 0 0 0 0
DEBUG: new current: 25 0 0 0 0 0 0 0
DEBUG: new current: 26 0 0 0 0 0 0 0
DEBUG: new current: 27 0 0 0 0 0 0 0
DEBUG: new current: 28 0 0 0 0 0 0 0
DEBUG: new current: 29 0 0 0 0 0 0 0
DEBUG: new current: 30 0 0 0 0 0 0 0
DEBUG: new current: 31 0 0 0 0 0 0 0
DEBUG: new current: 32 0 0 0 0 0 0 0
DEBUG: new current: 33 0 0 0 0 0 0 0
DEBUG: new current: 34 0 0 0 0 0 0 0
DEBUG: new current: 35 0 0 0 0 0 0 0
DEBUG: new current: 36 0 0 0 0 0 0 0
DEBUG: new current: 37 0 0 0 0 0 0 0
DEBUG: new current: 38 0 0 0 0 0 0 0
DEBUG: new current: 39 0 0 0 0 0 0 0
DEBUG: new current: 40 0 0 0 0 0 0 0
DEBUG: new current: 255 255 255 255 255 255 255 255
DEBUG: 
DEBUG: unit:  10  selector:  2
DEBUG: current: 1 0 0 0 0 0 0 16 39 0 0 0 0 0 0
DEBUG: minimum: 0 0 0 0 0 0 0 16 39 0 0 0 0 0 0
DEBUG: maximum: 7 0 0 0 0 0 0 244 21 5 0 0 0 0 0
DEBUG: new current: 0 0 0 0 0 0 0 16 39 0 0 0 0 0 0
DEBUG: new current: 1 0 0 0 0 0 0 16 39 0 0 0 0 0 0
DEBUG: new current: 2 0 0 0 0 0 0 16 39 0 0 0 0 0 0
DEBUG: new current: 3 0 0 0 0 0 0 16 39 0 0 0 0 0 0
DEBUG: new current: 4 0 0 0 0 0 0 16 39 0 0 0 0 0 0
DEBUG: new current: 5 0 0 0 0 0 0 16 39 0 0 0 0 0 0
DEBUG: new current: 6 0 0 0 0 0 0 16 39 0 0 0 0 0 0
DEBUG: new current: 7 0 0 0 0 0 0 16 39 0 0 0 0 0 0
DEBUG: new current: 0 0 0 0 0 0 0 17 39 0 0 0 0 0 0
DEBUG: new current: 1 0 0 0 0 0 0 17 39 0 0 0 0 0 0
DEBUG: new current: 2 0 0 0 0 0 0 17 39 0 0 0 0 0 0
DEBUG: new current: 3 0 0 0 0 0 0 17 39 0 0 0 0 0 0
DEBUG: new current: 4 0 0 0 0 0 0 17 39 0 0 0 0 0 0
DEBUG: new current: 5 0 0 0 0 0 0 17 39 0 0 0 0 0 0
DEBUG: new current: 6 0 0 0 0 0 0 17 39 0 0 0 0 0 0
DEBUG: new current: 7 0 0 0 0 0 0 17 39 0 0 0 0 0 0
DEBUG: new current: 0 0 0 0 0 0 0 18 39 0 0 0 0 0 0
DEBUG: new current: 1 0 0 0 0 0 0 18 39 0 0 0 0 0 0
DEBUG: new current: 2 0 0 0 0 0 0 18 39 0 0 0 0 0 0
DEBUG: new current: 3 0 0 0 0 0 0 18 39 0 0 0 0 0 0
DEBUG: new current: 4 0 0 0 0 0 0 18 39 0 0 0 0 0 0
DEBUG: new current: 5 0 0 0 0 0 0 18 39 0 0 0 0 0 0
DEBUG: new current: 6 0 0 0 0 0 0 18 39 0 0 0 0 0 0
DEBUG: new current: 7 0 0 0 0 0 0 18 39 0 0 0 0 0 0
DEBUG: new current: 0 0 0 0 0 0 0 19 39 0 0 0 0 0 0
DEBUG: new current: 1 0 0 0 0 0 0 19 39 0 0 0 0 0 0
DEBUG: new current: 2 0 0 0 0 0 0 19 39 0 0 0 0 0 0
DEBUG: new current: 3 0 0 0 0 0 0 19 39 0 0 0 0 0 0
DEBUG: new current: 4 0 0 0 0 0 0 19 39 0 0 0 0 0 0
DEBUG: new current: 5 0 0 0 0 0 0 19 39 0 0 0 0 0 0
DEBUG: new current: 6 0 0 0 0 0 0 19 39 0 0 0 0 0 0
DEBUG: new current: 7 0 0 0 0 0 0 19 39 0 0 0 0 0 0
DEBUG: new current: 0 0 0 0 0 0 0 20 39 0 0 0 0 0 0
DEBUG: new current: 1 0 0 0 0 0 0 20 39 0 0 0 0 0 0
DEBUG: new current: 2 0 0 0 0 0 0 20 39 0 0 0 0 0 0
DEBUG: new current: 3 0 0 0 0 0 0 20 39 0 0 0 0 0 0
DEBUG: new current: 4 0 0 0 0 0 0 20 39 0 0 0 0 0 0
DEBUG: new current: 5 0 0 0 0 0 0 20 39 0 0 0 0 0 0
DEBUG: new current: 6 0 0 0 0 0 0 20 39 0 0 0 0 0 0
DEBUG: new current: 7 0 0 0 0 0 0 20 39 0 0 0 0 0 0
DEBUG: new current: 0 0 0 0 0 0 0 21 39 0 0 0 0 0 0
DEBUG: new current: 7 0 0 0 0 0 0 244 21 5 0 0 0 0 0
DEBUG: 
DEBUG: Using current as minimum control.
DEBUG: unit:  10  selector:  6
DEBUG: current: 1 3 1 0 0 0 0 0 0
DEBUG: minimum: 1 3 1 0 0 0 0 0 0
DEBUG: maximum: 1 3 3 0 0 0 0 0 0
DEBUG: new current: 1 3 2 0 0 0 0 0 0
/usr/include/c++/13.2.1/bits/stl_vector.h:1125: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = std::unique_ptr<std::vector<int> >; _Alloc = std::allocator<std::unique_ptr<std::vector<int> > >; reference = std::unique_ptr<std::vector<int> >&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
[1]    6048 IOT instruction  sudo linux-enable-ir-emitter -v configure --no-gui
```

EmixamPP, 2023-11-25:
Let's quick fix your issue:
In `/dev/v4l/by-path/pci-0000:00:14.0-usbv2-0:5:1.2-video-index0_emitter0.driver` please write the following content:
```
device=pci-0000:00:14.0-usbv2-0:5:1.2-video-index0
unit=10
selector=6
control0=1
control1=3
control2=2
control3=0
control4=0
control5=0
control6=0
control7=0
control8=0
```
Then execute `sudo linux-enable-ir-emitter boot enable`

lwzhenglittle, 2023-12-02:
Sorry to see your reply so late again. I will test it soon.

lwzhenglittle, 2023-12-02:
I add that content in the file and execute the command, but the service didn't start successfully with the following message:

```
× linux-enable-ir-emitter.service - enable the infrared emitter
     Loaded: loaded (/usr/lib/systemd/system/linux-enable-ir-emitter.service; enabled; preset: disabled)
     Active: failed (Result: exit-code) since Sat 2023-12-02 12:23:52 CST; 10min ago
    Process: 476 ExecStartPre=/sbin/modprobe uvcvideo (code=exited, status=0/SUCCESS)
    Process: 538 ExecStart=/usr/bin/linux-enable-ir-emitter run (code=exited, status=1/FAILURE)
   Main PID: 538 (code=exited, status=1/FAILURE)
        CPU: 123ms

12月 02 12:23:51 lwzheng-V2 systemd[1]: Starting enable the infrared emitter...
12月 02 12:23:52 lwzheng-V2 linux-enable-ir-emitter[538]: CRITICAL: Unable to obtain the /dev/videoX path
12月 02 12:23:52 lwzheng-V2 systemd[1]: linux-enable-ir-emitter.service: Main process exited, code=exited, status=1/FAILURE
12月 02 12:23:52 lwzheng-V2 systemd[1]: linux-enable-ir-emitter.service: Failed with result 'exit-code'.
12月 02 12:23:52 lwzheng-V2 systemd[1]: Failed to start enable the infrared emitter.
```

I manually run `sudo linux-enable-ir-emitter run`

it shows: 

```
CRITICAL: Unable to obtain the /dev/videoX path
```

closed 2023-12-25 by EmixamPP