EmixamPP/linux-enable-ir-emitter #155: Segmentation Fault

azblurbit, 2023-11-28:
**Ouput of `linux-enable-ir-emitter -v configure`**
<details><summary>configure command output</summary>

```
$ sudo linux-enable-ir-emitter configure
INFO: Stand in front of and close to the camera and make sure the room is well lit.
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
Segmentation fault

```
</details>

**Output of `v4l2-ctl --list-devices`**
```
$ v4l2-ctl --list-devices
Integrated Camera: Integrated C (usb-0000:00:14.0-6):
	/dev/video0
	/dev/video1
	/dev/video2
	/dev/video3
	/dev/media0
	/dev/media1

```

**Ouput of `v4l2-ctl -d /dev/videoX --list-formats-ext`**

<details><summary>v4l2 device output</summary>

```
$ v4l2-ctl -d /dev/video0 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'MJPG' (Motion-JPEG, compressed)
		Size: Discrete 1280x720
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x180
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 352x288
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 424x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 848x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 960x540
			Interval: Discrete 0.033s (30.000 fps)
	[1]: 'YUYV' (YUYV 4:2:2)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x180
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 352x288
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 424x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 848x480
			Interval: Discrete 0.050s (20.000 fps)
		Size: Discrete 960x540
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.100s (10.000 fps)

$ v4l2-ctl -d /dev/video1 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

$  v4l2-ctl -d /dev/video2 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'GREY' (8-bit Greyscale)
		Size: Discrete 360x360
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)

$ v4l2-ctl -d /dev/video3 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

$ v4l2-ctl -d /dev/media0 --list-formats-ext
Unable to detect what device /dev/media0 is, exiting.

$ v4l2-ctl -d /dev/media1 --list-formats-ext
Unable to detect what device /dev/media1 is, exiting.

```
</details>

**Output of `ls -l /dev/v4l/by-path`**
```
$ ls -l /dev/v4l/by-path
total 0
lrwxrwxrwx. 1 root root 12 Nov 28 17:09 pci-0000:00:14.0-usb-0:6:1.0-video-index0 -> ../../video0
lrwxrwxrwx. 1 root root 12 Nov 28 17:09 pci-0000:00:14.0-usb-0:6:1.0-video-index1 -> ../../video1
lrwxrwxrwx. 1 root root 12 Nov 28 17:09 pci-0000:00:14.0-usb-0:6:1.2-video-index0 -> ../../video2
lrwxrwxrwx. 1 root root 12 Nov 28 17:09 pci-0000:00:14.0-usb-0:6:1.2-video-index1 -> ../../video3
lrwxrwxrwx. 1 root root 12 Nov 28 17:09 pci-0000:00:14.0-usbv2-0:6:1.0-video-index0 -> ../../video0
lrwxrwxrwx. 1 root root 12 Nov 28 17:09 pci-0000:00:14.0-usbv2-0:6:1.0-video-index1 -> ../../video1
lrwxrwxrwx. 1 root root 12 Nov 28 17:09 pci-0000:00:14.0-usbv2-0:6:1.2-video-index0 -> ../../video2
lrwxrwxrwx. 1 root root 12 Nov 28 17:09 pci-0000:00:14.0-usbv2-0:6:1.2-video-index1 -> ../../video3

```

**Give more information if you have**

**Additional info**
 - Computer (or camera) model: Razer Blade Stealth 13 (Late 2020)
 - Linux distribution: Fedora 39
 - Version of linux-enable-ir-emitter: 5.2.4

EmixamPP, 2023-12-04:
I'll try to solve this problem as soon as possible

EmixamPP, 2023-12-06:
Valgrind reports nothing, so honestly, I don't know what's causing this segfault

EmixamPP, 2023-12-25:
Hey, please try the v6 https://github.com/EmixamPP/linux-enable-ir-emitter/pull/154

klinkigt, 2024-01-14:
Hi,

first of all thank you for working on this. My setup is a ThinkPad X1 Yoga gen 8 with has a intel ipu6 system. It seems that the camera is not "connected" via USB but rather directly PCI.
```
v4l2-ctl --list-devices
```
```
ipu6 (PCI:pci:pci0000:00):
        /dev/video1
        /dev/video2
        /dev/video3
        /dev/video4
        /dev/video5
        /dev/video6
        /dev/video7
        /dev/video8

ipu6 (pci:pci0000:00):
        /dev/media0

Intel MIPI Camera (platform:v4l2loopback-000):
        /dev/video0
```

/dev/video0 is the main camera and now properly working via a v4l2loopback.

```
v4l2-ctl -d /dev/video1 --list-formats-ext
```
```
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture Multiplanar

        [0]: 'Y10 ' (10-bit Greyscale)
        [1]: 'BX24' (32-bit XRGB 8-8-8-8)
        [2]: 'Y12I' (Interleaved 12-bit Greyscale)
        [3]: 'Z16 ' (16-bit Depth)
        [4]: 'YUYV' (YUYV 4:2:2)
        [5]: 'Y8I ' (Interleaved 8-bit Greyscale)
        [6]: 'BG12' (12-bit Bayer BGBG/GRGR)
        [7]: 'GB12' (12-bit Bayer GBGB/RGRG)
        [8]: 'BA12' (12-bit Bayer GRGR/BGBG)
        [9]: 'RG12' (12-bit Bayer RGRG/GBGB)
        [10]: 'BG10' (10-bit Bayer BGBG/GRGR)
        [11]: 'GB10' (10-bit Bayer GBGB/RGRG)
        [12]: 'BA10' (10-bit Bayer GRGR/BGBG)
        [13]: 'RG10' (10-bit Bayer RGRG/GBGB)
        [14]: 'BA81' (8-bit Bayer BGBG/GRGR)
        [15]: 'GBRG' (8-bit Bayer GBGB/RGRG)
        [16]: 'GRBG' (8-bit Bayer GRGR/BGBG)
        [17]: 'RGGB' (8-bit Bayer RGRG/GBGB)
        [18]: 'GREY' (8-bit Greyscale)
```

and
```
ls -l /dev/v4l/by-path
```
```
lrwxrwxrwx 1 root root 12 Jan 14 11:45 pci-0000:00:05.0-video-index0 -> ../../video7
lrwxrwxrwx 1 root root 12 Jan 14 11:45 pci-0000:00:05.0-video-index1 -> ../../video1
lrwxrwxrwx 1 root root 12 Jan 14 11:45 pci-0000:00:05.0-video-index2 -> ../../video2
lrwxrwxrwx 1 root root 12 Jan 14 11:45 pci-0000:00:05.0-video-index3 -> ../../video3
lrwxrwxrwx 1 root root 12 Jan 14 11:45 pci-0000:00:05.0-video-index4 -> ../../video4
lrwxrwxrwx 1 root root 12 Jan 14 11:45 pci-0000:00:05.0-video-index5 -> ../../video5
lrwxrwxrwx 1 root root 12 Jan 14 11:45 pci-0000:00:05.0-video-index6 -> ../../video6
lrwxrwxrwx 1 root root 12 Jan 14 11:45 pci-0000:00:05.0-video-index7 -> ../../video8
```

I receive the same segmentation fault as the original author. I tried almost all versions v5 and v6 either the pre-compiled or self compiled. As the output is not very, is there a way for a more verbose version? It looks to me that the segmentation fault happens early on.

Thank you

EmixamPP, 2024-01-17:
Did you try to compile yourself the project?

What is your CPU reference? I'm wondering if this is not related to the SIMD extensions (https://en.wikipedia.org/wiki/Streaming_SIMD_Extensions) that opencv use.

klinkigt, 2024-01-17:
> Concerning the verbosity, there is the `-v` option, but indeed I should output more debug information. I will work on that.

This I was already using. As said, maybe it needs some print line before even something is happening :)

klinkigt, 2024-01-19:
Thank you. I recompiled and tried the new version and receive:
```
WARNING: The camera /dev/v4l/by-path/pci-0000:00:05.0-video-index1 is not in gray scale. This is probably your regular camera.
Press any key in the window to close
terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.8.1) /home/abuild/rpmbuild/BUILD/opencv-4.8.1/modules/highgui/src/window.cpp:971: error: (-215:Assertion failed) size.width>0 && size.height>0 in function 'imshow'
```
resulting in a segmentation fault again.

As for the error message. Not sure, maybe newer systems provide an RGB image instead of grayscale?

EmixamPP, 2024-01-19:
>   what():  OpenCV(4.8.1) /home/abuild/rpmbuild/BUILD/opencv-4.8.1/modules/highgui/src/window.cpp:971: error: (-215:Assertion failed) size.width>0 && size.height>0 in function 'imshow'

OpenCV fails to determine automatically the resolution to use, you have to specify it using the options `--height` and  `--width`.

> resulting in a segmentation fault again.

After the exception you showed, there is a segfault message printed?

>  Not sure, maybe newer systems provide an RGB image instead of grayscale?

Correct. We can see the in `v4l2-ctl -d /dev/video1 --list-formats-ext` that the last mode says that the camera can be in grayscale, but I don't know I have to do with OpenCV to select it since this is a pci camera. Usually with usb one, this is the resolution that drives this choice.

I'm not 100% sure that OpenCV supports pci camera for VideoCapture? Or maybe I need to do things differently? I will investigate in the coming days, but if you find something, please feel free to share it with me.

klinkigt, 2024-01-20:
Hi,
yes, after the exception there is still a segmentation fault. Sorry for not pasting this part, as it was in German. I also tried with setting the image size but no success here either. 
I think the direction that PCI devices are the issue might be correct. I had a lot of trouble making the main camera work with said loopback device. But as this is working now, I guess the drivers are correct.

EmixamPP, 2024-01-22:
I've pushed modifications that will explicitly select the grey mode of a camera if multiple possibility exists (like in your case). But this will not fix the height and width problem I think.

klinkigt, 2024-01-26:
Thank you for update. First I was thinking I had a problem with compiling, but the error is completely the same (not in gray scale)
It seems like that opencv just cannot work with this cameras at all. Maybe it will have to make a similar trick as with my main camera, looping it back with v4l2loopback. But is somehow strange that those cameras are listed under /dev/ but not useable. Something seems to be missing. I will try ti dig deeper in hope to find something.

richardbowman, 2024-02-01:
Hi, just trying this tool out in order to use Howdy. I'm on a Lenovo ThinkPad T16 Gen 1 (AMD Ryzen™ 7 PRO 6850U with Radeon™ Graphics × 16) running Fedora 39. On the latest stable version 5.2.6, my IR camera flashed for a while and a prompt came up to confirm if I saw flashing (which i said no to) before I received a segmentation fault. I followed this thread and installed the v6 version. It does not get very far at all before seg faulting. I also get no /etc/linux-enable-ir-emitter file.

<details>
<summary>Output of v4l2-ctl --list-devices</summary>

```
Integrated RGB Camera: Integrat (usb-0000:05:00.0-1):
	/dev/video0
	/dev/video1
	/dev/video2
	/dev/video3
	/dev/media0
	/dev/media1
```

</details>

<details>
<summary>Output of v4l2-ctl -d /dev/videoX --list-formats-ext</summary>

```
# /dev/video0
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'MJPG' (Motion-JPEG, compressed)
		Size: Discrete 1920x1080
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 960x540
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 848x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 424x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 352x288
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x180
			Interval: Discrete 0.033s (30.000 fps)
	[1]: 'YUYV' (YUYV 4:2:2)
		Size: Discrete 1920x1080
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 960x540
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 848x480
			Interval: Discrete 0.050s (20.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 424x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 352x288
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x180
			Interval: Discrete 0.033s (30.000 fps)

# /dev/video1
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

# /dev/video2
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'GREY' (8-bit Greyscale)
		Size: Discrete 640x360
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.033s (30.000 fps)

# /dev/video3
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

```
</details>

<details>
<summary>Output of ls -l /dev/v4l/by-path</summary>

```
total 0
lrwxrwxrwx. 1 root root 12 Jan 31 21:58 pci-0000:05:00.0-usb-0:1:1.0-video-index0 -> ../../video0
lrwxrwxrwx. 1 root root 12 Jan 31 21:58 pci-0000:05:00.0-usb-0:1:1.0-video-index1 -> ../../video1
lrwxrwxrwx. 1 root root 12 Jan 31 21:58 pci-0000:05:00.0-usb-0:1:1.2-video-index0 -> ../../video2
lrwxrwxrwx. 1 root root 12 Jan 31 21:58 pci-0000:05:00.0-usb-0:1:1.2-video-index1 -> ../../video3
lrwxrwxrwx. 1 root root 12 Jan 31 21:58 pci-0000:05:00.0-usbv2-0:1:1.0-video-index0 -> ../../video0
lrwxrwxrwx. 1 root root 12 Jan 31 21:58 pci-0000:05:00.0-usbv2-0:1:1.0-video-index1 -> ../../video1
lrwxrwxrwx. 1 root root 12 Jan 31 21:58 pci-0000:05:00.0-usbv2-0:1:1.2-video-index0 -> ../../video2
lrwxrwxrwx. 1 root root 12 Jan 31 21:58 pci-0000:05:00.0-usbv2-0:1:1.2-video-index1 -> ../../video3
```

</details>

EmixamPP, 2024-02-01:
I've just pushed multiple improvements, and one specific for the imshow exception.
Please use the version provided in the PR #154 (see the bot comment).

If it stills not working, could you try three things:
* Add `-g` to the configure command
* Specify the width and height of your camera using `--width` and `--height`
* Both in the same time

richardbowman, 2024-02-04:
So the latest version does not segfault on my machine, but it does hang forever (never exits). I can see it has left the main camera on (white light on my laptop is on), but the IR camera isn't on (can't see the red light). I tried -g but it says that's not a valid argument. I also tried --width and --height which it said were not valid. the help suggested -w and -t were width and height so I did try passing -w 640 -t 360 for one run, which didn't help (not sure which resolution to pass? that's the only resolution reported by the IR camera in the v4l output).

```
x@thinkpad-t16-gen-1:~$ sudo linux-enable-ir-emitter -g configure
usage: linux-enable-ir-emitter [-h] [-V] [-v] [-d device] [-w width]
                               [-t height]
                               {run,configure,tweak,test,boot,delete} ...
linux-enable-ir-emitter: error: unrecognized arguments: -g

x@thinkpad-t16-gen-1:~$ sudo linux-enable-ir-emitter -V
linux-enable-ir-emitter 6.0.0
Developped by Maxime Dirksen - EmixamPP
MIT License

```

EmixamPP, 2024-02-05:
If the segfault problem is resolved, please let's open a new issue using the configuration has failed template.
Like this, I can give proper help to every of you.

Note that the tool will not hang forever, if you enable verbosity (`-v`, as requested in the issue template) you will see outputs

But first, could you reinstall again the tool? I've fixed an issue with the boot service.

richardbowman, 2024-02-08:
Hmm I take it back, the segfault is intermittent. I've tried the latest build. It either hangs forever (well, I've let it sit for 5-10min), or finally segmentation faults. I am running with -v and -g:

```
$ sudo linux-enable-ir-emitter -v -w 640 -t 360 configure -g
DEBUG: Executing configure command.
INFO: Stand in front of and close to the camera and make sure the room is well lit.
INFO: Ensure to not use the camera during the execution.
DEBUG: Checking if /dev/video0 is a greyscale camera.
DEBUG: Checking if /dev/video1 is a greyscale camera.
DEBUG: Checking if /dev/video2 is a greyscale camera.
DEBUG: /dev/video2 is a greyscale camera.
INFO: Configuring the camera /dev/video2
DEBUG: yaml error: bad file: /etc/linux-enable-ir-emitter/pci-0000:05:00.0-usb-0:1:1.2-video-index0
DEBUG: No previous configuration found.
Is the ir emitter flashing (not just turn on) ? Yes/No ? No
DEBUG: no  inputed.
DEBUG: Instruction applied: unit: 4, selector: 2, control: 1 0
DEBUG: Instruction applied: unit: 4, selector: 2, control: 2 0
DEBUG: Instruction applied: unit: 4, selector: 2, control: 3 0
DEBUG: Instruction applied: unit: 4, selector: 2, control: 4 0
DEBUG: Instruction applied: unit: 4, selector: 2, control: 5 0
DEBUG: Instruction applied: unit: 4, selector: 2, control: 6 0
DEBUG: Instruction applied: unit: 4, selector: 2, control: 7 0
DEBUG: Instruction applied: unit: 4, selector: 2, control: 8 0
DEBUG: Instruction applied: unit: 4, selector: 2, control: 9 0
DEBUG: Instruction applied: unit: 4, selector: 2, control: 255 255
DEBUG: Instruction applied: unit: 4, selector: 3, control: 1
Segmentation fault
$ 
```

azblurbit, 2024-02-09:
Apologies for taking so long to get back to you on this ticket. Got sidetracked with the holidays. I really appreciate you helping with this. This is a great project.

I don't appear to be getting the segmentation fault with the latest v6 so that's good. It's now just hanging. It runs and I start to see the IR camera light up after video2 (I think, it happens quick). However it comes on. I did let it run for a few minutes and then halted it because it doesn't seem to stop.

> $sudo linux-enable-ir-emitter -v configure -g
> DEBUG: Executing configure command.
> INFO: Stand in front of and close to the camera and make sure the room is well lit.
> INFO: Ensure to not use the camera during the execution.
> DEBUG: Checking if /dev/video0 is a greyscale camera.
> DEBUG: Checking if /dev/video1 is a greyscale camera.
> DEBUG: Checking if /dev/video2 is a greyscale camera.

I did also try running it with -g and -m but that didn't seem to change anything.

If you'd like I can open a new ticket for the configuration not working since it seems for me the segmentation fault at least is resolved.

EmixamPP, 2024-02-10:
I've published new commits with again better logging for me to identify exactly where the segfault happens, and why the process seems stuck.

Could you please both try again with this new version? 
I hope after that I will be able to provide an answer to your issues!

azblurbit, 2024-02-14:
I seemed to have gotten the same sort of issue. Are there logs being written to anywhere now in the new version that I can grab?

Also I'm sort of new to github so I did download the right tarball, correct?
[image]

Assuming I did I got the same sort of error with Segmentation fault:

> $sudo linux-enable-ir-emitter -v configure
> DEBUG: Executing configure command.
> INFO: Stand in front of and close to the camera and make sure the room is well lit.
> INFO: Ensure to not use the camera during the execution.
> DEBUG: Checking if /dev/video0 is a greyscale camera.
> DEBUG: Checking if /dev/video1 is a greyscale camera.
> DEBUG: Checking if /dev/video2 is a greyscale camera.
> Segmentation fault

forcemagic, 2024-02-18:
Hi everyone! I seem to have a similar issue, running `linux-enable-ir-emitter v5.2.4-1 (reported by -V flag: v5.2.3)` from AUR.

```
➜  ~ sudo -E linux-enable-ir-emitter -v configure
INFO: Stand in front of and close to the camera and make sure the room is well lit.
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
/usr/include/c++/13.2.1/bits/stl_vector.h:1125: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = std::unique_ptr<std::vector<int> >; _Alloc = std::allocator<std::unique_ptr<std::vector<int> > >; reference = std::unique_ptr<std::vector<int> >&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
[1]    9510 IOT instruction  sudo -E linux-enable-ir-emitter -v configure
```

The same thing happens when recompiling project from master. I am running on `Yoga Pro 7 14ARP8 (s/n: YX06SB5Y)` with Camera `5986:215d`.

Happy to help in debugging anything that's related to this issue:) Thank you for the amazing piece of software!

EmixamPP, 2024-02-21:
Sorry for the late reply @azblurbit, I forgot to tell you to uninstall the v5 version before, please consult the [README.md](https://github.com/EmixamPP/linux-enable-ir-emitter/blob/master/README.md#uninstallation)

EmixamPP, 2024-03-17:
It has been merged on master. You can directly download this v6 pre version here: https://github.com/EmixamPP/linux-enable-ir-emitter/releases/tag/6.0.0-pre

hulilikii, 2024-05-06:
I have been encountering a failure at the same place using v6 beta on my thinkpad z13 with IMC integrated IR
`DEBUG: Instruction applied: unit: 4, selector: 3, control: 1
/usr/include/c++/11/bits/stl_vector.h:1063: std::vector<_Tp, _Alloc>::const_reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) const [with _Tp = int; _Alloc = std::allocator<int>; std::vector<_Tp, _Alloc>::const_reference = const int&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__n < this->size()' failed.
Aborted`
this is where it hangs and eventually aborts requiring a reboot.
Any ways to skip this instruction?

EmixamPP, 2024-05-06:
> Any ways to skip this instruction?

Yes you can! Edit the yaml configuration file that is in `/etc/linux-enable-ir-emitter`, find the item with the fields `unit: 4, selector: 3, control: 1`, then change the field `corrupted: false` to `true`

EmixamPP, 2024-05-12:
@hulilikii see #172, use the `--manual` option

hulilikii, 2024-05-12:
Thanks for the help. I was successful using the config file and setting any instructions that crashed the camera as true. Seems to work fine now after configuring.

EmixamPP, 2024-05-14:
I might be that both the assertion error and the segfault are related.
If so, in the PR #176 you can find a bot comment with a link to a tarball and installation instructions that should fix the issue.

EmixamPP, 2024-06-14:
The v6 has been published. Feel free to open a new issue if the problem persists.

closed 2024-06-14 by EmixamPP