EmixamPP/linux-enable-ir-emitter #161: X1 Yoga 3rd gen IR LEDs work in GNOME Snapshot but configure gives error "Impossible to find an infrared camera"

ctrl-alt-breathe, 2024-01-09:
**Ouput of `linux-enable-ir-emitter -v configure`**
<details><summary>configure command output</summary>

```
INFO: Stand in front of and close to the camera and make sure the room is well lit.
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
CRITICAL: Impossible to find an infrared camera.
```
</details>

**Output of `v4l2-ctl --list-devices`**
```
Integrated IR Camera: Integrate (usb-0000:00:14.0-5):
	/dev/video0
	/dev/video1
	/dev/media0

Integrated Camera: Integrated C (usb-0000:00:14.0-8):
	/dev/video2
	/dev/video3
	/dev/media1
```

**Ouput of `v4l2-ctl -d /dev/videoX --list-formats-ext`**

<details><summary>v4l2 device output</summary>

```
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'YUYV' (YUYV 4:2:2)
		Size: Discrete 340x340
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
```
</details>

**Output of `ls -l /dev/v4l/by-path`**
```
total 0
lrwxrwxrwx 1 root root 12 Jan  8 23:38 pci-0000:00:14.0-usb-0:5:1.0-video-index0 -> ../../video0
lrwxrwxrwx 1 root root 12 Jan  8 23:38 pci-0000:00:14.0-usb-0:5:1.0-video-index1 -> ../../video1
lrwxrwxrwx 1 root root 12 Jan  8 23:38 pci-0000:00:14.0-usb-0:8:1.0-video-index0 -> ../../video2
lrwxrwxrwx 1 root root 12 Jan  8 23:38 pci-0000:00:14.0-usb-0:8:1.0-video-index1 -> ../../video3
lrwxrwxrwx 1 root root 12 Jan  8 23:38 pci-0000:00:14.0-usbv2-0:5:1.0-video-index0 -> ../../video0
lrwxrwxrwx 1 root root 12 Jan  8 23:38 pci-0000:00:14.0-usbv2-0:5:1.0-video-index1 -> ../../video1
lrwxrwxrwx 1 root root 12 Jan  8 23:38 pci-0000:00:14.0-usbv2-0:8:1.0-video-index0 -> ../../video2
lrwxrwxrwx 1 root root 12 Jan  8 23:38 pci-0000:00:14.0-usbv2-0:8:1.0-video-index1 -> ../../video3
```

**Give more information if you have**
The camera and IR LEDs work in GNOME Snapshot and flicker when I run the configure command, but it still returns "Impossible to find an infrared camera". It also returns "Unable to obtain the /dev/videoX path" when I specify /dev/video0 as the camera

**Additional info**
 - Computer (or camera) model: ThinkPad X1 Yoga 3rd gen
 - Linux distribution: ArcoLinux
 - Version of linux-enable-ir-emitter: 5.2.4

EmixamPP, 2024-01-09:
If your IR emitters are working, for what purpose you want to use this tool?

>It also returns "Unable to obtain the /dev/videoX path" when I specify /dev/video0 as the camera

Please try the v6 #154, it fixes this error; there is a bot comment in the PR that explains how to install it.

ctrl-alt-breathe, 2024-01-09:
Because howdy only blinks the LEDs once then says no face found

ctrl-alt-breathe, 2024-01-09:
I followed the uninstallation steps then installed that v6 version, but it still gives the videoX error

EmixamPP, 2024-01-09:
Strange, I had the same problem as you because of usbv2 v4l paths, and the fix I added in the v6 solved the problem on my setup.
Could you show me the output of:
```
linux-enable-ir-emitter test
linux-enable-ir-emitter -d /dev/video0 test
linux-enable-ir-emitter -d /dev/video2 test
```
Thanks!

EmixamPP, 2024-01-13:
This seems very weird. 
I've updated the error message to get more information. Could you go in the PR and install the tool again? (no need to uninstall, you can overwrite)
Then, could you copy past the outputs of all these commands:
```
find -L /dev/v4l/by-path -samefile /dev/video0
linux-enable-ir-emitter -d  /dev/video0 test
linux-enable-ir-emitter -d  /dev/v4l/by-path/pci-0000:00:14.0-usb-0:5:1.0-video-index0 test
linux-enable-ir-emitter test
```
Please do not resume to what is shown, but really copy past to avoid confusion.

EmixamPP, 2024-02-02:
You can try again if you want to, I've refactored the system of devices discovering.

EmixamPP, 2024-03-04:
Closing because of inactivity. Feel free to comment if you want to re-open

closed 2024-03-04 by EmixamPP