EmixamPP/linux-enable-ir-emitter #168: IR emitter fails after booting into windows, Systemd service always fails

WebF1yin, 2024-03-07:
**Output of `systemctl status linux-enable-ir-emitter.service` immediately after boot**
```
linux-enable-ir-emitter.service - enable the infrared emitter
     Loaded: loaded (/etc/systemd/system/linux-enable-ir-emitter.service; enabled; preset: disabled)
     Active: failed (Result: exit-code) since Thu 2024-03-07 13:13:34 MST; 35s ago
    Process: 661 ExecStartPre=modprobe uvcvideo (code=exited, status=0/SUCCESS)
    Process: 687 ExecStart=linux-enable-ir-emitter -v run (code=exited, status=1/FAILURE)
   Main PID: 687 (code=exited, status=1/FAILURE)
        CPU: 93ms

 systemd[1]: Starting enable the infrared emitter...
linux-enable-ir-emitter[687]: DEBUG: Executing run command.
linux-enable-ir-emitter[687]: DEBUG: filesystem error: cannot make canonical path: No such file or directory [/dev/v4l/by-path/pci-0000:04:00.4-usb-0:4:1.2-video-index0]
linux-enable-ir-emitter[687]: ERROR: Impossible to obtain the v4l name of /dev/v4l/by-path/pci-0000:04:00.4-usb-0:4:1.2-video-index0
linux-enable-ir-emitter[687]: WARNING: Failed to load a configuration for /dev/v4l/by-path/pci-0000:04:00.4-usb-0:4:1.2-video-index0
systemd[1]: linux-enable-ir-emitter.service: Main process exited, code=exited, status=1/FAILURE
systemd[1]: linux-enable-ir-emitter.service: Failed with result 'exit-code'.
systemd[1]: Failed to start enable the infrared emitter.

```

**Output of `linux-enable-ir-emitter boot status` after running `systemctl start linux-enable-ir-emitter.service`**
```
linux-enable-ir-emitter.service - enable the infrared emitter
     Loaded: loaded (/etc/systemd/system/linux-enable-ir-emitter.service; enabled; preset: disabled)
     Active: failed (Result: exit-code) since Thu 2024-03-07 13:17:01 MST; 5s ago
    Process: 3466 ExecStartPre=modprobe uvcvideo (code=exited, status=0/SUCCESS)
    Process: 3467 ExecStart=linux-enable-ir-emitter -v run (code=exited, status=1/FAILURE)
   Main PID: 3467 (code=exited, status=1/FAILURE)
        CPU: 50ms

systemd[1]: Starting enable the infrared emitter...
linux-enable-ir-emitter[3467]: DEBUG: Executing run command.
linux-enable-ir-emitter[3467]: DEBUG: yaml error: bad file: /etc/linux-enable-ir-emitter/pci-0000:04:00.4-usbv2-0:4:1.2-video-index0
linux-enable-ir-emitter[3467]: WARNING: Failed to load a configuration for /dev/v4l/by-path/pci-0000:04:00.4-usb-0:4:1.2-video-index0
systemd[1]: linux-enable-ir-emitter.service: Main process exited, code=exited, status=1/FAILURE
systemd[1]: linux-enable-ir-emitter.service: Failed with result 'exit-code'.
systemd[1]: Failed to start enable the infrared emitter.

```

**Output of `cat /etc/linux-enable-ir-emitter/*`**
```
 cat: /etc/linux-enable-ir-emitter/: Is a directory
```

**Output of `v4l2-ctl --list-devices*`**
```
 Integrated Camera: Integrated C (usb-0000:04:00.4-4):
        /dev/video0
        /dev/video1
        /dev/video2
        /dev/video3
        /dev/media0
        /dev/media1
```

**Output of `ls -l /dev/v4l/by-id/*`**
```
lrwxrwxrwx 1 root root 12 Mar  7 13:14 usb-Azurewave_Integrated_Camera_200901010001-video-index0 -> ../../video0
lrwxrwxrwx 1 root root 12 Mar  7 13:14 usb-Azurewave_Integrated_Camera_200901010001-video-index1 -> ../../video3
```

**Output of `ls -l  /dev/v4l/by-path/*`**
```
total 0
lrwxrwxrwx 1 root root 12 Mar  7 13:13 pci-0000:04:00.4-usb-0:4:1.0-video-index0 -> ../../video0
lrwxrwxrwx 1 root root 12 Mar  7 13:13 pci-0000:04:00.4-usb-0:4:1.0-video-index1 -> ../../video1
lrwxrwxrwx 1 root root 12 Mar  7 13:13 pci-0000:04:00.4-usb-0:4:1.2-video-index0 -> ../../video2
lrwxrwxrwx 1 root root 12 Mar  7 13:13 pci-0000:04:00.4-usb-0:4:1.2-video-index1 -> ../../video3
lrwxrwxrwx 1 root root 12 Mar  7 13:13 pci-0000:04:00.4-usbv2-0:4:1.0-video-index0 -> ../../video0
lrwxrwxrwx 1 root root 12 Mar  7 13:13 pci-0000:04:00.4-usbv2-0:4:1.0-video-index1 -> ../../video1
lrwxrwxrwx 1 root root 12 Mar  7 13:13 pci-0000:04:00.4-usbv2-0:4:1.2-video-index0 -> ../../video2
lrwxrwxrwx 1 root root 12 Mar  7 13:13 pci-0000:04:00.4-usbv2-0:4:1.2-video-index1 -> ../../video3
```

**Give more information if you have**

I'm having a weird issue that may not be directly linked to linux-enable-ir-emitter. I am dual booting windows and Arch, and after booting into windows and using facial recognition there, the next time I boot Arch, the IR emitter doesn't work. The service always fails when booting Arch whether I'm coming from windows or not. If I manually run linux-enable-ir-emitter, although the service still fails, the IR emitter works until I boot into windows again.

Let me know if you need any other information,
Thanks!

**Additional info**
 - Computer (or camera) model: Razer-blade 14 2021
 - Linux distribution: Arch, linux-zen
 - Version of linux-enable-ir-emitter: 6.0.0

WebF1yin, 2024-03-08:
An update, if it matters: holding the power button to force the computer to shut down, from what seems like any screen in either OS, fixes the IR emitter in Arch as well.

EmixamPP, 2024-03-08:
> Output of systemctl status linux-enable-ir-emitter.service immediately after boot

Hum, it seems that the v4l paths are not yet setup

> Output of linux-enable-ir-emitter boot status after running systemctl start linux-enable-ir-emitter.service

And here, this is a problem to deserialize the configuration file.

You forgot the `*` here: 
>  cat: /etc/linux-enable-ir-emitter/: Is a directory

Also, could you provide the output of `ls /etc/linux-enable-ir-emitter`?

WebF1yin, 2024-03-08:
<details>
  <summary>
    Oops I did forget the *, sorry:
  </summary>

```

- corrupted: false
  unit: 9
  selector: 10
  current:
    - 0
    - 8
    - 0
    - 187
    - 187
    - 187
    - 187
    - 187
  maximum:
    - 255
    - 255
    - 255
    - 255
    - 255
    - 255
    - 255
    - 255
  minimum:
    - 0
    - 0
    - 0
    - 0
    - 0
    - 0
    - 0
    - 0
- corrupted: false
  unit: 10
  selector: 2
  current:
    - 1
    - 0
    - 0
    - 0
    - 0
    - 0
    - 0
    - 17
    - 39
    - 0
    - 0
    - 0
    - 0
    - 0
    - 0
  maximum:
    - 7
    - 0
    - 0
    - 0
    - 0
    - 0
    - 0
    - 128
    - 150
    - 152
    - 0
    - 0
    - 0
    - 0
    - 0
  minimum:
    - 0
    - 0
    - 0
    - 0
    - 0
    - 0
    - 0
    - 16
    - 39
    - 0
    - 0
    - 0
    - 0
    - 0
    - 0
- corrupted: false
  unit: 10
  selector: 3
  current:
    - 16
    - 0
    - 0
    - 0
    - 0
    - 0
    - 0
    - 100
    - 0
    - 0
    - 0
  maximum:
    - 16
    - 0
    - 0
    - 0
    - 0
    - 0
    - 0
    - 14
    - 6
    - 0
    - 0
  minimum:
    - 16
    - 0
    - 0
    - 0
    - 0
    - 0
    - 0
    - 100
    - 0
    - 0
    - 0
- corrupted: false
  unit: 10
  selector: 6
  current:
    - 1
    - 3
    - 2
    - 0
    - 0
    - 0
    - 0
    - 0
    - 0
  maximum:
    - 1
    - 3
    - 3
    - 0
    - 0
    - 0
    - 0
    - 0
    - 0
- corrupted: false
  unit: 10
  selector: 9
  current:
    - 0
    - 0
    - 0
    - 0
  maximum:
    - 1
    - 0
    - 0
    - 0
  minimum:
    - 0
    - 0
    - 0
    - 0
- corrupted: false
  unit: 14
  selector: 14
  current:
    - 0
    - 0
  maximum:
    - 2
    - 100
  minimum:
    - 0
    - 0

```

</details>

Here's the output of `ls /etc/linux-enable-ir-emitter`:
```
pci-0000:04:00.4-usb-0:4:1.2-video-index0
```

EmixamPP, 2024-03-10:
I've pushed a possible fix to the error, please try #169. There is a bot comment in the PR that provides the tarball.

This should solve the second problem; the one not immediately after boot.
Simply execute `linux-enable-ir-emitter -v run` to see if it succeeds to run the configuration saved (i.e. this is the command executed by `systemctl start linux-enable-ir-emitter`.

WebF1yin, 2024-03-22:
Sorry I left you hanging for so long. This fix seems to work! It still fails to start the service on boot, but ```linux-enable-ir-emitter -v run``` or ```systemctl start linux-enable-ir-emitter``` makes the emitter work again. If the latter command is used, ```systemctl status``` reports the service ran fine.

EmixamPP, 2024-03-23:
Great. For this problem, could you try if works for you by adding some delay in the /etc/systemd/system/linux-enable-ir-emitter.service file: 

```
...
[Service]
ExecStartPre=/bin/sleep 2
...
```

EmixamPP, 2024-03-27:
Oops, the issue was closed by mistake

WebF1yin, 2024-03-30:
Adding the delay worked! The service seems like it starts properly every time, and the IR emitter works at the login screen. Thanks so much!

EmixamPP, 2024-03-31:
Could you try if 1sec is sufficient? I'm thinking to integrate this delay for everyone

closed 2024-04-11 by EmixamPP