EmixamPP/linux-enable-ir-emitter #170: Pre-Release Missing Folder (/etc/linux-enable-ir-emitter)

ironveil, 2024-03-24:
**Describe the bug**
Using the latest pre-release (6.0.0-pre), it seems like it's pointing to a configuration directory that doesn't exist. On comparing the 6.0.0-pre & 5.2.4 .tar.gz files, it appears that the pre-release doesn't have the `/etc/linux-enable-ir-emitter` folder to store config files. If this folder is created, everything works fine.

**How to reproduce**
1. Install v6.0.0-pre
2. Run any command that requires a config (e.g. `sudo linux-enable-ir-emitter configure` or `sudo linux-enable-ir-emitter status`)
3. Output:
```
Traceback (most recent call last):
  File "/usr/local/bin/linux-enable-ir-emitter", line 164, in <module>
    res = boot(args.boot_status)
          ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/linux-enable-ir-emitter/command/boot.py", line 29, in boot
    devices = get_devices()
              ^^^^^^^^^^^^^
  File "/usr/local/lib/linux-enable-ir-emitter/utils/globals.py", line 54, in get_devices
    for config in os.listdir(SAVE_FOLDER_CONFIG_PATH):
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: '/etc/linux-enable-ir-emitter'
```

**Additional info**
 - Computer (or camera) model: Lenovo Yoga Slim7
 - Linux distribution: Arch Linux
 - Version of linux-enable-ir-emitter:  6.0.0-pre

EmixamPP, 2024-03-24:
Did you already configured the camera? Because to execute the `boot enable` command, you need to first run the `configure` one.

Is there is a directory at `/etc/linux-enable-ir-emitter`? Could you create it?

Thank you for your help!

SWi7CHbladE, 2024-03-24:
You may manually create the folder and then go ahead with the installation

EmixamPP, 2024-03-27:
I've pushed the fix on the master branch and updated the pre-release tarball. Thank you for your help!

closed 2024-03-27 by EmixamPP