EmixamPP/linux-enable-ir-emitter #172: Assertion failed while configuring

fervidnerd, 2024-04-02:
**Ouput of `linux-enable-ir-emitter -v configure`**

```
DEBUG: Executing configure command.
INFO: Stand in front of and close to the camera and make sure the room is well lit.
INFO: Ensure to not use the camera during the execution.
DEBUG: Checking if /dev/video3 is a greyscale camera.
DEBUG: Checking if /dev/video2 is a greyscale camera.
/usr/include/c++/11/bits/stl_vector.h:1063: std::vector<_Tp, _Alloc>::const_reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) const [with _Tp = int; _Alloc = std::allocator<int>; std::vector<_Tp, _Alloc>::const_reference = const int&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__n < this->size()' failed.
[1]    112803 IOT instruction  sudo linux-enable-ir-emitter -v configure

```

**Output of `cat /etc/linux-enable-ir-emitter/*`**
```
- corrupted: false
  unit: 4
  selector: 6
  current:
    - 1
    - 3
    - 1
    - 0
    - 0
    - 0
    - 0
    - 0
    - 0
  maximum:
    - 1
    - 3
    - 3
    - 0
    - 0
    - 0
    - 0
    - 0
    - 0
- corrupted: false
  unit: 4
  selector: 9
  current:
    - 1
    - 0
    - 0
    - 0
  maximum:
    - 1
    - 0
    - 0
    - 0
  minimum:
    - 0
    - 0
    - 0
    - 0

```

**Output of `v4l2-ctl --list-devices`**
```
paste here
```

**Ouput of `for dev in /dev/video*; do echo $dev && v4l2-ctl -d $dev --list-formats-ext; done`**
```
paste here
```

**Output of `ls -l /dev/v4l/by-path`**
```
lrwxrwxrwx 1 root root 12 Apr  2 20:13 pci-0000:0a:00.3-usb-0:3.2.1:1.0-video-index0 -> ../../video0
lrwxrwxrwx 1 root root 12 Apr  2 20:13 pci-0000:0a:00.3-usb-0:3.2.1:1.0-video-index1 -> ../../video1
lrwxrwxrwx 1 root root 12 Apr  2 20:13 pci-0000:0a:00.3-usb-0:3.2.1:1.2-video-index0 -> ../../video2
lrwxrwxrwx 1 root root 12 Apr  2 20:13 pci-0000:0a:00.3-usb-0:3.2.1:1.2-video-index1 -> ../../video3

```

**Give more information if you have**

**Additional info**
 - Computer (or camera) model: https://www.nexigo.com/collections/all/products/hellocam
 - Linux distribution: Ubuntu 22.04.4
 - Version of linux-enable-ir-emitter: 6.0.0-pre

fervidnerd, 2024-04-03:
> Could you retry with the command linux-enable-ir-emitter -v configure -d /dev/video2?

Unfortunately I get the same error as above, even with the suggested changes.

```
➜  ~ sudo linux-enable-ir-emitter -v -d /dev/video2 configure
DEBUG: Executing configure command.
INFO: Stand in front of and close to the camera and make sure the room is well lit.
INFO: Ensure to not use the camera during the execution.
/usr/include/c++/11/bits/stl_vector.h:1063: std::vector<_Tp, _Alloc>::const_reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) const [with _Tp = int; _Alloc = std::allocator<int>; std::vector<_Tp, _Alloc>::const_reference = const int&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__n < this->size()' failed.
[1]    8506 IOT instruction  sudo linux-enable-ir-emitter -v -d /dev/video2 configure
```

elegos, 2024-05-11:
Hello there!

Same issue, same cam D:

I was previously prompted by you @EmixamPP to use the very same config file, but I can confirm that the error does not go away:

```bash
❯ sudo linux-enable-ir-emitter -v -d /dev/video2 configure
DEBUG: Executing configure command.
INFO: Stand in front of and close to the camera and make sure the room is well lit.
INFO: Ensure to not use the camera during the execution.
/usr/include/c++/11/bits/stl_vector.h:1063: std::vector<_Tp, _Alloc>::const_reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) const [with _Tp = int; _Alloc = std::allocator<int>; std::vector<_Tp, _Alloc>::const_reference = const int&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__n < this->size()' failed.
[1]    24758 IOT instruction  sudo linux-enable-ir-emitter -v -d /dev/video2 configure
```

The behavior is strange though: when you first try the face recognition, it won't blink and fail (image too dark). When you _try_ to configure the device, even if failing, the next time you try to authenticate, it will blink and work, until the next restart.

EmixamPP, 2024-05-11:
Can someone run `gdb`? Since I cannot reproduce, I need help to find which line raises the error
1. Download and place in `/usr/local/bin` that the file inside the zip :
[linux-enable-ir-emitter.zip](https://github.com/EmixamPP/linux-enable-ir-emitter/files/15284206/linux-enable-ir-emitter.zip)
2. Execute the tool inside gdb : `gdb -ex run --args linux-enable-ir-emitter -v configure`
3. When it fails, write inside gdb: `backtrace`
4. Paste all the outputs in a comment

elegos, 2024-05-11:
Here we go:

```
(gdb) backtrace
#0  0x0000000000454fcb in compute_sum_intensities_variation (diffs=std::vector of length 0, capacity 0) at ../camera/autocamera.cpp:49
#1  0x000000000045507b in AutoCamera::intensity_variation_sum (this=0xa71530) at ../camera/autocamera.cpp:73
#2  0x00000000004551ab in AutoCamera::AutoCamera (this=0xa71530, device="/dev/video2", width=-1, height=-1, capture_time_ms=1000) at ../camera/autocamera.cpp:99
#3  0x000000000048882e in std::_Construct<AutoCamera, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int&, int&> (__p=0xa71530)
    at /usr/include/c++/13/bits/stl_construct.h:119
#4  0x0000000000488347 in std::allocator_traits<std::allocator<void> >::construct<AutoCamera, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int&, int&> (__p=0xa71530)
    at /usr/include/c++/13/bits/alloc_traits.h:661
#5  std::_Sp_counted_ptr_inplace<AutoCamera, std::allocator<void>, (__gnu_cxx::_Lock_policy)2>::_Sp_counted_ptr_inplace<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int&, int&> (this=0xa71520, __a=...) at /usr/include/c++/13/bits/shared_ptr_base.h:604
#6  0x0000000000487cf7 in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count<AutoCamera, std::allocator<void>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int&, int&> (this=0x7fffffffcf98, __p=@0x7fffffffcf90: 0x0, __a=...) at /usr/include/c++/13/bits/shared_ptr_base.h:971
#7  0x000000000048788e in std::__shared_ptr<AutoCamera, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<std::allocator<void>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int&, int&> (this=0x7fffffffcf90, __tag=...) at /usr/include/c++/13/bits/shared_ptr_base.h:1712
#8  0x0000000000487435 in std::shared_ptr<AutoCamera>::shared_ptr<std::allocator<void>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int&, int&>
    (this=0x7fffffffcf90, __tag=...) at /usr/include/c++/13/bits/shared_ptr.h:464
#9  0x0000000000487024 in std::make_shared<AutoCamera, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int&, int&> () at /usr/include/c++/13/bits/shared_ptr.h:1010
#10 0x00000000004868d1 in CreateCamera<AutoCamera> (device=std::optional [no contained value], width=-1, height=-1, no_gui=false) at ../command/commands.hpp:70
#11 0x0000000000485c8e in configure (device=std::optional [no contained value], width=-1, height=-1, manual=false, emitters=1, neg_answer_limit=10, no_gui=false) at ../command/configure.cpp:45
#12 0x0000000000433f57 in main (argc=3, argv=0x7fffffffe418) at /home/maxime/projects/linux-enable-ir-emitter/build/linux-enable-ir-emitter.cpp:111
```

EmixamPP, 2024-05-11:
Ok very good, thank you! I will be able to find a fix.

To avoid that issue, you can add the `--manual` option.

EmixamPP, 2024-05-14:
Otherwise, in the PR #176 you can find a bot comment with a link to a tarball and installation instructions that should fix the issue.

closed 2024-05-16 by EmixamPP