EmixamPP/linux-enable-ir-emitter #173: Immutable distro does not allow root commands after manual build

Woofweasel, 2024-04-07:
**Ouput of `linux-enable-ir-emitter -v configure`**

```
CRITICAL: Please run as root.
```

**Output of `cat /etc/linux-enable-ir-emitter/*`**
```
cat: '/etc/linux-enable-ir-emitter/*': No such file or directory
```

**Output of `ls -l /dev/v4l/by-path`**
```
total 0
lrwxrwxrwx. 1 root root 12 Apr  7 11:37 pci-0000:05:00.4-usb-0:4:1.0-video-index0 -> ../../video0
lrwxrwxrwx. 1 root root 12 Apr  7 11:37 pci-0000:05:00.4-usb-0:4:1.0-video-index1 -> ../../video1
lrwxrwxrwx. 1 root root 12 Apr  7 11:37 pci-0000:05:00.4-usb-0:4:1.2-video-index0 -> ../../video2
lrwxrwxrwx. 1 root root 12 Apr  7 11:37 pci-0000:05:00.4-usb-0:4:1.2-video-index1 -> ../../video3
lrwxrwxrwx. 1 root root 12 Apr  7 11:37 pci-0000:05:00.4-usbv2-0:4:1.0-video-index0 -> ../../video0
lrwxrwxrwx. 1 root root 12 Apr  7 11:37 pci-0000:05:00.4-usbv2-0:4:1.0-video-index1 -> ../../video1
lrwxrwxrwx. 1 root root 12 Apr  7 11:37 pci-0000:05:00.4-usbv2-0:4:1.2-video-index0 -> ../../video2
lrwxrwxrwx. 1 root root 12 Apr  7 11:37 pci-0000:05:00.4-usbv2-0:4:1.2-video-index1 -> ../../video3
```

**Give more information if you have**
I built the package manually using meson by following the instructions on docs/manual-build.md, including changing the installation directory to $HOME/.local because I'm on an immutable system. If I run `linux-enable-ir emitter configure`, the output is `CRITICAL: Please run as root.`. However, if I then run the same command using sudo, I get `sudo: linux-enable-ir-emitter: command not found`. 

**Additional info**
 - Computer (or camera) model: Lenovo Yoga Slim 7 Pro 16ACH6
 - Linux distribution: Fedora Silverblue 39
 - Version of linux-enable-ir-emitter: linux-enable-ir-emitter 6.0.0-pre

EmixamPP, 2024-04-08:
Ok, I see where is the bug. In the meantime I publish a fix, please execute `sudo /home/<your home>/.local/bin/linux-enable-ir emitter configure`

Woofweasel, 2024-04-08:
Many many thanks! your tool has fixed the issues I was having with Howdy, and you've been incredibly friendly and helpful :)

EmixamPP, 2024-04-08:
You're welcome! If you like the project, do not hesitate to star the repository to support me, thank you!

EmixamPP, 2024-04-08:
I think I'll officially provide a tarball for immutable distros, to avoid manual build.
Because, is your build static ? Or did you layered the yaml and opencv libs package?

Woofweasel, 2024-04-08:
I installed all the dependencies using rpm-ostree, including yaml and opencv.

EmixamPP, 2024-04-08:
I'll send a comment here when the installation archive for immutable distro is ready, it'll be easier for the tool not to break when you update the distro, and it'll also reduce your number of layered packages

Woofweasel, 2024-04-08:
I've just noticed a new bug- the emitter is now not working again after having restarted my system, so I will have to reconfigure it. Will this be solved when the new tarball is ready?

EmixamPP, 2024-04-08:
Don't need to re-configure, execute `linux-enable-ir-emitter run`.

I think there is a problem with the systemd script.
Can past the output of `sudo systemctl status linux-enable-ir-emitter`?

Could you also paste here where are installed the different files, thus what is output by the `ninja install -C build` step

Woofweasel, 2024-04-08:
`sudo systemctl status linux-enable-ir-emitter` gives:
```
× linux-enable-ir-emitter.service - enable the infrared emitter
     Loaded: loaded (/etc/systemd/system/linux-enable-ir-emitter.service; enabl>
    Drop-In: /usr/lib/systemd/system/service.d
             └─10-timeout-abort.conf
     Active: failed (Result: exit-code) since Mon 2024-04-08 12:37:19 BST; 58mi>
    Process: 5688 ExecStartPre=modprobe uvcvideo (code=exited, status=0/SUCCESS)
    Process: 5689 ExecStart=linux-enable-ir-emitter -v run (code=exited, status>
   Main PID: 5689 (code=exited, status=203/EXEC)
        CPU: 13ms

Apr 08 12:37:19 Portaweasel systemd[1]: Starting linux-enable-ir-emitter.servic>
Apr 08 12:37:19 Portaweasel (-emitter)[5689]: linux-enable-ir-emitter.service: >
Apr 08 12:37:19 Portaweasel (-emitter)[5689]: linux-enable-ir-emitter.service: >
Apr 08 12:37:19 Portaweasel systemd[1]: linux-enable-ir-emitter.service: Main p>
Apr 08 12:37:19 Portaweasel systemd[1]: linux-enable-ir-emitter.service: Failed>
Apr 08 12:37:19 Portaweasel systemd[1]: Failed to start linux-enable-ir-emitter>
lines 1-16/16 (END)
```

Output of `ninja install -C build`:

```
ninja: Entering directory `build'
[0/1] Installing files.
Installing libcommands.so to /var/home/woofweasel/.local/lib64/linux-enable-ir-emitter
Installing new directory /etc/udev/rules.d
'/etc/udev/rules.d': Unable to set permissions 493: Operation not permitted, ignoring...
Installing new directory /etc/linux-enable-ir-emitter
'/etc/linux-enable-ir-emitter': Unable to set permissions 493: Operation not permitted, ignoring...
Installing /var/home/woofweasel/linux-enable-ir-emitter/build/linux-enable-ir-emitter to /var/home/woofweasel/.local/bin
Installation failed due to insufficient permissions.
Attempt to use /usr/bin/sudo to gain elevated privileges? [y/n] y
Attempting facial authentication
[sudo] password for woofweasel: 
Identified face as woofweasel
Installing libcommands.so to /var/home/woofweasel/.local/lib64/linux-enable-ir-emitter
Installing new directory /etc/udev/rules.d
Installing new directory /etc/linux-enable-ir-emitter
Installing /var/home/woofweasel/linux-enable-ir-emitter/build/linux-enable-ir-emitter to /var/home/woofweasel/.local/bin
Installing /var/home/woofweasel/linux-enable-ir-emitter/build/globals.py to /var/home/woofweasel/.local/lib64/linux-enable-ir-emitter/utils
Installing /var/home/woofweasel/linux-enable-ir-emitter/utils/__init__.py to /var/home/woofweasel/.local/lib64/linux-enable-ir-emitter/utils
Installing /var/home/woofweasel/linux-enable-ir-emitter/command/__init__.py to /var/home/woofweasel/.local/lib64/linux-enable-ir-emitter/command
Installing /var/home/woofweasel/linux-enable-ir-emitter/command/boot.py to /var/home/woofweasel/.local/lib64/linux-enable-ir-emitter/command
Installing /var/home/woofweasel/linux-enable-ir-emitter/command/load_cpp_commands.py to /var/home/woofweasel/.local/lib64/linux-enable-ir-emitter/command
Installing /var/home/woofweasel/linux-enable-ir-emitter/boot_service/__init__.py to /var/home/woofweasel/.local/lib64/linux-enable-ir-emitter/boot_service
Installing /var/home/woofweasel/linux-enable-ir-emitter/boot_service/base_boot_service.py to /var/home/woofweasel/.local/lib64/linux-enable-ir-emitter/boot_service
Installing /var/home/woofweasel/linux-enable-ir-emitter/LICENSE to /var/home/woofweasel/.local/lib64/linux-enable-ir-emitter
Installing /var/home/woofweasel/linux-enable-ir-emitter/boot_service/systemd/__init__.py to /var/home/woofweasel/.local/lib64/linux-enable-ir-emitter/boot_service/systemd
Installing /var/home/woofweasel/linux-enable-ir-emitter/boot_service/systemd/systemd.py to /var/home/woofweasel/.local/lib64/linux-enable-ir-emitter/boot_service/systemd
Installing /var/home/woofweasel/linux-enable-ir-emitter/boot_service/systemd/linux-enable-ir-emitter.service to /etc/systemd/system
Installing /var/home/woofweasel/linux-enable-ir-emitter/autocomplete/linux-enable-ir-emitter.bash to /var/home/woofweasel/.local/share/bash-completion/completions
Installing /var/home/woofweasel/linux-enable-ir-emitter/autocomplete/linux-enable-ir-emitter.zsh to /var/home/woofweasel/.local/share/zsh/site-functions
```

EmixamPP, 2024-04-10:
Could you modify the `/etc/systemd/system/linux-enable-ir-emitter.service` file and change the line:
`ExecStart=linux-enable-ir-emitter -v run` with:
`ExecStart=/var/home/woofweasel/.local/bin/linux-enable-ir-emitter -v run`

Woofweasel, 2024-04-11:
After making that change, it seems that functionally nothing has changed. If I restart my device, the program will work absolutely fine. If I shut down and then start it up, linux-enable-ir-emitter will not be enabled until I run `linux-enable-ir-emitter run`. 

`sudo systemctl status linux-enable-ir-emitter` now gives:
```
× linux-enable-ir-emitter.service - enable the infrared emitter
     Loaded: loaded (/etc/systemd/system/linux-enable-ir-emitter.service; enabled; preset: disabled)
    Drop-In: /usr/lib/systemd/system/service.d
             └─10-timeout-abort.conf
     Active: failed (Result: exit-code) since Thu 2024-04-11 15:46:59 BST; 7min ago
    Process: 1038 ExecStartPre=modprobe uvcvideo (code=exited, status=0/SUCCESS)
    Process: 1053 ExecStart=/var/home/woofweasel/.local/bin/linux-enable-ir-emitter -v run (code=exited, status=1/FAILURE)
   Main PID: 1053 (code=exited, status=1/FAILURE)
        CPU: 74ms

Apr 11 15:46:59 fedora linux-enable-ir-emitter[1053]:   File "/var/home/woofweasel/.local/lib64/linux-enable-ir-emitter/command/load_cpp_commands.py", line 5, in <module>
Apr 11 15:46:59 fedora linux-enable-ir-emitter[1053]:     cpp_commands = ctypes.CDLL(CPP_COMMANDS_LIB_PATH)
Apr 11 15:46:59 fedora linux-enable-ir-emitter[1053]:                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Apr 11 15:46:59 fedora linux-enable-ir-emitter[1053]:   File "/usr/lib64/python3.12/ctypes/__init__.py", line 379, in __init__
Apr 11 15:46:59 fedora linux-enable-ir-emitter[1053]:     self._handle = _dlopen(self._name, mode)
Apr 11 15:46:59 fedora linux-enable-ir-emitter[1053]:                    ^^^^^^^^^^^^^^^^^^^^^^^^^
Apr 11 15:46:59 fedora linux-enable-ir-emitter[1053]: OSError: /var/home/woofweasel/.local/lib64/linux-enable-ir-emitter/libcommands.so: failed to map segment from shared object
Apr 11 15:46:59 fedora systemd[1]: linux-enable-ir-emitter.service: Main process exited, code=exited, status=1/FAILURE
Apr 11 15:46:59 fedora systemd[1]: linux-enable-ir-emitter.service: Failed with result 'exit-code'.
Apr 11 15:46:59 fedora systemd[1]: Failed to start linux-enable-ir-emitter.service - enable the infrared emitter.
lines 1-20
```

EmixamPP, 2024-04-11:
What shows: `ls -l /var/home/woofweasel/.local/lib64/linux-enable-ir-emitter/libcommands.so`?

Woofweasel, 2024-04-11:
`ls -l /var/home/woofweasel/.local/lib64/linux-enable-ir-emitter/libcommands.so` gives:

`-rwxr-xr-x. 1 root root 437776 Apr  7 00:29 /var/home/woofweasel/.local/lib64/linux-enable-ir-emitter/libcommands.so`

> Could you execute `chmod 777 /var/home/woofweasel/.local/lib64/linux-enable-ir-emitter/libcommands.so` and see if the issue persists?

This appears to have fixed it, thanks! I'll do some further testing just to make absolutely sure, and if so I'll close the issue.

EmixamPP, 2024-04-11:
Hum, as you can see `/var/home/woofweasel/.local/lib64/linux-enable-ir-emitter/libcommands.so` is owned by `root`, and this is the problem I think. This is caused by the meson/ninja installer I guess. With the specific tarball for immutable distro I'm creating that problem shouldn't happen I think

Woofweasel, 2024-04-11:
That makes sense, thanks mate! You've been really helpful :)

Woofweasel, 2024-04-12:
Hey, unfortunately it appears that the issue has persisted- not sure why it worked yesterday. I'm very happy to wait for the new tarball and simply run the program when necessary until then, would you recommend that?

EmixamPP, 2024-04-12:
Same output for the systemd service ?

Woofweasel, 2024-04-12:
What command should I run to find this out? Apologies, I'm relatively new to Linux.

EmixamPP, 2024-04-12:
No problem! `sudo systemctl status linux-enable-ir-emitter`

Woofweasel, 2024-04-12:
that outputs:

```
× linux-enable-ir-emitter.service - enable the infrared emitter
     Loaded: loaded (/etc/systemd/system/linux-enable-ir-emitter.service; enabl>
    Drop-In: /usr/lib/systemd/system/service.d
             └─10-timeout-abort.conf
     Active: failed (Result: exit-code) since Fri 2024-04-12 13:45:17 BST; 19mi>
    Process: 1046 ExecStartPre=modprobe uvcvideo (code=exited, status=0/SUCCESS)
    Process: 1067 ExecStart=/var/home/woofweasel/.local/bin/linux-enable-ir-emi>
   Main PID: 1067 (code=exited, status=1/FAILURE)
        CPU: 54ms

Apr 12 13:45:17 fedora linux-enable-ir-emitter[1067]:   File "/var/home/woofwea>
Apr 12 13:45:17 fedora linux-enable-ir-emitter[1067]:     cpp_commands = ctypes>
Apr 12 13:45:17 fedora linux-enable-ir-emitter[1067]:                    ^^^^^^>
Apr 12 13:45:17 fedora linux-enable-ir-emitter[1067]:   File "/usr/lib64/python>
Apr 12 13:45:17 fedora linux-enable-ir-emitter[1067]:     self._handle = _dlope>
Apr 12 13:45:17 fedora linux-enable-ir-emitter[1067]:                    ^^^^^^>
Apr 12 13:45:17 fedora linux-enable-ir-emitter[1067]: OSError: /var/home/woofwe>
Apr 12 13:45:17 fedora systemd[1]: linux-enable-ir-emitter.service: Main proces>
Apr 12 13:45:17 fedora systemd[1]: linux-enable-ir-emitter.service: Failed with>
Apr 12 13:45:17 fedora systemd[1]: Failed to start linux-enable-ir-emitter.serv>
lines 1-20/20 (END)
```

EmixamPP, 2024-04-17:
Hey, in the PR #174, you can find a bot comment that contains instructions and a link to download and install my latest changes.

Please uninstall first your current version:
```
sudo rm -rf $HOME/.local/lib/linux-enable-ir-emitter \
$HOME/.local/bin/linux-enable-ir-emitter \
$HOME/.local/share/bash-completition/completitions/linux-enable-ir-emitter \
$HOME/.local/share/zsh/site-functions/_linux-enable-ir-emitter \
/etc/systemd/system/linux-enable-ir-emitter.service \
/etc/udev/rules.d/99-linux-enable-ir-emitter.rules
```

If you don't delete the `/etc/linux-enable-ir-emitter`, you do not need to reconfigure

Woofweasel, 2024-04-17:
That command worked after having changed

> `$HOME/.local/lib/linux-enable-ir-emitter`

to `$HOME/.local/lib64/linux-enable-ir-emitter`

However, while I have checked that all those files are deleted, I can still run `linux-enable-ir-emitter run` in order to fix my ir emitter upon a full restart of my device, making me wonder if there's something else that needs to be uninstalled before your new version can be installed?

Also, unfortunately I am very unfamiliar with tarballs, and so I am struggling with the installation of [this one](https://github.com/EmixamPP/linux-enable-ir-emitter/actions/runs/8712096637). Upon downloading it, I execute `unzip linux-enable-ir-emitter*.tar.gz.zip` only to be met with:
```
unzip:  cannot find or open linux-enable-ir-emitter*.tar.gz.zip, linux-enable-ir-emitter*.tar.gz.zip.zip or linux-enable-ir-emitter*.tar.gz.zip.ZIP.

No zipfiles found.
```

EmixamPP, 2024-04-17:
> However, while I have checked that all those files are deleted, I can still run linux-enable-ir-emitter run in order to fix my ir emitter upon a full restart of my device, making me wonder if there's something else that needs to be uninstalled before your new version can be installed?

Execute `which linux-enable-ir-emitter` to see where the executable is located.

> Also, unfortunately I am very unfamiliar with tarballs, and so I am struggling with the installation of [this one](https://github.com/EmixamPP/linux-enable-ir-emitter/actions/runs/8712096637). Upon downloading it, I execute unzip linux-enable-ir-emitter*.tar.gz.zip only to be met with:

Are you in the directory where the archive has been downloaded? e.g. `cd $HOME/Downloads`

Woofweasel, 2024-04-17:
> Execute `which linux-enable-ir-emitter` to see where the executable is located.

This worked and I used `sudo rm -rf` to uninstall the executable. Strangely, after this step, all my issues with howdy have been fixed and my ir emitter works even after a system shut down. Thank you so much for all the work you've put into solving this for me :)

However, just to see if linux-enable-ir-emitter itself worked on my system, I tried installing the tarball (this time from the downloads directory)- the unzip works, but `sudo tar -C / --no-same-owner -h -vxzf linux-enable-ir-emitter*.tar.gz` then outputs:
```
./
./usr/
./usr/local/
./usr/local/bin/
./usr/local/bin/linux-enable-ir-emitter
./usr/local/share/
./usr/local/share/zsh/
./usr/local/share/zsh/site-functions/
./usr/local/share/zsh/site-functions/_linux-enable-ir-emitter
./usr/local/share/licences/
./usr/local/share/licences/linux-enable-ir-emitter/
./usr/local/share/licences/linux-enable-ir-emitter/LICENSE
./usr/local/share/bash-completion/
./usr/local/share/bash-completion/completions/
./usr/local/share/bash-completion/completions/linux-enable-ir-emitter
./etc/
tar: ./usr: Cannot utime: Read-only file system
./etc/systemd/
./etc/systemd/system/
./etc/systemd/system/linux-enable-ir-emitter.service
./etc/linux-enable-ir-emitter/
tar: .: Cannot utime: Operation not permitted
tar: Exiting with failure status due to previous errors
```

EmixamPP, 2024-04-17:
Please try this command: `sudo tar -C / --no-same-owner --no-overwrite-dir -h -vxzf linux-enable-ir-emitter*.tar.gz`

Many thanks to you also for your contribution to this project!

EmixamPP, 2024-04-17:
> Strangely, after this step, all my issues with howdy have been fixed and my ir emitter works even after a system shut down. Thank you so much for all the work you've put into solving this for me :)

Hum, not sure if this will persist

Woofweasel, 2024-04-17:
> Hum, not sure if this will persist

You were spot on, it's broken again.

`sudo tar -C / --no-same-owner --no-overwrite-dir -h -vxzf linux-enable-ir-emitter*.tar.gz` just output:

```
./
./usr/
./usr/local/
./usr/local/bin/
./usr/local/bin/linux-enable-ir-emitter
./usr/local/share/
./usr/local/share/zsh/
./usr/local/share/zsh/site-functions/
./usr/local/share/zsh/site-functions/_linux-enable-ir-emitter
./usr/local/share/licences/
./usr/local/share/licences/linux-enable-ir-emitter/
./usr/local/share/licences/linux-enable-ir-emitter/LICENSE
./usr/local/share/bash-completion/
./usr/local/share/bash-completion/completions/
./usr/local/share/bash-completion/completions/linux-enable-ir-emitter
./etc/
tar: ./usr: Cannot utime: Read-only file system
./etc/systemd/
./etc/systemd/system/
./etc/systemd/system/linux-enable-ir-emitter.service
./etc/linux-enable-ir-emitter/
tar: .: Cannot utime: Operation not permitted
tar: Exiting with failure status due to previous errors
```

EmixamPP, 2024-04-17:
And this one:
`sudo tar -C / --no-same-owner --atime-preserve=system -h -vxzf linux-enable-ir-emitter*.tar.gz`

Woofweasel, 2024-04-17:
Same output

EmixamPP, 2024-04-17:
Hum, ok let's try these two: 
* `sudo tar -C / --no-same-owner -m -h -vxzf linux-enable-ir-emitter*.tar.gz`
* `sudo tar -C / --no-same-owner --no-overwrite-dir -m -h -vxzf linux-enable-ir-emitter*.tar.gz`
Thanks a lot for your time and very prompt response by the way

Woofweasel, 2024-04-17:
That first command seemed to work! The output did not contain the errors:

```
./
./usr/
./usr/local/
./usr/local/bin/
./usr/local/bin/linux-enable-ir-emitter
./usr/local/share/
./usr/local/share/zsh/
./usr/local/share/zsh/site-functions/
./usr/local/share/zsh/site-functions/_linux-enable-ir-emitter
./usr/local/share/licences/
./usr/local/share/licences/linux-enable-ir-emitter/
./usr/local/share/licences/linux-enable-ir-emitter/LICENSE
./usr/local/share/bash-completion/
./usr/local/share/bash-completion/completions/
./usr/local/share/bash-completion/completions/linux-enable-ir-emitter
./etc/
./etc/systemd/
./etc/systemd/system/
./etc/systemd/system/linux-enable-ir-emitter.service
./etc/linux-enable-ir-emitter/
```

I can now run `linux-enable-ir-emitter run` just fine and it works, I will test to see if it works even after a shut down now.

> And does something is extracted? Is /usr/local/bin/linux-enable-ir-emitter exists?

Yep, it exists (after running the above command, not sure if it did before).

EmixamPP, 2024-04-17:
Also execute `sudo systemctl enable --now linux-enable-ir-emitter`

And if the issue persists, show me the output of `systemctl status linux-enable-ir-emitter`

Woofweasel, 2024-04-17:
> Also execute `sudo systemctl enable --now linux-enable-ir-emitter`

This appears not to have changed anything.

> And if the issue persists, show me the output of `systemctl status linux-enable-ir-emitter`

```
○ linux-enable-ir-emitter.service - enable the infrared emitter
     Loaded: loaded (/etc/systemd/system/linux-enable-ir-emitter.service; enabled; preset: disabled)
    Drop-In: /usr/lib/systemd/system/service.d
             └─10-timeout-abort.conf
     Active: inactive (dead) since Wed 2024-04-17 22:26:28 BST; 5min ago
    Process: 1037 ExecStartPre=modprobe uvcvideo (code=exited, status=0/SUCCESS)
    Process: 1057 ExecStartPre=sleep 2 (code=exited, status=0/SUCCESS)
    Process: 1255 ExecStart=linux-enable-ir-emitter -v run (code=exited, status=0/SUCCESS)
   Main PID: 1255 (code=exited, status=0/SUCCESS)
        CPU: 61ms

Apr 17 22:26:26 fedora systemd[1]: Starting linux-enable-ir-emitter.service - enable the infrared emitter...
Apr 17 22:26:28 fedora linux-enable-ir-emitter[1255]: 6.0.0-beta
Apr 17 22:26:28 fedora systemd[1]: linux-enable-ir-emitter.service: Deactivated successfully.
Apr 17 22:26:28 fedora systemd[1]: Finished linux-enable-ir-emitter.service - enable the infrared emitter.
```

EmixamPP, 2024-04-18:
> Main PID: 1255 (code=exited, status=0/SUCCESS)

No no, not the same. Now the systemd service does not exit with error, and the output in the bottom is not the same.

However, I forgot that I changed the shortname for the verbosity. It is not anymore `-v`. Could you do:
1. Open with root privilege `/etc/systemd/system/linux-enable-ir-emitter.service, so e.g. `sudo nano /etc/systemd/system/linux-enable-ir-emitter.service`
2. Modify the `-v` with `--verbose`, thus the line becomes: `linux-enable-ir-emitter --verbose run`
3. Save (with nano: ctrl+o, enter, ctrl+x)
5. `sudo systemctl enable --now linux-enable-ir-emitter`

After that everything should be good!