EmixamPP/linux-enable-ir-emitter #177: ThinkPad X280: Webcam completely dead in both Windows and Linux after configure failed (Unknown USB Device (Device Descriptor Request Failed))

th1nhhdk, 2024-05-21:
**Ouput of `linux-enable-ir-emitter -v configure`**

```
DEBUG: Executing configure command.
INFO: Stand in front of and close to the camera and make sure the room is well lit.
INFO: Ensure to not use the camera during the execution.
DEBUG: Checking if /dev/video0 is a greyscale camera.
^C Ctrl-c again if you really want to, be careful this could break the camera.
```

**Output of `cat /etc/linux-enable-ir-emitter/*`**
```
- corrupted: false
  unit: 3
  selector: 2
  current:
    - 2
  maximum:
    - 3
  minimum:
    - 3
- corrupted: false
  unit: 4
  selector: 2
  current:
    - 0
    - 32
  maximum:
    - 255
    - 255
  minimum:
    - 0
    - 0
- corrupted: false
  unit: 4
  selector: 3
  current:
    - 0
  maximum:
    - 255
  minimum:
    - 0
- corrupted: false
  unit: 4
  selector: 4
  current:
    - 0
    - 0
  maximum:
    - 255
    - 255
  minimum:
    - 0
    - 0
- corrupted: false
  unit: 4
  selector: 5
  current:
    - 255
    - 0
  maximum:
    - 255
    - 255
  minimum:
    - 0
    - 0
- corrupted: false
  unit: 4
  selector: 6
  current:
[... 2564 lines cut ...]
```

**Output of `ls -l /dev/v4l/by-path`**
```
total 0
lrwxrwxrwx. 1 root root 12 May 21 13:37 pci-0000:00:14.0-usb-0:5:1.0-video-index0 -> ../../video0
lrwxrwxrwx. 1 root root 12 May 21 13:37 pci-0000:00:14.0-usb-0:5:1.0-video-index1 -> ../../video1
lrwxrwxrwx. 1 root root 12 May 21 13:37 pci-0000:00:14.0-usb-0:8:1.0-video-index0 -> ../../video2
lrwxrwxrwx. 1 root root 12 May 21 13:37 pci-0000:00:14.0-usb-0:8:1.0-video-index1 -> ../../video3
lrwxrwxrwx. 1 root root 12 May 21 13:37 pci-0000:00:14.0-usbv2-0:5:1.0-video-index0 -> ../../video0
lrwxrwxrwx. 1 root root 12 May 21 13:37 pci-0000:00:14.0-usbv2-0:5:1.0-video-index1 -> ../../video1
lrwxrwxrwx. 1 root root 12 May 21 13:37 pci-0000:00:14.0-usbv2-0:8:1.0-video-index0 -> ../../video2
lrwxrwxrwx. 1 root root 12 May 21 13:37 pci-0000:00:14.0-usbv2-0:8:1.0-video-index1 -> ../../video3
```

**Give more information if you have**

- I Ctrl + C'ed because at the press Y or N no window appeared for a long time
- After that the device names are messed up, both the webcam and IR camera in both Windows and Linux stopped working
- In device manager the device show up as "Unknown USB Device (Device Descriptor Request Failed)"

**Additional info**
 - Computer (or camera) model:  ThinkPad X280
 - Linux distribution: Fedora Linux 40 (KDE Plasma)
 - Version of linux-enable-ir-emitter: linux-enable-ir-emitter 6.0.0-beta

th1nhhdk, 2024-05-21:
- The IR camera starts working again after reinstalling the firmware
- But the Webcam remains broken (Showing wrong name in Device Manager)

Settordici, 2024-05-24:
I have the same issue, on Fedora 40 as well. I ran linux-enable-ir-emitter configure and after answering yes to a question everything is stuck. I'm forced to Ctrl-C and potentially break everything. I hope this is not the case, I will update later my comment if something changes

EmixamPP, 2024-05-25:
I will try to investigate in the coming days your troubles (I'm currently quite busy, so I can't promise).

I think I should definitively handle the ctrl-c better. And secondly, I should also print some progress bar to let know if the process is really stuck or not.

EmixamPP, 2024-05-26:
The PR #178 will ensure that issue will never happen anymore. However, I cannot undone what happened, I hope reinstalling firmware and drivers through windows will fix your issue.

th1nhhdk, 2024-06-01:
You should made the program only run if the user provides `-d`

Some SunplusIT Cameras will be destroyed if the tool tries to find the IR Camera because the user doesn't provide `-d`, you should also add that warning

EmixamPP, 2024-06-01:
Ok, I will force to always specify the infrared camera. And maybe allow the automatic detection using an option then.

Did you manage to fix your camera?

terrabys, 2024-06-02:
So is there no way to make SunplusIT camera working?

th1nhhdk, 2024-06-02:
> Did you manage to fix your camera?

I contacted SunplusIT, they told me to contact Lenovo instead

Lenovo forums said to reinstall the driver using the system image, I'll try it later (probably won't work)

terrabys, 2024-06-02:
> these

Thx, sorry for bad explanation, the IR camera itself works fine, but I couldn't make the IR emitter to work, the utility gets to step where it runs 255 and it hangs.

terrabys, 2024-06-02:
> > So is there no way to make SunplusIT camera to work?
> 
> I don't know if SunplusIT's Cameras work but I know that the program may brick these cameras

Thx, sorry for bad explanation, the IR camera itself works fine, but I couldn't make the IR emitter to work, the utility gets to step where it runs 255 and it hangs.

EmixamPP, 2024-06-02:
> Thx, sorry for bad explanation, the IR camera itself works fine, but I couldn't make the IR emitter to work, the utility gets to step where it runs 255 and it hangs.

If I understood well, you enables the verbose mode (-v) and you clearly saw that no more logs were outputed?

th1nhhdk, 2024-06-05:
> > Did you manage to fix your camera?
> 
> I contacted SunplusIT, they told me to contact Lenovo instead
> 
> Lenovo forums said to reinstall the driver using the system image, I'll try it later (probably won't work)

Lenovo forums told me to file a support ticket, after comtacting support and trying basic solutions they told me to bring the machine to the service center, but it's not worth it.

So, yeah, the Webcam's completely dead and there's no way to fix it

closed 2024-06-05 by th1nhhdk

aarontyliu, 2024-07-28:
@th1nhhdk I encountered a similar issue (X1C6). Do you think installing a new set of integrated camera would work?

th1nhhdk, 2024-07-28:
> @th1nhhdk I encountered a similar issue (X1C6). Do you think installing a new set of integrated camera would work (i.e. is the webcam really dead)? 

I can comfirm that the firmware is broken, if you can reflash the firmware, webcam will start working again, but Lenovo won't provide the firmware no matter how much I asked

EmixamPP, 2024-07-28:
Maybe try your chance with the camera constructor (azureware?)

EmixamPP, 2024-07-28:
> @th1nhhdk I encountered a similar issue (X1C6). Do you think installing a new set of integrated camera would work (i.e. is the webcam really dead)? 

When you say similar, could you develop?

aarontyliu, 2024-07-28:
> > @th1nhhdk I encountered a similar issue (X1C6). Do you think installing a new set of integrated camera would work (i.e. is the webcam really dead)?
> 
> When you say similar, could you develop?

I also Ctrl + C'ed during the process (Y/N questions about IR camera) for it took a bit long.

```bash
v4l2-ctl --list-devices
SPCA2085 PC Camera: PC Camera (usb-0000:00:14.0-5):
	/dev/video1
	/dev/video2
	/dev/media0

SPCA2087 PC Camera: PC Camera (usb-0000:00:14.0-8):
	/dev/video3
	/dev/video4
	/dev/media1
```

aarontyliu, 2024-07-30:
> Maybe try your chance with the camera constructor (azureware?)

Do you know if there's any alternative for ubuntu?

aarontyliu, 2024-08-10:
Update: I swapped a new camera module and it's working now.