EmixamPP/linux-enable-ir-emitter #180: Video feed is completely black

Anonymous-ABC-XYZ, 2024-07-07:
**Ouput of `cat /var/local/log/linux-enable-ir-emitter`**
```
[00:19:28] [info] Stand in front of and close to the camera and make sure the room is well lit.
[00:19:28] [info] Ensure to not use the camera during the execution.
[00:19:28] [critical] OpenCV cannot access to /dev/video1
[00:30:16] [info] Stand in front of and close to the camera and make sure the room is well lit.
[00:30:16] [info] Ensure to not use the camera during the execution.
[00:30:16] [critical] OpenCV cannot access to /dev/video2
[00:30:18] [info] Stand in front of and close to the camera and make sure the room is well lit.
[00:30:18] [info] Ensure to not use the camera during the execution.
[00:30:18] [critical] OpenCV cannot access to /dev/video2
[00:30:23] [info] Stand in front of and close to the camera and make sure the room is well lit.
[00:30:23] [info] Ensure to not use the camera during the execution.
[00:30:23] [critical] OpenCV cannot access to /dev/video2
[00:30:24] [info] Stand in front of and close to the camera and make sure the room is well lit.
[00:30:24] [info] Ensure to not use the camera during the execution.
[00:30:24] [critical] OpenCV cannot access to /dev/video2
[00:30:25] [info] Stand in front of and close to the camera and make sure the room is well lit.
[00:30:25] [info] Ensure to not use the camera during the execution.
[00:30:25] [critical] OpenCV cannot access to /dev/video2
[00:30:26] [info] Stand in front of and close to the camera and make sure the room is well lit.
[00:30:26] [info] Ensure to not use the camera during the execution.
[00:30:26] [critical] OpenCV cannot access to /dev/video2
```

**Output of `v4l2-ctl --list-devices`**
```
USB2.0 FHD UVC WebCam: USB2.0 F (usb-0000:65:00.4-1):
	/dev/video0
	/dev/video1
	/dev/video2
	/dev/video3
	/dev/media0
	/dev/media1
```

**Ouput of `for dev in /dev/video*; do echo $dev && v4l2-ctl -d $dev --list-formats-ext; done`**
```
/dev/video0
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'MJPG' (Motion-JPEG, compressed)
		Size: Discrete 1920x1080
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 800x600
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 352x288
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 176x144
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 160x120
			Interval: Discrete 0.033s (30.000 fps)
	[1]: 'YUYV' (YUYV 4:2:2)
		Size: Discrete 1920x1080
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 800x600
			Interval: Discrete 0.050s (20.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 352x288
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 176x144
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 160x120
			Interval: Discrete 0.033s (30.000 fps)
/dev/video1
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

/dev/video2
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'GREY' (8-bit Greyscale)
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
/dev/video3
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture
```

**Output of `ls -l /dev/v4l/by-path`**
```
lrwxrwxrwx 1 root root 12 Jul  8 01:50 pci-0000:65:00.4-usb-0:1:1.0-video-index0 -> ../../video0
lrwxrwxrwx 1 root root 12 Jul  8 01:50 pci-0000:65:00.4-usb-0:1:1.0-video-index1 -> ../../video1
lrwxrwxrwx 1 root root 12 Jul  8 01:50 pci-0000:65:00.4-usb-0:1:1.2-video-index0 -> ../../video2
lrwxrwxrwx 1 root root 12 Jul  8 01:50 pci-0000:65:00.4-usb-0:1:1.2-video-index1 -> ../../video3
lrwxrwxrwx 1 root root 12 Jul  8 01:50 pci-0000:65:00.4-usbv2-0:1:1.0-video-index0 -> ../../video0
lrwxrwxrwx 1 root root 12 Jul  8 01:50 pci-0000:65:00.4-usbv2-0:1:1.0-video-index1 -> ../../video1
lrwxrwxrwx 1 root root 12 Jul  8 01:50 pci-0000:65:00.4-usbv2-0:1:1.2-video-index0 -> ../../video2
lrwxrwxrwx 1 root root 12 Jul  8 01:50 pci-0000:65:00.4-usbv2-0:1:1.2-video-index1 -> ../../video3
```

**Give more information if you have**

The feed is entirely blank, leaving me no option to click yes or no. Additionally, the IR emitter is also black, with no light coming from it.

[image]

**Additional info**
 - Computer (or camera) model: ROG Zephyrus G14 GA402XV
 - Linux distribution: Arch Linus
 - Version of linux-enable-ir-emitter: 6.0

Anonymous-ABC-XYZ, 2024-07-09:
Changed distro to KDE Neon still the same issue as the previous comment

EmixamPP, 2024-07-09:
You have to click on the window, then press the keyboard key Y or N; for yes the emitter is blinking or no it isn't.
Since the message seems not clear, do you have suggestion to avoid the same confusion to future users?

Anonymous-ABC-XYZ, 2024-07-09:
I would suggest that you change this(Is the video flashing? Press Y or N in the window.) to: `Is the video flashing? Please click on the video feed, and then press the keyboard key Y (yes the emitter is blinking) or N(no it isn't) ` 

Also, it worked perfectly now, thank you!

closed 2024-07-09 by Anonymous-ABC-XYZ