EmixamPP/linux-enable-ir-emitter #181: Configuration Lenovo Yoga 9 2-in-1 14IMH9 camera

binSmile, 2024-07-10:
**Ouput of `cat /var/local/log/linux-enable-ir-emitter`**
```
[12:00:08] [info] Stand in front of and close to the camera and make sure the room is well lit.
[12:00:08] [info] Ensure to not use the camera during the execution.
[12:00:09] [info] Configuring the camera /dev/video2.
[12:00:09] [debug] Previous configuration found.
[12:00:26] [debug] Instruction applied: unit: 4, selector: 2, control: 1 0.
[12:00:28] [debug] Instruction applied: unit: 4, selector: 2, control: 2 0.
[12:00:29] [debug] Instruction applied: unit: 4, selector: 2, control: 3 0.
[12:00:31] [debug] Instruction applied: unit: 4, selector: 2, control: 4 0.
[12:00:32] [debug] Instruction applied: unit: 4, selector: 2, control: 5 0.
[12:00:34] [debug] Instruction applied: unit: 4, selector: 2, control: 6 0.
[12:00:35] [debug] Instruction applied: unit: 4, selector: 2, control: 7 0.
[12:00:37] [debug] Instruction applied: unit: 4, selector: 2, control: 8 0.
[12:00:38] [debug] Instruction applied: unit: 4, selector: 2, control: 9 0.
[12:00:40] [debug] Instruction applied: unit: 4, selector: 2, control: 255 255.
[12:00:41] [debug] Reseting to the instruction: unit: 4, selector: 2, control: 0 32.
[12:00:41] [debug] Instruction applied: unit: 4, selector: 3, control: 1.
[12:05:42] [debug] Instruction applied: unit: 4, selector: 3, control: 2.
[12:10:42] [debug] Instruction applied: unit: 4, selector: 3, control: 3.
[12:15:43] [debug] Instruction applied: unit: 4, selector: 3, control: 4.
[12:20:44] [debug] Instruction applied: unit: 4, selector: 3, control: 5.
[12:25:45] [debug] Instruction applied: unit: 4, selector: 3, control: 6.
[12:30:45] [debug] Instruction applied: unit: 4, selector: 3, control: 7.
[12:35:46] [debug] Instruction applied: unit: 4, selector: 3, control: 8.
[12:40:47] [debug] Instruction applied: unit: 4, selector: 3, control: 9.
[12:45:47] [debug] Instruction applied: unit: 4, selector: 3, control: 255.
[12:45:47] [debug] The instruction is not valid.
[12:45:47] [debug] Reseting to the instruction: unit: 4, selector: 3, control: 0.
[12:45:47] [debug] Instruction applied: unit: 4, selector: 4, control: 1 0.
[12:45:47] [debug] The instruction is not valid.
[12:45:47] [debug] Instruction applied: unit: 4, selector: 4, control: 2 0.
[12:45:47] [debug] The instruction is not valid.
[12:45:47] [debug] Instruction applied: unit: 4, selector: 4, control: 3 0.
[12:45:47] [debug] The instruction is not valid.
[12:45:47] [debug] Instruction applied: unit: 4, selector: 4, control: 4 0.
[12:45:47] [debug] The instruction is not valid.
[12:45:47] [debug] Instruction applied: unit: 4, selector: 4, control: 5 0.
[12:45:47] [debug] The instruction is not valid.
[12:45:47] [debug] Instruction applied: unit: 4, selector: 4, control: 6 0.
[12:45:47] [debug] The instruction is not valid.
[12:45:47] [debug] Instruction applied: unit: 4, selector: 4, control: 7 0.
[12:45:47] [debug] The instruction is not valid.
[12:45:47] [debug] Instruction applied: unit: 4, selector: 4, control: 8 0.
[12:45:47] [debug] The instruction is not valid.
[12:45:47] [debug] Instruction applied: unit: 4, selector: 4, control: 9 0.
[12:45:47] [debug] The instruction is not valid.
[12:45:47] [debug] Instruction applied: unit: 4, selector: 4, control: 255 255.
[12:45:47] [debug] The instruction is not valid.
[12:45:47] [debug] Reseting to the instruction: unit: 4, selector: 4, control: 0 0.
[12:45:47] [debug] Instruction applied: unit: 4, selector: 5, control: 1 0.
[12:45:47] [debug] The instruction is not valid.
[12:45:47] [debug] Instruction applied: unit: 4, selector: 5, control: 2 0.
[12:45:47] [debug] The instruction is not valid.
[12:45:47] [debug] Instruction applied: unit: 4, selector: 5, control: 3 0.
[12:45:47] [debug] The instruction is not valid.
[12:45:47] [debug] Instruction applied: unit: 4, selector: 5, control: 4 0.
[12:45:47] [debug] The instruction is not valid.
[12:45:47] [debug] Instruction applied: unit: 4, selector: 5, control: 5 0.
[12:45:47] [debug] The instruction is not valid.
[12:45:47] [debug] Instruction applied: unit: 4, selector: 5, control: 6 0.
[12:45:47] [debug] The instruction is not valid.
[12:45:47] [debug] Instruction applied: unit: 4, selector: 5, control: 7 0.
[12:45:47] [debug] The instruction is not valid.
[12:45:47] [debug] Instruction applied: unit: 4, selector: 5, control: 8 0.
[12:45:47] [debug] The instruction is not valid.
[12:45:47] [debug] Instruction applied: unit: 4, selector: 5, control: 9 0.
[12:45:47] [debug] The instruction is not valid.
[12:45:47] [debug] Instruction applied: unit: 4, selector: 5, control: 255 255.
[12:45:47] [debug] The instruction is not valid.
[... 179 lines cut ...]
lrwxrwxrwx 1 root root 12 july 10 14:11 pci-0000:00:14.0-usb-0:8:1.2-video-index0 -> ../../video2
lrwxrwxrwx 1 root root 12 july 10 14:11 pci-0000:00:14.0-usb-0:8:1.2-video-index1 -> ../../video3
lrwxrwxrwx 1 root root 12 july 10 14:11 pci-0000:00:14.0-usbv2-0:8:1.0-video-index0 -> ../../video0
lrwxrwxrwx 1 root root 12 july 10 14:11 pci-0000:00:14.0-usbv2-0:8:1.0-video-index1 -> ../../video1
lrwxrwxrwx 1 root root 12 july 10 14:11 pci-0000:00:14.0-usbv2-0:8:1.2-video-index0 -> ../../video2
lrwxrwxrwx 1 root root 12 july 10 14:11 pci-0000:00:14.0-usbv2-0:8:1.2-video-index1 -> ../../video3

```

**Give more information if you have**
I use that comand for configuration:
`sudo linux-enable-ir-emitter -d /dev/video2 -v configure -l`

I have windows on that laptop. If you have guide, I could sniff control comands on windows. 

Also, probably, something goes wrong (may be i stoped with force configuration process), and now I should open qv4l and turn on "auto exprosure"  after each boot...

Thank you very much!

**Additional info**
 - Computer (or camera) model: 
    - Lenovo Yoga 9 2-in-1 14IMH9 (83AC003RYA)
    - Bus 003 Device 008: ID 04f2:b7ed Chicony Electronics Co., Ltd Integrated Camera
 - Linux distribution: kubuntu (trixie/sid)
 - Version of linux-enable-ir-emitter: 0x5ecbd7238080

EmixamPP, 2024-07-10:
> Sorry, I don't understatnd what you mean.

My bad, the right command is: `find /etc/linux-enable-ir-emitter -type f -exec sh -c 'echo "File: {}"; cat {}; echo ""' \;`

EmixamPP, 2024-07-13:
Or just execute `sudo linux-enable-ir-emitter tweak` and past here the output

EmixamPP, 2024-07-13:
Could you provide me the gdb backtrace for the configure command? 
Please follow the instructions of the bot comment in the PR: https://github.com/EmixamPP/linux-enable-ir-emitter/pull/184. Thank you

binSmile, 2024-07-17:
> > Sorry, I don't understatnd what you mean.
> 
> My bad, the right command is: `find /etc/linux-enable-ir-emitter -type f -exec sh -c 'echo "File: {}"; cat {}; echo ""' \;`

Thanks! I put the output to the first message.

```
$ sudo linux-enable-ir-emitter tweak
Tweaking the camera /dev/video2
Caution, you could break the camera.
0) unit: 4, selector: 2, control: 0 32
1) unit: 4, selector: 3, control: 0
2) unit: 4, selector: 4, control: 0 0
3) unit: 4, selector: 5, control: 255 255
4) unit: 4, selector: 6, control: 1
5) unit: 4, selector: 7, control: 1
6) unit: 4, selector: 8, control: 3
7) unit: 4, selector: 9, control: 0
8) unit: 4, selector: 12, control: 0
9) unit: 4, selector: 13, control: 0 0 0 0
10) unit: 4, selector: 14, control: 0 0 0 0
11) unit: 4, selector: 15, control: 149 40 4 242 183 237 2 0 1 255 65 255 0 0 0 0 0 0 16 0 0 0 0 1 2 252 12 253 163 57 40 239 113 9 219 72 255 218 243 205 105 174 73 148 47 211 219 178 155 107 118 245 114 10 15 177 0 0 0 1 0 3 240 0 0 0 0 1 2 252 12 253 163 57 40 239 113 9 219 72 255 218 243 205 105 174 73 148 47 211 219 178 155 107 118 245 114 10 15 177 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255
12) unit: 4, selector: 16, control: 3
13) unit: 4, selector: 17, control: 0 8 0 143 170 41 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
14) unit: 4, selector: 21, control: 1
15) unit: 4, selector: 22, control: 149 40 4 242 183 237 2 0 1 255 65 255 0 0 0 0 0 0 16 0 0 0 0 1 2 252 12 253 163 57 40 239 113 9 219 72 255 218 243 205 105 174 73 148 47 211 219 178 155 107 118 245 114 10 15 177 0 0 0 1 0 3 240 0
16) unit: 4, selector: 23, control: 0 0 0
17) unit: 4, selector: 25, control: 1
18) unit: 4, selector: 26, control: 255 255 255 255
19) unit: 4, selector: 27, control: 255 255 255 255
20) unit: 5, selector: 2, control: 0
21) unit: 5, selector: 4, control: 183 0
22) unit: 7, selector: 2, control: 1 0 0 0 0 0 0 96 97 2 0 0 0 0 0
23) unit: 7, selector: 3, control: 16 0 0 0 0 0 0 0 0 0 0
24) unit: 7, selector: 6, control: 1 3 3 0 0 0 0 0 0
25) unit: 7, selector: 9, control: 1 0 0 0
26) exit

```

binSmile, 2024-07-17:
> Could you provide me the gdb backtrace for the configure command? Please follow the instructions of the bot comment in the PR: #184. Thank you

Yes, sure!
But, I'll try to do it at the evening.

binSmile, 2024-07-18:
> Could you provide me the gdb backtrace for the configure command? Please follow the instructions of the bot comment in the PR: #184. Thank you

Finally, I did it!

`~$ sudo gdb --ex run --args linux-enable-ir-emitter -d /dev/video2 -v configure -l`
<details>
  <summary>Click to expand</summary>
```
GNU gdb (Ubuntu 15.0.50.20240403-0ubuntu1) 15.0.50.20240403-git
Copyright (C) 2024 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from linux-enable-ir-emitter...
Starting program: /usr/local/bin/linux-enable-ir-emitter -d /dev/video2 -v configure -l
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Executing configure command.
Stand in front of and close to the camera and make sure the room is well lit.
Ensure to not use the camera during the execution.
[New Thread 0x7ffff5a006c0 (LWP 7298)]
[New Thread 0x7ffff50006c0 (LWP 7299)]
[New Thread 0x7fffefe006c0 (LWP 7300)]
[New Thread 0x7fffef4006c0 (LWP 7301)]
[New Thread 0x7fffeea006c0 (LWP 7302)]
[New Thread 0x7fffee0006c0 (LWP 7303)]
[New Thread 0x7fffed6006c0 (LWP 7304)]
[New Thread 0x7fffecc006c0 (LWP 7305)]
[New Thread 0x7fffe7e006c0 (LWP 7306)]
[New Thread 0x7fffe74006c0 (LWP 7307)]
[New Thread 0x7fffe6a006c0 (LWP 7308)]
[New Thread 0x7fffe60006c0 (LWP 7309)]
[New Thread 0x7fffe56006c0 (LWP 7310)]
[New Thread 0x7fffe4c006c0 (LWP 7311)]
[New Thread 0x7fffdbe006c0 (LWP 7312)]
[New Thread 0x7fffdb4006c0 (LWP 7313)]
[New Thread 0x7fffdaa006c0 (LWP 7314)]
[New Thread 0x7fffda0006c0 (LWP 7315)]
[New Thread 0x7fffd96006c0 (LWP 7316)]
[New Thread 0x7fffd8c006c0 (LWP 7317)]
[New Thread 0x7fffcfe006c0 (LWP 7318)]
Configuring the camera /dev/video2.
Previous configuration found.
Is the video flashing? Press Y or N in the window. [Detaching after fork from child process 7319]
n
N pressed.
Instruction applied: unit: 4, selector: 2, control: 1 0.
Instruction applied: unit: 4, selector: 2, control: 2 0.
Instruction applied: unit: 4, selector: 2, control: 3 0.
Instruction applied: unit: 4, selector: 2, control: 4 0.
Instruction applied: unit: 4, selector: 2, control: 5 0.
Instruction applied: unit: 4, selector: 2, control: 6 0.
Instruction applied: unit: 4, selector: 2, control: 7 0.
Instruction applied: unit: 4, selector: 2, control: 8 0.
Instruction applied: unit: 4, selector: 2, control: 9 0.
Instruction applied: unit: 4, selector: 2, control: 255 255.
Reseting to the instruction: unit: 4, selector: 2, control: 0 32.
Instruction applied: unit: 4, selector: 3, control: 1.
Instruction applied: unit: 4, selector: 3, control: 2.
Instruction applied: unit: 4, selector: 3, control: 3.
[... 155 lines cut ...]
[Thread 0x7fffdaa006c0 (LWP 7314) exited]
[Thread 0x7fffdb4006c0 (LWP 7313) exited]
[Thread 0x7fffdbe006c0 (LWP 7312) exited]
[Thread 0x7fffe4c006c0 (LWP 7311) exited]
[Thread 0x7fffe56006c0 (LWP 7310) exited]
[Thread 0x7fffe60006c0 (LWP 7309) exited]
[Thread 0x7fffe6a006c0 (LWP 7308) exited]
[Thread 0x7fffe74006c0 (LWP 7307) exited]
[Thread 0x7fffe7e006c0 (LWP 7306) exited]
[Thread 0x7fffecc006c0 (LWP 7305) exited]
[Thread 0x7fffed6006c0 (LWP 7304) exited]
[Thread 0x7fffee0006c0 (LWP 7303) exited]
[Thread 0x7fffeea006c0 (LWP 7302) exited]
[Thread 0x7fffef4006c0 (LWP 7301) exited]
[Thread 0x7fffefe006c0 (LWP 7300) exited]
[Thread 0x7ffff5a006c0 (LWP 7298) exited]
[Thread 0x7ffff5daf380 (LWP 7295) exited]
[Thread 0x7ffff50006c0 (LWP 7299) exited]
[New process 7295]

Program terminated with signal SIGABRT, Aborted.
The program no longer exists.

```
</details>

EmixamPP, 2024-07-18:
Aah, what I really needed was the result of the `bt` (or `backtrace`) command inside gdb after the segfault :confused:
Sorry, may be the explanation wasn't clear

binSmile, 2024-07-20:
My fault. I was inattentive...

binSmile, 2024-07-31:
I apologize for the delay.

1. I attempted to retrieve the bt, but gdb reported that it doesn't have any data (perhaps the process faulted or something similar).
2. Do you have any tips on how to bypass the successful steps and quickly reach the failure point? It's quite time-consuming.
3. Additionally, it shut down the main camera, making it difficult to proceed during the day

EmixamPP, 2024-07-31:
No problem, thanks for your precious time!

1. Could you show me the outputs? Are you indeed using the version of the PR #184? Because it contains debug symbols that allow gdb to analyze what happen, while the "usual" version not.
2. Yes, you can add   to the configure command `--limit 1` to reduce the search to one try per instruction pattern, but also `--no-gui` to remove the loading of the video feedback, and you can also add `-m` to answer yourself in the terminal and not let the tool determine itself by comparing the frames if the emitter is working or not.
So you can add `-l 1 -g -m`.
3. I am not sure to understand your question, but maybe my previous answer will solve that

Anyway, if you have trouble do not worry too much, this bug exists for a long time, and I was never able to find why it occurs for a few people. So now, by providing a debug version, I was hoping to have more inputs to find the cause.

But I do not want to take you more time. I maybe have an idea to enable your emitter, by bypassing the automatic configuration that crashes.
Execute the tweak command, input the number `24`, and enter `1 3 2 0 0 0 0 0 0` as value. If it works, you may see the video feedback blinking / or see a red light at your camera.

EmixamPP, 2024-09-01:
Feel free to re-comment to reopen the discussion

binSmile, 2024-10-13:
Hi, @EmixamPP !
Finally, i did it! =)
Is it right output?

<details>
  <summary>Click to expand</summary>

```
$ sudo gdb --ex run --args linux-enable-ir-emitter -d /dev/video2 -v configure -l
GNU gdb (Ubuntu 15.0.50.20240403-0ubuntu1) 15.0.50.20240403-git
Copyright (C) 2024 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from linux-enable-ir-emitter...
Starting program: /usr/local/bin/linux-enable-ir-emitter -d /dev/video2 -v configure -l
warning: could not find '.gnu_debugaltlink' file for /lib/x86_64-linux-gnu/libgdk-3.so.0
warning: could not find '.gnu_debugaltlink' file for /lib/x86_64-linux-gnu/libgobject-2.0.so.0
warning: could not find '.gnu_debugaltlink' file for /lib/x86_64-linux-gnu/libglib-2.0.so.0
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
warning: could not find '.gnu_debugaltlink' file for /lib/x86_64-linux-gnu/libgmodule-2.0.so.0
warning: could not find '.gnu_debugaltlink' file for /lib/x86_64-linux-gnu/libbrotlidec.so.1
warning: could not find '.gnu_debugaltlink' file for /lib/x86_64-linux-gnu/libbrotlicommon.so.1
warning: could not find '.gnu_debugaltlink' file for /lib/x86_64-linux-gnu/libcap.so.2
Executing configure command.
Stand in front of and close to the camera and make sure the room is well lit.
Ensure to not use the camera during the execution.
[New Thread 0x7ffff5a006c0 (LWP 107447)]
[New Thread 0x7ffff50006c0 (LWP 107448)]
[New Thread 0x7fffefe006c0 (LWP 107449)]
[New Thread 0x7fffef4006c0 (LWP 107450)]
[New Thread 0x7fffeea006c0 (LWP 107451)]
[New Thread 0x7fffee0006c0 (LWP 107452)]
[New Thread 0x7fffed6006c0 (LWP 107453)]
[New Thread 0x7fffecc006c0 (LWP 107454)]
[New Thread 0x7fffe7e006c0 (LWP 107455)]
[New Thread 0x7fffe74006c0 (LWP 107456)]
[New Thread 0x7fffe6a006c0 (LWP 107457)]
[New Thread 0x7fffe60006c0 (LWP 107458)]
[New Thread 0x7fffe56006c0 (LWP 107459)]
[New Thread 0x7fffe4c006c0 (LWP 107460)]
[New Thread 0x7fffdbe006c0 (LWP 107461)]
[New Thread 0x7fffdb4006c0 (LWP 107462)]
[New Thread 0x7fffdaa006c0 (LWP 107463)]
[New Thread 0x7fffda0006c0 (LWP 107464)]
[New Thread 0x7fffd96006c0 (LWP 107465)]
[New Thread 0x7fffd8c006c0 (LWP 107466)]
[New Thread 0x7fffcfe006c0 (LWP 107467)]
Configuring the camera /dev/video2.
Previous configuration found.
Is the video flashing? Press Y or N in the window. warning: could not find '.gnu_debugaltlink' file for /usr/lib/x86_64-linux-gnu/gio/modules/libgvfsdbus.so
warning: could not find '.gnu_debugaltlink' file for /usr/lib/x86_64-linux-gnu/gvfs/libgvfscommon.so
N pressed.
Instruction applied: unit: 4, selector: 2, control: 1 0.
Instruction applied: unit: 4, selector: 2, control: 2 0.
Instruction applied: unit: 4, selector: 2, control: 3 0.
Instruction applied: unit: 4, selector: 2, control: 4 0.
Instruction applied: unit: 4, selector: 2, control: 5 0.
Instruction applied: unit: 4, selector: 2, control: 6 0.
[... 240 lines cut ...]
kernel: usb 3-8: Failed to query (GET_LEN) UVC control 13 on unit 4: -19 (exp. 2).
kernel: usb 3-8: Failed to query (GET_LEN) UVC control 13 on unit 4: -19 (exp. 2).
kernel: usb 3-8: Failed to query (GET_LEN) UVC control 13 on unit 4: -19 (exp. 2).
kernel: usb 3-8: Failed to query (GET_LEN) UVC control 13 on unit 4: -19 (exp. 2).
kernel: usb 3-8: Failed to query (GET_LEN) UVC control 13 on unit 4: -19 (exp. 2).
kernel: usb 3-8: Failed to query (GET_LEN) UVC control 13 on unit 4: -19 (exp. 2).
kernel: usb 3-8: Failed to query (GET_LEN) UVC control 13 on unit 4: -19 (exp. 2).
kernel: usb 3-8: Failed to query (GET_LEN) UVC control 13 on unit 4: -19 (exp. 2).
kernel: usb 3-8: Failed to query (GET_LEN) UVC control 13 on unit 4: -19 (exp. 2).
kernel: usb 3-8: Failed to query (GET_LEN) UVC control 13 on unit 4: -19 (exp. 2).
kernel: usb 3-8: Failed to query (GET_LEN) UVC control 13 on unit 4: -19 (exp. 2).
kernel: usb 3-8: Failed to query (GET_LEN) UVC control 14 on unit 4: -19 (exp. 2).
kernel: usb 3-8: Failed to query (GET_LEN) UVC control 14 on unit 4: -19 (exp. 2).
kernel: usb 3-8: Failed to query (GET_LEN) UVC control 14 on unit 4: -19 (exp. 2).
kernel: usb 3-8: Failed to query (GET_LEN) UVC control 14 on unit 4: -19 (exp. 2).
kernel: usb 3-8: Failed to query (GET_LEN) UVC control 14 on unit 4: -19 (exp. 2).
kernel: usb 3-8: Failed to query (GET_LEN) UVC control 14 on unit 4: -19 (exp. 2).
kernel: usb 3-8: Failed to query (GET_LEN) UVC control 14 on unit 4: -19 (exp. 2).
kernel: usb 3-8: Failed to query (GET_LEN) UVC control 14 on unit 4: -19 (exp. 2).
kernel: usb 3-8: Failed to query (GET_LEN) UVC control 14 on unit 4: -19 (exp. 2).
kernel: usb 3-8: Failed to query (GET_LEN) UVC control 14 on unit 4: -19 (exp. 2).
kernel: usb 3-8: Failed to query (GET_LEN) UVC control 14 on unit 4: -19 (exp. 2).

```
</details>

binSmile, 2024-10-13:
> Execute the tweak command, input the number 24, and enter 1 3 2 0 0 0 0 0 0 as value. If it works, you may see the video feedback blinking / or see a red light at your camera.

Yeah! It's work!

binSmile, 2024-10-24:
Hi, @EmixamPP !

> Execute the tweak command, input the number 24, and enter 1 3 2 0 0 0 0 0 0 as value. If it works, you may see the video feedback blinking / or see a red light at your camera.

What I meant is that after tweaking, my color camera starts in 'Manual Mode' for the Auto Exposure setting instead of 'Aperture Priority Mode'. Do you know how to fix that? I had similar behavior with the camera on startup when I tried to configure it automatically.

Also, could you share any notes or resources on how to better understand the camera's parameters?

EmixamPP, 2024-11-23:
> What I meant is that after tweaking, my color camera starts in 'Manual Mode' for the Auto Exposure setting instead of 'Aperture Priority Mode'. Do you know how to fix that? I had similar behavior with the camera on startup when I tried to configure it automatically.

> Also, could you share any notes or resources on how to better understand the camera's parameters?

Hum, that I do not know, the issue is that without any driver/firmware documentation we cannot understand anything from these cameras. So the only way to get them working as we want is with trial-error.

I can only advise you to play with `sudo linux-enable-ir-emitter tweak`  by changing the control value for each instruction. There is a small description in the README https://github.com/EmixamPP/linux-enable-ir-emitter?tab=readme-ov-file#how-do-i-tweak-my-camera, if you need more help please feel to ask!

EmixamPP, 2024-11-23:
And thank you for the gdb backtrack! One bug less! I merged the fix on master, the tarball can be downloaded here: https://github.com/EmixamPP/linux-enable-ir-emitter/actions/runs/11986854302