EmixamPP/linux-enable-ir-emitter #182: Lenovo Ideapad Pro 5 configuraiton failing

eu-david, 2024-07-13:
**Ouput of `cat /var/local/log/linux-enable-ir-emitter`**
```
Executing configure command.
Stand in front of and close to the camera and make sure the room is well lit.
Ensure to not use the camera during the execution.
Checking if /dev/video0 is a grayscale camera.
Checking if /dev/video1 is a grayscale camera.
Checking if /dev/video2 is a grayscale camera.
/dev/video2 is a grayscale camera.
Configuring the camera /dev/video2.
Previous configuration found.
Is the video flashing? Press Y or N in the window. 
** (linux-enable-ir-emitter:4753): WARNING **: 11:06:23.046: AT-SPI: Could not obtain desktop path or name

** (linux-enable-ir-emitter:4753): WARNING **: 11:06:23.047: atk-bridge: GetRegisteredEvents returned message with unknown signature

** (linux-enable-ir-emitter:4753): WARNING **: 11:06:23.047: atk-bridge: get_device_events_reply: unknown signature
N pressed.
Instruction applied: unit: 4, selector: 2, control: 1 0.
Instruction applied: unit: 4, selector: 2, control: 2 0.
Instruction applied: unit: 4, selector: 2, control: 3 0.
Instruction applied: unit: 4, selector: 2, control: 4 0.
Instruction applied: unit: 4, selector: 2, control: 5 0.
Instruction applied: unit: 4, selector: 2, control: 6 0.
Instruction applied: unit: 4, selector: 2, control: 7 0.
Instruction applied: unit: 4, selector: 2, control: 8 0.
Instruction applied: unit: 4, selector: 2, control: 9 0.
Instruction applied: unit: 4, selector: 2, control: 255 255.
Reseting to the instruction: unit: 4, selector: 2, control: 0 32.
Instruction applied: unit: 4, selector: 3, control: 1.
Instruction applied: unit: 4, selector: 3, control: 2.
Instruction applied: unit: 4, selector: 3, control: 3.
Instruction applied: unit: 4, selector: 3, control: 4.
Instruction applied: unit: 4, selector: 3, control: 5.
Instruction applied: unit: 4, selector: 3, control: 6.
Instruction applied: unit: 4, selector: 3, control: 7.
Instruction applied: unit: 4, selector: 3, control: 8.
Instruction applied: unit: 4, selector: 3, control: 9.
Instruction applied: unit: 4, selector: 3, control: 255.
The instruction is not valid.
Reseting to the instruction: unit: 4, selector: 3, control: 0.
Instruction applied: unit: 4, selector: 4, control: 1 0.
The instruction is not valid.
Instruction applied: unit: 4, selector: 4, control: 2 0.
The instruction is not valid.
Instruction applied: unit: 4, selector: 4, control: 3 0.
The instruction is not valid.
Instruction applied: unit: 4, selector: 4, control: 4 0.
The instruction is not valid.
Instruction applied: unit: 4, selector: 4, control: 5 0.
The instruction is not valid.
Instruction applied: unit: 4, selector: 4, control: 6 0.
The instruction is not valid.
Instruction applied: unit: 4, selector: 4, control: 7 0.
The instruction is not valid.
Instruction applied: unit: 4, selector: 4, control: 8 0.
The instruction is not valid.
Instruction applied: unit: 4, selector: 4, control: 9 0.
The instruction is not valid.
Instruction applied: unit: 4, selector: 4, control: 255 255.
The instruction is not valid.
Reseting to the instruction: unit: 4, selector: 4, control: 0 0.
Instruction applied: unit: 4, selector: 5, control: 1 0.
The instruction is not valid.
Instruction applied: unit: 4, selector: 5, control: 2 0.
The instruction is not valid.
Instruction applied: unit: 4, selector: 5, control: 3 0.
The instruction is not valid.
Instruction applied: unit: 4, selector: 5, control: 4 0.
The instruction is not valid.
[... 3484 lines cut ...]
                        Interval: Discrete 0.033s (30.000 fps)
/dev/video3
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

```

**Output of `ls -l /dev/v4l/by-path`**
```
total 0
lrwxrwxrwx. 1 root root 12 Jul 13 12:05 pci-0000:63:00.4-usb-0:1:1.0-video-index0 -> ../../video0
lrwxrwxrwx. 1 root root 12 Jul 13 12:05 pci-0000:63:00.4-usb-0:1:1.0-video-index1 -> ../../video1
lrwxrwxrwx. 1 root root 12 Jul 13 12:05 pci-0000:63:00.4-usb-0:1:1.2-video-index0 -> ../../video2
lrwxrwxrwx. 1 root root 12 Jul 13 12:05 pci-0000:63:00.4-usb-0:1:1.2-video-index1 -> ../../video3
lrwxrwxrwx. 1 root root 12 Jul 13 12:05 pci-0000:63:00.4-usbv2-0:1:1.0-video-index0 -> ../../video0
lrwxrwxrwx. 1 root root 12 Jul 13 12:05 pci-0000:63:00.4-usbv2-0:1:1.0-video-index1 -> ../../video1
lrwxrwxrwx. 1 root root 12 Jul 13 12:05 pci-0000:63:00.4-usbv2-0:1:1.2-video-index0 -> ../../video2
lrwxrwxrwx. 1 root root 12 Jul 13 12:05 pci-0000:63:00.4-usbv2-0:1:1.2-video-index1 -> ../../video3

```

**Additional info**
 - Computer (or camera) model: Lenovo Ideapad Pro 5 14AHP9
 - Linux distribution: Fedora 40 KDE Spin
 - Version of linux-enable-ir-emitter: 0x55e3a98b1a80

EmixamPP, 2024-07-13:
Execute `sudo linux-enable-ir-emitter tweak` and modify by selecting the number of the line that shows this information:
```
unit: 7
selector: 6
current:
  - 1
  - 3
  - 1
  - 0
  - 0
  - 0
  - 0
  - 0
  - 0
  ```
  With `1 3 2 0 0 0 0 0 0`, you should see your emitter blinking, I hope

eu-david, 2024-07-13:
it worked, the IR emitter started blinking, how do I save and generate the driver with this specific config?

EmixamPP, 2024-07-13:
Nothing special, once you made the change, you can exit the command and it is already saved

eu-david, 2024-07-13:
This way it works only once. After reboot it will start the IR emitter and will work only one time, if I try to initiate it again, it starts the camera without the IR emitter.

eu-david, 2024-07-13:
Also found that it will works for 1 more time after running the `linux-enable-ir-emitter run`. So it is either reboot or running the `run` command.

EmixamPP, 2024-07-13:
Could you show me the output of `sudo linux-enable-ir-emitter tweak` and `systemctl status linux-enable-ir-emitter`

eu-david, 2024-07-13:
`sudo linux-enable-ir-emitter tweak`

```
Tweaking the camera /dev/video2
Caution, you could break the camera.
0) unit: 4, selector: 2, control: 0 32
1) unit: 4, selector: 3, control: 0
2) unit: 4, selector: 4, control: 0 0
3) unit: 4, selector: 5, control: 255 0
4) unit: 4, selector: 6, control: 1
5) unit: 4, selector: 7, control: 1
6) unit: 4, selector: 8, control: 3
7) unit: 4, selector: 9, control: 0
8) unit: 4, selector: 12, control: 0
9) unit: 4, selector: 13, control: 0 0 0 0
10) unit: 4, selector: 14, control: 0 0 0 0
11) unit: 4, selector: 15, control: 149 40 89 134 33 105 2 0 1 255 65 255 0 0 0 0 0 0 16 0 0 0 0 1 52 132 195 138 158 90 33 85 103 168 67 15 68 70 142 223 152 153 46 81 78 162 199 245 149 98 203 245 68 71 149 229 0 0 0 1 0 3 240 0 0 0 0 1 52 132 195 138 158 90 33 85 103 168 67 15 68 70 142 223 152 153 46 81 78 162 199 245 149 98 203 245 68 71 149 229 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255
12) unit: 4, selector: 16, control: 0
13) unit: 4, selector: 17, control: 0 8 0 143 170 41 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
14) unit: 4, selector: 21, control: 0
15) unit: 4, selector: 22, control: 149 40 89 134 33 105 2 0 1 255 65 255 0 0 0 0 0 0 16 0 0 0 0 1 52 132 195 138 158 90 33 85 103 168 67 15 68 70 142 223 152 153 46 81 78 162 199 245 149 98 203 245 68 71 149 229 0 0 0 1 0 3 240 0
16) unit: 4, selector: 23, control: 0 0 0
17) unit: 4, selector: 25, control: 0
18) unit: 4, selector: 26, control: 0 0 0 0
19) unit: 4, selector: 27, control: 0 0 0 0
20) unit: 5, selector: 2, control: 0
21) unit: 5, selector: 4, control: 114 0
22) unit: 7, selector: 2, control: 1 0 0 0 0 0 0 96 97 2 0 0 0 0 0
23) unit: 7, selector: 3, control: 16 0 0 0 0 0 0 0 0 0 0
24) unit: 7, selector: 6, control: 1 3 2 0 0 0 0 0 0
[... cut ...]
Jul 13 17:22:49 fedora linux-enable-ir-emitter[5302]: Applying instruction unit: 7, selector: 6, control: 1 3 2 0 0 0 0 0 0 on /dev/v4l/by-path/pci-0000:63:00.4-usb-0:1:1.2-video-index0.
Jul 13 17:22:49 fedora linux-enable-ir-emitter[5302]: Applying instruction unit: 7, selector: 9, control: 0 0 0 0 on /dev/v4l/by-path/pci-0000:63:00.4-usb-0:1:1.2-video-index0.
Jul 13 17:22:49 fedora systemd[1]: linux-enable-ir-emitter.service: Deactivated successfully.
Jul 13 17:22:49 fedora systemd[1]: Finished linux-enable-ir-emitter.service - enable the infrared emitter.
```

EmixamPP, 2024-07-13:
> if I try to initiate it again, it starts the camera without the IR emitter.

Could you develop? Because everything looks good to me.
Is after restart you call `linux-enable-ir-emitter -d /dev/video2 test` multiple times, is the emitter blinks each times?

eu-david, 2024-07-14:
It starts the IR emitter only for 1 session, after which it starts just the camera without the IR emitter.

So if I run `linux-enable-ir-emitter -d /dev/video2 test` for the first time, it will work properly, the second and next times running the command will start just the camera, with no IR emitter blinking.

Also, running the `linux-enable-ir-emitter run` will reset this behavior so it does as described above, initiate IR emitter for one time and stop working for the following times.

EmixamPP, 2024-07-15:
Hum strange. And is it the same behaviour if you set the value to `1 3 3 0 0 0 0 0 0`?

eu-david, 2024-07-15:
When setting to `1 3 3 0 0 0 0 0 0` it does not start the IR emitter at all.

EmixamPP, 2024-07-15:
No one never opened an issue with that problem, so I couldn't really guess the solution, and if there exists one (using my tool).

So for the moment, I could only advise you to play with the `tweak` command, I just released it in the v6 for this kind of special case where a single instruction cannot make the IR emitter working properly.

Personally, I would try to modify the instructions: 1, 4, 5, 7, 12, 14, 17, 20, since they seem to be true 1/ false 0 condition (you can check this by selecting any of these instructions and see the minimum/maximum value). Maybe there is one instruction that enable / disable the reset of the camera instructions after each usage...

eu-david, 2024-07-15:
Thanks, I will play with it and report back if I manage to find the right config.

EmixamPP, 2024-07-15:
In case you want to reset everything to the default value, there a `.ini` file in `/etc/linux-enable-ir-emitter`, just overwritte the non-.ini variant with that content and then execute the `run` command to apply the changes.

eu-david, 2024-07-21:
Tried the `tweak` command for all the `true / false` instructions that I found but without any result. As a workaround is to have a cron job that runs the `linux-enable-ir-emitter run` command every few minutes.

Kundros, 2024-09-19:
I have similar issue. I have same laptop, with Fedora 40 and gnome. After setting the instruction to 1 3 2 0 0 0 0 0 0, it starts blinking, but after leaving the tweak and running test, then it blinks only one time and stops. Even after reboot it behaves the same.

CEN90, 2024-10-30:
I have this problem as well on my Lenovo Yoga Pro 7, running Fedora 41. I used `tweak` to set `1 3 2 0 0 0 0 0 0` and the ir emitters blinks once on the first run of `test`, but nothing after. To get it to blink again I have to turn off the laptop. This seems to me to be the same problem as #195.

EmixamPP, 2024-11-26:
I just want to let you know that I am working on a kernel space solution (instead of user space as currently), this should solve this kind of issue (I think).

CEN90, 2025-01-31:
I tested the latest version released a few days ago just now and it worked right away. My laptop feels complete now, thank you.

pako8x, 2025-08-19:
Same problem in 6.1.1 (Fedora gnome 42 on Lenovo Yoga pro 7 AMD AI). Tried everything possible but nothing happened.
I have to re-run the command 'run'  to make it work just once.
I am trying to solve with some sort of workaround to launch the service after every time I use the camera.

pako8x, 2025-08-19:
Thank you @EmixamPP , can you link me a simple procedure? I don't know where to start

EmixamPP, 2025-09-07:
I assume you are using it in combination with Howdy.

In all file returned by `grep -rl howdy /etc/pam.d`, add the following line before the one mentioning "howdy":
```
auth optional pam_exec.so /usr/local/bin/linux-enable-ir-emitter run
```

closed 2025-09-07 by EmixamPP