EmixamPP/linux-enable-ir-emitter #189: Error configuring ir on Wayland

Sniptzler, 2024-08-22:
**Describe the bug**
When trying to configure the ir emitter an error occurs.
```
terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.9.0) /home/runner/work/linux-enable-ir-emitter/linux-enable-ir-emitter/.github/workflows/deps/build/opencv/src/opencv/modules/highgui/src/window_gtk.cpp:638: error: (-2:Unspecified error) Can't initialize GTK backend in function 'cvInitSystem'
```

**How to reproduce**
Run sudo linux-enable-ir-emitter configure

**Additional info**
 - Computer (or camera) model:
 - Linux distribution: ArchLinux with hyprland and Wayland
 - Version of linux-enable-ir-emitter: 6.0.0

EmixamPP, 2024-08-22:
I think something is missing in your system.
Do you use the binary provided here on github or the AUR package?

You can use the `--no-gui` option to solve problem.

Sniptzler, 2024-08-23:
First I tried the AUR Package and got a different error. This one is with the provided binary. Now I know that my AUR package related error has to do with using sudo on GUIs with wayland. Not using the gui works perfectly, thanks.