EmixamPP/linux-enable-ir-emitter #190: Web Camera is broken

mgite, 2024-08-23:
**Ouput of `cat /var/local/log/linux-enable-ir-emitter/linux-enable-ir-emitter.log`**
```
```

**Output of `sudo linux-enable-ir-emitter tweak`**
```
there is no output
```

**Output of `v4l2-ctl --list-devices`**
```
SPCA2087 PC Camera: PC Camera (usb-0000:00:14.0-8):
        /dev/video0
        /dev/video1
        /dev/media0
```

**Ouput of `for dev in /dev/video*; do echo $dev && v4l2-ctl -d $dev --list-formats-ext; done`**
```
/dev/video0
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'YUYV' (YUYV 4:2:2)
                Size: Discrete 160x120
                        Interval: Discrete 0.033s (30.000 fps)
/dev/video1
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture
```

**Output of `ls -l /dev/v4l/by-path`**
```
lrwxrwxrwx 1 root root 12 Aug 23 15:12 pci-0000:00:14.0-usb-0:8:1.0-video-index0 -> ../../video0
lrwxrwxrwx 1 root root 12 Aug 23 15:12 pci-0000:00:14.0-usb-0:8:1.0-video-index1 -> ../../video1
lrwxrwxrwx 1 root root 12 Aug 23 15:12 pci-0000:00:14.0-usbv2-0:8:1.0-video-index0 -> ../../video0
lrwxrwxrwx 1 root root 12 Aug 23 15:12 pci-0000:00:14.0-usbv2-0:8:1.0-video-index1 -> ../../video1
```

**Give more information if you have**

After i installed this package and tried to configure /dev/video0 it did nothing just flashed webcam light. Then i tried /dev/video1 and it said opencv can not open /dev/video1. I was trying to make howdy work. After it did not work i tried rebooting laptop, also i shutdown and start it, but now the webcamera stopped working. I tried to remove this package but still when i try to use webcamera it's light just flashes like it did when i tried to configure it. I tried modprobe -r uvcvideo and then modprobe it  still does not work. This package broke something in my system. What could it be and how can i fix my webcamera>

**Additional info**
 - Computer (or camera) model: Thinkpad T480
 - Linux distribution: Arch Linux
 - Version of linux-enable-ir-emitter: Latest

mgite, 2024-08-23:
Just tried live ubuntu usb and it did same thing. Flashlight flashes and camera does not work. This package physically damaged my laptop WTF? Why there is no warning about this possibility?

EmixamPP, 2024-09-01:
I see you do not give the output logs from /var/local/log/linux-enable-ir-emitter/linux-enable-ir-emitter.log. So it is hard for me to understand what exactly happened. 

Note the camera is probably not physically damaged, it seems its firmware is in a wrong state.

mgite, 2024-09-01:
> I see you do not give the output logs from /var/local/log/linux-enable-ir-emitter/linux-enable-ir-emitter.log. So it is hard for me to understand what exactly happened.
> 
> Note the camera is probably not physically damaged, it seems its firmware is in a wrong state.

Yes quickly tried to get that log but file was not there. Here it is

```
 ~ cat /var/log/linux-enable-ir-emitter/linux-enable-ir-emitter.log
[14:25:39] [info] Stand in front of and close to the camera and make sure the room is well lit.
[14:25:39] [info] Ensure to not use the camera during the execution.
[14:25:40] [critical] No infrared camera has been found.
[14:25:44] [info] Stand in front of and close to the camera and make sure the room is well lit.
[14:25:44] [info] Ensure to not use the camera during the execution.
[14:25:44] [critical] No infrared camera has been found.
[14:25:54] [info] Stand in front of and close to the camera and make sure the room is well lit.
[14:25:54] [info] Ensure to not use the camera during the execution.
[14:25:54] [critical] No infrared camera has been found.
[14:25:58] [info] Stand in front of and close to the camera and make sure the room is well lit.
[14:25:58] [info] Ensure to not use the camera during the execution.
[14:25:59] [critical] No infrared camera has been found.
[14:26:03] [info] Stand in front of and close to the camera and make sure the room is well lit.
[14:26:03] [info] Ensure to not use the camera during the execution.
[14:26:04] [critical] No infrared camera has been found.
[14:30:18] [info] Stand in front of and close to the camera and make sure the room is well lit.
[14:30:18] [info] Ensure to not use the camera during the execution.
[14:30:18] [critical] No infrared camera has been found.
[14:30:31] [info] Stand in front of and close to the camera and make sure the room is well lit.
[14:30:31] [info] Ensure to not use the camera during the execution.
[14:30:32] [critical] No infrared camera has been found.
[14:38:10] [info] Stand in front of and close to the camera and make sure the room is well lit.
[14:38:10] [info] Ensure to not use the camera during the execution.
[14:38:10] [critical] No infrared camera has been found.
[14:38:30] [info] Tweaking the camera /dev/video1
[14:38:30] [info] Caution, you could break the camera.
[14:38:31] [critical] OpenCV cannot access to /dev/video1
[14:38:39] [info] Tweaking the camera /dev/video1
[14:38:39] [info] Caution, you could break the camera.
[14:38:39] [critical] OpenCV cannot access to /dev/video1
[14:44:29] [info] Stand in front of and close to the camera and make sure the room is well lit.
[14:44:29] [info] Ensure to not use the camera during the execution.
[14:44:29] [critical] OpenCV cannot access to /dev/video1
[14:44:54] [info] Stand in front of and close to the camera and make sure the room is well lit.
[14:44:54] [info] Ensure to not use the camera during the execution.
[14:44:54] [critical] OpenCV cannot access to /dev/video1
[14:51:37] [info] Stand in front of and close to the camera and make sure the room is well lit.
[14:51:37] [info] Ensure to not use the camera during the execution.
[14:51:37] [critical] OpenCV cannot access to /dev/video1
[15:25:04] [info] Tweaking the camera /dev/video1
[15:25:04] [info] Caution, you could break the camera.
[15:25:04] [critical] OpenCV cannot access to /dev/video1
```

EmixamPP, 2024-09-01:
Hum, you use the package from AUR. I see it also removed the log when you uninstalled the package, so what is inside is pointless now.

Unfortunately, unless contacting the camera vendor/producer, to get the firmware in order to reset, I'm not sure to be able to do something for you, but I will try my best.

Do you still have the file which is inside /etc/linux-enable-ir-emitter ?

mgite, 2024-09-01:
Yes there are 2 files

pci-0000:00:14.0-usb-0:8:1.0-video-index0.ini  pci-0000:00:14.0-usb-0:8:1.0-video-index1

EmixamPP, 2024-09-01:
Could you copy paste their content?

mgite, 2024-09-01:
pci-0000\:00\:14.0-usb-0\:8\:1.0-video-index1
[]

pci-0000:00:14.0-usb-0:8:1.0-video-index0.ini
https://pastebin.com/NdWcv87y

mgite, 2024-09-01:
I was searching for fix and i noticed in this video on 3:26 https://youtu.be/d_hucSz3mQI?si=Hlcv8-GeskPfyyVN&t=206

There is RESET button beside camera in Lenovo vantage on windows. Maybe that will fix it? But i have not used windows in about 15 years already. I'll have to replace ssd and install windows first.

EmixamPP, 2024-09-02:
> But i have not used windows in about 15 years already. I'll have to replace ssd and install windows first.

You can use a VM

TuzelKO, 2024-10-22:
So... I have a problem and the log file...

```
[21:02:39] [debug] Executing configure command.
[21:02:39] [info] Stand in front of and close to the camera and make sure the room is well lit.
[21:02:39] [info] Ensure to not use the camera during the execution.
[21:02:39] [debug] Checking if /dev/video0 is a grayscale camera.
[21:02:41] [debug] Checking if /dev/video1 is a grayscale camera.
[21:02:41] [debug] Checking if /dev/video2 is a grayscale camera.
[21:02:44] [debug] /dev/video2 is a grayscale camera.
[21:02:44] [info] Configuring the camera /dev/video2.
[21:02:44] [debug] yaml error: bad file: /etc/linux-enable-ir-emitter/pci-0000:64:00.4-usb-0:1:1.2-video-index0.ini.
[21:02:44] [debug] No previous configuration found.
[21:02:44] [debug] Configuration for /dev/video2 saved here: /etc/linux-enable-ir-emitter/pci-0000:64:00.4-usb-0:1:1.2-video-index0.ini.
[21:03:06] [debug] Instruction applied: unit: 4, selector: 2, control: 1 0.
[21:03:08] [debug] The instruction does not enable the emitter.
[21:03:08] [debug] Instruction applied: unit: 4, selector: 2, control: 2 0.
[21:03:10] [debug] The instruction does not enable the emitter.
[21:03:10] [debug] Instruction applied: unit: 4, selector: 2, control: 3 0.
[21:03:11] [debug] The instruction does not enable the emitter.
[21:03:11] [debug] Instruction applied: unit: 4, selector: 2, control: 4 0.
[21:03:13] [debug] The instruction does not enable the emitter.
[21:03:13] [debug] Instruction applied: unit: 4, selector: 2, control: 5 0.
[21:03:14] [debug] The instruction does not enable the emitter.
[21:03:14] [debug] Instruction applied: unit: 4, selector: 2, control: 6 0.
[21:03:16] [debug] The instruction does not enable the emitter.
[21:03:16] [debug] Instruction applied: unit: 4, selector: 2, control: 7 0.
[21:03:17] [debug] The instruction does not enable the emitter.
[21:03:17] [debug] Instruction applied: unit: 4, selector: 2, control: 8 0.
[21:03:19] [debug] The instruction does not enable the emitter.
[21:03:19] [debug] Instruction applied: unit: 4, selector: 2, control: 9 0.
[21:03:21] [debug] The instruction does not enable the emitter.
[21:03:21] [debug] Instruction applied: unit: 4, selector: 2, control: 255 255.
[21:03:22] [debug] The instruction does not enable the emitter.
[21:03:22] [debug] Reseting to the instruction: unit: 4, selector: 2, control: 0 32.
[21:03:22] [debug] Instruction applied: unit: 4, selector: 3, control: 1.
[21:04:47] [info] The process will exit as soon as possible.
[21:07:46] [info] The process will exit as soon as possible.
[21:07:47] [info] The process will exit as soon as possible.
[21:07:48] [info] The process will exit as soon as possible.
[21:07:48] [info] The process will exit as soon as possible.
[21:07:48] [debug] The instruction does not enable the emitter.
[21:07:48] [debug] Reseting to the instruction: unit: 4, selector: 3, control: 0.
[21:07:48] [error] The configuration failed.
[21:07:48] [info] Please retry in manual mode by adding the '-m' option.
[21:07:48] [info] Do not hesitate to visit the GitHub!
[21:07:48] [info] https://github.com/EmixamPP/linux-enable-ir-emitter/blob/master/docs/README.md
[21:07:57] [debug] Executing configure command.
[21:07:57] [info] Stand in front of and close to the camera and make sure the room is well lit.
[21:07:57] [info] Ensure to not use the camera during the execution.
[21:07:57] [debug] Checking if /dev/video0 is a grayscale camera.
[21:07:59] [debug] Checking if /dev/video1 is a grayscale camera.
[21:07:59] [debug] Checking if /dev/video2 is a grayscale camera.
[21:08:02] [debug] /dev/video2 is a grayscale camera.
[21:08:02] [info] Configuring the camera /dev/video2.
[21:08:02] [debug] Previous configuration found.
[21:08:05] [debug] Instruction applied: unit: 4, selector: 2, control: 1 0.
[21:08:06] [debug] The instruction does not enable the emitter.
[21:08:06] [debug] Instruction applied: unit: 4, selector: 2, control: 2 0.
[21:08:08] [debug] The instruction does not enable the emitter.
[21:08:08] [debug] Instruction applied: unit: 4, selector: 2, control: 3 0.
[21:08:09] [debug] The instruction does not enable the emitter.
[21:08:09] [debug] Instruction applied: unit: 4, selector: 2, control: 4 0.
[21:08:11] [debug] The instruction does not enable the emitter.
[21:08:11] [debug] Instruction applied: unit: 4, selector: 2, control: 5 0.
[21:08:13] [debug] The instruction does not enable the emitter.
[21:08:13] [debug] Instruction applied: unit: 4, selector: 2, control: 6 0.
[21:08:14] [debug] The instruction does not enable the emitter.
[21:08:14] [debug] Instruction applied: unit: 4, selector: 2, control: 7 0.
[21:08:16] [debug] The instruction does not enable the emitter.
[... 344 lines cut ...]
[22:19:31] [debug] The instruction does not enable the emitter.
[22:19:31] [debug] Instruction applied: unit: 4, selector: 13, control: 9 0 0 0.
[22:19:31] [debug] The instruction does not enable the emitter.
[22:19:31] [debug] Instruction applied: unit: 4, selector: 13, control: 255 255 255 0.
[22:19:31] [debug] The instruction does not enable the emitter.
[22:19:31] [debug] Reseting to the instruction: unit: 4, selector: 13, control: 0 0 0 0.
[22:19:31] [debug] Instruction applied: unit: 4, selector: 14, control: 1 0 0 0.
[22:19:31] [debug] The instruction does not enable the emitter.
[22:19:31] [debug] Instruction applied: unit: 4, selector: 14, control: 2 0 0 0.
[22:19:31] [debug] The instruction does not enable the emitter.
[... cut ...]
[22:19:31] [debug] The instruction does not enable the emitter.
[22:19:31] [debug] Instruction applied: unit: 4, selector: 14, control: 6 0 0 0.
[22:19:31] [debug] The instruction does not enable the emitter.
[22:19:31] [debug] Instruction appli
( ...App crash and restart laptop... )
[22:39:45] [debug] Executing configure command.
[22:39:45] [info] Stand in front of and close to the camera and make sure the room is well lit.
[22:39:45] [info] Ensure to not use the camera during the execution.
[22:39:45] [critical] No infrared camera has been found.
```

EmixamPP, 2024-10-22:
Thanks for the log, I will investigate a possible solution

TuzelKO, 2024-10-22:
After turning off the laptop and turning it on again, it came back. Several reboots before that did not help. It was necessary to turn it off and on again.

igloo861, 2025-01-10:
Fix on Lenovo laptop

Hi, just had the same problem. I left the configuration running unattended for a couple of hours.  When i came back the terminal said "Segmentation fault (core dumped) and the camera no longer appeared when i tried to test it. 

What i fixed by turning the computer off and starting it with the NOVO button on my Lenovo laptop and choosing system recovery. After the reboot the camera reappeared and functioned fine.

Pleas note that i booted to Windows while using the recovery function. I don't know how Lenovo's recovery tool works, so i don't know if booting to Linux makes a difference.

LainerDonet, 2026-08-16:
I want to report a very similar case on the same laptop model, which may help narrow down what's causing this.

**Hardware/software:**
- ThinkPad T480
- Ubuntu 24.04.4 LTS, GNOME 46, Wayland
- linux-enable-ir-emitter 6.1.2 (systemd release variant)

**What happened:**

The first time I ran `sudo linux-enable-ir-emitter configure`, the process completed and appeared to work correctly. On the second run, I killed the process manually (`kill`) partway through. Right after that, I noticed the webcam's white LED stayed permanently on — it did not turn off on its own, and it stayed on through a normal reboot. It only turned off after a full cold power-off (holding the power button until the system powered down completely).

After that point, `v4l2-ctl --list-devices` started showing the IR camera under a different identity:

**Before (working correctly):**

Integrated IR Camera: Integrate (usb-0000:00:14.0-5):
/dev/video0
/dev/video1
/dev/media0

**After (current state):**

SPCA2085 PC Camera: PC Camera (usb-0000:00:14.0-5):
/dev/video0
/dev/video1
/dev/media0

Same USB bus location, different reported name — no longer recognized as the IR camera. A camera app that used to show the IR feed in grayscale no longer shows it at all.

**Recovery attempts that did NOT work:**
1. Normal reboot
2. BIOS toggle: Security → I/O Port Access → Integrated Camera → Disable, save/reboot, then re-enable, save/reboot
3. Firmware check via `fwupdmgr` — already up to date, nothing to apply
4. Full power drain: shutdown, unplug AC, remove the removable battery, hold power button ~30s, reinsert battery, cold boot

None of these restored the original device identification.

This matches the warning in the tool about not killing the process mid-run ("you could break the camera") — in my case, killing it seems to have left the camera's onboard controller in a bad state that persists across both software and hardware-level resets. Posting this in case it helps correlate with other reports, or in case anyone has found a way to recover from this state.

Shlok-Bhakta, 2026-08-16:
@LainerDonet This literally just happened to me. I did the exact same thing. if anyone with a firmware dump of their working camera could send it, that would be awesome!

Shlok-Bhakta, 2026-08-16:
Same class of failure as this issue: interrupted configure on a ThinkPad RGB camera, device becomes `1BCF:0B13` / `SPCA2087 PC Camera` / 160×120 only.

On a T480s here, two independent **read-only** 64 KiB SPI dumps (UVC XU selector 14 = read address, selector 22 = 64-byte read; **never** 9/10/12) are byte-identical. Sector 0 (`0x0000–0x0FFF`) is erased `FF`. From `0x1000` onward the original OEM image is still there, including the `04F2:B614` descriptor. I am not asking anyone to flash firmware, and I will not write anything to volunteer hardware.

I need a **read-only 64 KiB SPI dump from a known-good `04F2:B614`**.

```text
BEST:              T480s + 04F2:B614 + FRU 01HW038 / 01HW039
STILL VERY USEFUL: any T480s + 04F2:B614
USEFUL:            T480 / X280 / X380 / X1 Carbon 6th / etc. + 04F2:B614
```

Zero-risk check (stop if no match):

```bash
lsusb -nn | grep -i '04f2:b614'
```

If that hits, [`dump-b614-readonly.py`](https://gist.github.com/Shlok-Bhakta/7df42a1cdab03d1c0f095a2005b8dd5e) (plain Python, no extra packages):

1. Refuses every VID:PID except `04F2:B614` (including IR `04F2:B615`).
2. `--check` identifies the camera from sysfs and exits without any UVC XU ioctl.
3. Dump path uses only SET_CUR selector 14 (SPI *read* pointer) and GET_CUR selector 22 (64-byte read). No ISP enable, no erase, no write, no finish.
4. Dumps 64 KiB twice and will not emit a zip unless both reads match.
5. Zip contains both dumps, SHA256, and `lsusb -v`.

```bash
curl -fsSL -o dump-b614-readonly.py \
  https://gist.githubusercontent.com/Shlok-Bhakta/7df42a1cdab03d1c0f095a2005b8dd5e/raw/dump-b614-readonly.py
less dump-b614-readonly.py

python3 dump-b614-readonly.py --check
python3 dump-b614-readonly.py
```

Please attach the zip and mention:

- ThinkPad model
- the `lsusb -nn` line
- camera FRU only if you already know it (do not open the laptop)

Thank you.

kaovilai, 2026-09-01:
Same failure on a different OEM (HP, not Lenovo) — plus a real lead on the write protocol

Wanted to add a data point since this exact bug clearly isn't Lenovo-specific — it hit my HP Spectre x360 15-bl0XX (`HP TrueVision FHD RGB-IR`, originally `064E:3401`, manufacturer string "Suyin Corp.") the same way: ran `linux-enable-ir-emitter configure`, it hung (turned out two copies were running concurrently in separate terminals), killed both, and the camera has been stuck ever since — same `SPCA2085 PC Camera` (`1bcf:0b09`) fallback identity everyone else here is seeing.

**Diagnostic data point:** I retargeted @Shlok-Bhakta's read-only dump script (https://gist.github.com/Shlok-Bhakta/7df42a1cdab03d1c0f095a2005b8dd5e) at my own device's `1bcf:0b09` identity (same read-only allowlist, unmodified — just changed the VID/PID it checks for). Result: **sector 0 (0x0000-0x0FFF) is fully erased** (4096/4096 bytes `0xFF`), but **sectors 0x1000-0x8000 (~32KB) still contain non-erased data** — the application firmware payload itself looks intact; only the boot/vector sector pointing to it is gone. SHA256 of the 64KB dump: `ceb0d7510e1f16b9d77bffb9a9e1fa3a7f9a2a3234e82512a9dccd50157df926`. Happy to share the full dump if useful.

**Possible path forward — fwupd's `sunplus-camera` plugin:** I found that `fwupd` (the official Linux firmware-update daemon) merged a plugin for this exact chip family in March 2026 (https://github.com/fwupd/fwupd/tree/main/plugins/sunplus-camera, by @snizmowski / Sean Rhodes @ Star Labs). Its README explicitly names our exact failure state: *"It does not support recovery-state devices that enumerate as `1bcf:0b09`."* — independent confirmation this is a known, generic Sunplus-chip-family identity, not OEM-specific.

More importantly, the plugin's source (`fu-sunplus-camera-device.c`, `fu-sunplus-camera.rs`) documents a complete, production-tested UVC Extension Unit ISP protocol on Unit 4:

```
selector 2/3  = ASIC register select/data (16-bit register, 8-bit value)
selector 9    = enable ISP/download mode
selector 10   = stream firmware payload (chunk size is queried from the device, not hardcoded)
selector 11   = XOR checksum verify (checksum includes 0xFF padding on the final chunk)
selector 12   = finish/finalize
selector 14   = set read address (matches what the read-only dump script already uses safely)
selector 22   = read chunk (ditto)
ASIC reg 0x2501 = download-state flag, cleared before a write if nonzero
soft-reset sequence: reg 0x2034=0x00, 0x2030=0xFF, 0x2031=0x00, 0x2032=0x00, 0x2033=0x02, 0x2034=0x01
```

Shlok-Bhakta, 2026-09-07:
more vibe slop for everyone. I replaced my camera and then ripped the firmware myself. I got a base64 encoded gist of the firmware here https://gist.github.com/Shlok-Bhakta/5334482f8affd25b9e04d6b5b5b83dd5. since most yall be using agents anyway you should be able to get it from here. Just note I am using a Thinkpad T480s and I am not responsible if you break things further (a replacement camera module is also only $20 on ebay so no harm rlly) how to get the dump and verify:

```
curl -fsSL -o dump-a.bin.b64 \
  https://gist.githubusercontent.com/Shlok-Bhakta/5334482f8affd25b9e04d6b5b5b83dd5/raw/dump-a.bin.b64
base64 -d dump-a.bin.b64 > dump-a.bin
echo bab4aa48a13ce02267e44501e5c66aa810172c5d5df5a3f96fcee9080ad7cc75  dump-a.bin | sha256sum -c
```

I think your agents should be able to take it from here. I suggest telling your agent to dump the firmware of your own camera using the dump script https://gist.github.com/Shlok-Bhakta/7df42a1cdab03d1c0f095a2005b8dd5e and then comparing if they are similar if they dont be similar then dont flash this lol. hopefully someone can save a few bucks and get the cam module working again. Good luck!

Netsrot1150, 2026-10-06:
Another HP data point, same module as @kaovilai: **HP Spectre x360 13-ac0XX** (board 827E, BIOS F.52), Ubuntu 26.04, kernel 7.0, `linux-enable-ir-emitter` 6.1.2 (systemd release tarball).

The camera was `064e:3401` **HP TrueVision FHD RGB-IR** (manufacturer "SuYin", serial `HF2066-P98B-OV07-REV0101`, bcdDevice 0.11). `configure` was run once (not killed, not run twice). About 40 s later the camera dropped off the bus and came back as the `1bcf:0b09` fallback:

```
01:16:44 kernel: usb 1-5: New USB device found, idVendor=064e, idProduct=3401, bcdDevice= 0.11
01:16:44 kernel: usb 1-5: Product: HP TrueVision FHD RGB-IR
01:16:44 kernel: usb 1-5: Manufacturer: SuYin
01:18:48 sudo: COMMAND=/usr/local/bin/linux-enable-ir-emitter configure
01:19:27 kernel: uvcvideo 1-5:1.0: UVC non compliance: GET_CUR control 10 on unit 4 returned 1 bytes when we expected 1024.
01:19:27 kernel: usb 1-5: USB disconnect, device number 2
01:19:27 kernel: uvcvideo 1-5:1.1: Failed to resubmit video URB (-19).
01:19:28 kernel: usb 1-5: New USB device found, idVendor=1bcf, idProduct=0b09, bcdDevice= 1.00
01:19:28 kernel: usb 1-5: Product: SPCA2085 PC Camera
01:19:28 kernel: usb 1-5: Manufacturer: SunplusIT Inc
01:19:28 kernel: usb 1-5: SerialNumber: 01.00.00
```

The last logged access before the disconnect was a `GET_CUR` on unit 4, selector 10 (0x0A). In fwupd's `sunplus-camera` plugin, that selector is `Access`, which the plugin uses for the firmware payload transfer. The kernel doesn't log `SET_CUR`, so I can't tell which write actually caused it. Still, the tool's control scan seems to reach the ISP/flash selectors on unit 4. It might be worth excluding unit 4 selectors 0x09–0x0C (or the whole XU GUID `63610682-5070-49ab-b8cc-b3855e8d221d`) from the scan for this chip family.

Current state, the same as everyone else here:
- `1bcf:0b09` with one VideoStreaming interface and only YUYV 160x120@30. The RGB camera is gone too.
- `STREAMON` succeeds and bulk IN transfers on EP 0x81 are submitted, but no data arrives (confirmed with usbmon).
- V4L2 control ranges are garbage (e.g. brightness min -15011 / max 26419 / step -17217).
- XU unit 4, selector 22: `GET_INFO` = 0x03, `GET_CUR` all `0xFF`, and `GET_MIN`/`GET_MAX`/`GET_RES` return 1 byte (0x00 / 0xFF / 0x01).
- Windows (generic `usbvideo.inf`) sees it as well, but it produces no image there either.
- A full cold power-off (AC unplugged, power button held for 30 s) did **not** help, and neither did several reboots.

@kaovilai, since your 15-bl0XX has the same SuYin `064e:3401` module: did you get anywhere with restoring it? If anyone has a dump from a *working* `064e:3401` (HP TrueVision FHD RGB-IR), that would probably help several HP Spectre owners here. I haven't written anything to the device myself.

kaovilai, 2026-10-07:
nope

wonderfulyeah, 2026-10-08:
# [HELP] ThinkPad X1 Yoga Gen 3 — RGB camera firmware lost (Sunplus SPCA208x, stuck in 1bcf:0b13 ROM mode). Need a read-only flash dump from a healthy unit

## TL;DR

If your ThinkPad's **RGB camera works** and it's the Sunplus module
(**FRU 01HW040** family — X1 Yoga Gen 3 / X1 Carbon 6th / X280 / X380 Yoga /
L380 / L390 Yoga / L480 / P1), please run a **strictly read-only** dump
script on any Linux live USB and send me the resulting `donor-dump.bin`:

```bash
sudo python3 donor-dump-probe.py        # https://github.com/wonderfulyeah/sunplus-camera-rescue
```

You do **not** need to know your camera's USB ID — the script discovers it.
It cannot write to the camera (the flash-write controls are hard-blocked in
code) and it reads every dump twice, keeping results only if both passes
match. A replacement module for these models costs ~$20, so the risk to a
donor is essentially zero, but the dump would save me the same cost and a
lot of e-waste.

## The situation

My X1 Yoga Gen 3 (Type 20LE) has the dual-sensor camera bar (FRU 01HW040 —
RGB + IR). Forensic reads over the vendor UVC extension-unit protocol
(confirmed against fwupd's `sunplus-camera` plugin) established:

- The bar carries **two independent controllers, each with its own 64 KiB
  SPI flash**:
  - IR: SPCA2093B → healthy, enumerates as `5986:211a`, fw version 5465
    (restored with Lenovo's official *Integrated IR Camera Firmware Update
    Utility*, ds505129).
  - RGB: SPCA2087-family → **bricked**, stuck in the mask-ROM state
    `1bcf:0b13` "SPCA2087 PC Camera".
- The RGB controller's flash currently contains a mis-flashed copy of the
  **IR** image (byte-for-byte `sf.bin` from the IR package except one flag
  byte), i.e. **the original RGB firmware is gone** from the chip.
- Lenovo never published an RGB firmware for this generation. I checked:
  Lenovo support channels (chat bot loop, China support → "no download,
  use a repair center"), the Microsoft Update Catalog, LVFS/fwupd metadata
  (all 850 components, text + GUID search), driver packs (the Sunplus INFs
  don't even list this module). Nothing.
- The camera's healthy USB identity was never recorded on my install
  (X1 Yoga Gen 3 units came with either a Chicony RGB camera — the
  `04f2:b61e` on ArchWiki — or this Sunplus one), so I can't even screen
  donors by ID. The probe script handles that automatically.

## What I'm asking

Run this on any machine with a **working** RGB camera on the 01HW040-family
module (Linux live USB is fine, no OS changes):

```bash
git clone https://github.com/wonderfulyeah/sunplus-camera-rescue
cd sunplus-camera-rescue
python3 donor-dump-probe.py --check    # inventory only, no camera I/O yet
sudo python3 donor-dump-probe.py       # read-only probe + dump
```

Then send me `donor-dump.bin` and the `lsusb -nn` line of your machine
(either directly or via the GitHub repo). Units with the Chicony RGB
variant (`04f2:b61e`) will simply fail validation ("no Sunplus firmware
header") — nothing breaks, you just can't donate. Only Sunplus-variant
machines produce a valid image.

## Why this is safe for donors

- The only USB transfers the script can emit are the vendor's flash-read
  pair (`SET_CUR sel 0x0E` seek + `GET_CUR sel 0x16` 64-byte read) on
  extension unit 4 — the same pair the vendor flasher uses to *verify*
  programming, and the same pair fwupd's official plugin uses.
- Flash-write controls (selectors 9/10/11/12) and register writes (2/3)
  are hard-blocked before any ioctl; a coding mistake raises instead of
  transmitting.
- Target allowlists skip all known IR cameras and all known ROM states.
- The dump contains only the camera controller's firmware/calibration
  blob — a webcam flash holds no user data.

## If you can't run the script

Boosting this thread helps. Also useful: if you own a bricked
`1bcf:0b13` camera yourself, `brick-forensics-0b13.py` (same repo,
read-only) reads your module's flash so we can compare damage patterns.

---

*Once I receive a validated dump I'll flash it back over the same UVC
interface (write path per the fwupd plugin: selectors 9/10/11/12 + XOR
checksum + verify) and document the whole procedure in the repo, so the
next person with a bricked 01HW040 RGB camera can self-serve.*

---

## Short version (for replying inside existing threads)

> Looking for a read-only 64 KiB firmware dump from a **healthy** Sunplus
> RGB camera (FRU 01HW040 family: X1 Yoga Gen 3 / X1C6 / X280 / X380 /
> L380 / L390 / P1) to recover a bricked X1Y3 camera stuck in ROM mode
> (`1bcf:0b13`). One command on a live USB:
> `sudo python3 donor-dump-probe.py` from
> https://github.com/wonderfulyeah/sunplus-camera-rescue — strictly
> read-only (write selectors hard-blocked), double-read verified, no
> disassembly. Send back `donor-dump.bin` + your `lsusb -nn` line. Thanks!

deviant, 2026-10-10:
Unsubscribing from this thread since it’s just clankers going back and forth begging for firmware dumps at this point. Seriously guys, it’s not that hard to write your own comment, and it’s definitely a hell of a lot more considerate than splurging word-vomit into everyone’s inbox. And emoji reacts exist for a reason! Use those instead of dumping a 10k word “me too” into the thread next time, please…