EmixamPP/linux-enable-ir-emitter #193: 6.0.3 fails to build

iyanmv, 2024-09-06:
**Describe the bug**
Hi,

Latest release (6.0.3) fails to build. I tried to build it exactly [the same way as 6.0.0](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=linux-enable-ir-emitter) and these are the errors I get:

```
==> Starting build()...
+ exec meson setup --prefix /usr --libexecdir lib --sbindir bin --buildtype plain --auto-features enabled --wrap-mode nodownload -D b_pie=true -D python.bytecompile=1 linux-enable-ir-emitter-6.0.3 build
The Meson build system
Version: 1.5.1
Source dir: /build/linux-enable-ir-emitter/src/linux-enable-ir-emitter-6.0.3
Build dir: /build/linux-enable-ir-emitter/src/build
Build type: native build
Project name: linux-enable-ir-emitter
Project version: 6.0.3
C++ compiler for the host machine: c++ (gcc 14.2.1 "c++ (GCC) 14.2.1 20240805")
C++ linker for the host machine: c++ ld.bfd 2.43.0
Host machine cpu family: x86_64
Host machine cpu: x86_64
Found pkg-config: YES (/usr/bin/pkg-config) 2.1.1
Run-time dependency opencv4 found: YES 4.10.0
Run-time dependency gtk+-3.0 found: YES 3.24.43
Run-time dependency yaml-cpp found: YES 0.8.0
Run-time dependency argparse found: YES 3.1.0
Run-time dependency spdlog found: YES 1.14.1
Configuring linux-enable-ir-emitter.cpp using configuration
Configuring configuration.hpp using configuration
Configuring logger.hpp using configuration
Build targets in project: 1

linux-enable-ir-emitter 6.0.3

  User defined options
    auto_features     : enabled
    buildtype         : plain
    libexecdir        : lib
    prefix            : /usr
    sbindir           : bin
    wrap_mode         : nodownload
    python.bytecompile: 1
    b_pie             : true

Found ninja-1.12.1 at /usr/bin/ninja
ninja: Entering directory `/build/linux-enable-ir-emitter/src/build'
[1/13] Compiling C++ object src/linux-enable-ir-emitter.p/camera_autocamera.cpp.o
[2/13] Compiling C++ object src/linux-enable-ir-emitter.p/configuration_finder.cpp.o
FAILED: src/linux-enable-ir-emitter.p/configuration_finder.cpp.o 
c++ -Isrc/linux-enable-ir-emitter.p -Isrc -I../linux-enable-ir-emitter-6.0.3/src -I/usr/include/gtk-3.0 -I/usr/include/pango-1.0 -I/usr/include/cairo -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/atk-1.0 -I/usr/include/freetype2 -I/usr/include/glib-2.0 -I/usr/lib/glib-2.0/include -I/usr/include/harfbuzz -I/usr/include/libmount -I/usr/include/blkid -I/usr/include/libpng16 -I/usr/include/pixman-1 -I/usr/include/cloudproviders -I/usr/include/at-spi2-atk/2.0 -I/usr/include/at-spi-2.0 -I/usr/include/dbus-1.0 -I/usr/lib/dbus-1.0/include -I/usr/include/fribidi -I/usr/include/sysprof-6 -I/usr/include/gio-unix-2.0 -fdiagnostics-color=always -D_GLIBCXX_ASSERTIONS=1 -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -Wpedantic -Wcast-qual -Wconversion -Wfloat-equal -Wformat=2 -Winline -Wmissing-declarations -Wredundant-decls -Wshadow -Wundef -Wuninitialized -Wwrite-strings -Wdisabled-optimization -Wpacked -Wpadded -Wmultichar -Wswitch-default -Wswitch-enum -Wunused-macros -Wmissing-include-dirs -Wunsafe-loop-optimizations -Wstack-protector -Wstrict-overflow=5 -Warray-bounds=2 -Wlogical-op -Wstrict-aliasing=3 -Wvla -Wdouble-promotion -Wsuggest-attribute=const -Wsuggest-attribute=noreturn -Wsuggest-attribute=pure -Wtrampolines -Wvector-operation-performance -Wsuggest-attribute=format -Wdate-time -Wformat-signedness -Wnormalized=nfc -Wduplicated-cond -Wnull-dereference -Wshift-negative-value -Wshift-overflow=2 -Wunused-const-variable=2 -Walloca -Walloc-zero -Wformat-overflow=2 -Wformat-truncation=2 -Wstringop-overflow=3 -Wduplicated-branches -Wcast-align=strict -Wsuggest-attribute=cold -Wsuggest-attribute=malloc -Wattribute-alias=2 -Wanalyzer-too-complex -Warith-conversion -Wbidi-chars=ucn -Wopenacc-parallelism -Wtrivial-auto-var-init -Wctor-dtor-privacy -Weffc++ -Wnon-virtual-dtor -Wold-style-cast -Woverloaded-virtual -Wsign-promo -Wstrict-null-sentinel -Wnoexcept -Wzero-as-null-pointer-constant -Wabi-tag -Wuseless-cast -Wconditionally-supported -Wsuggest-final-methods -Wsuggest-final-types -Wsuggest-override -Wmultiple-inheritance -Wplacement-new=2 -Wvirtual-inheritance -Waligned-new=all -Wnoexcept-type -Wregister -Wcatch-value=3 -Wextra-semi -Wdeprecated-copy-dtor -Wredundant-move -Wcomma-subscript -Wmismatched-tags -Wredundant-tags -Wvolatile -Wdeprecated-enum-enum-conversion -Wdeprecated-enum-float-conversion -Winvalid-imported-macros -Werror -std=c++20 -Wno-unknown-warning-option -Wno-padded -Wno-shadow -Wno-abi-tag -Wno-effc++ -Wno-unused-const-variable -Wno-suggest-attribute=const -Wno-inline -Wno-suggest-attribute=pure -Wno-suggest-final-types -Wno-suggest-final-methods -march=x86-64 -mtune=generic -O2 -pipe -fno-plt -fexceptions -Wp,-D_FORTIFY_SOURCE=3 -Wformat -Werror=format-security -fstack-clash-protection -fcf-protection -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -Wp,-D_GLIBCXX_ASSERTIONS -g -ffile-prefix-map=/build/linux-enable-ir-emitter/src=/usr/src/debug/linux-enable-ir-emitter -flto=auto -fPIE -DSPDLOG_SHARED_LIB -DSPDLOG_COMPILED_LIB -DSPDLOG_FMT_EXTERNAL -pthread -isystem/usr/include/opencv4 -MD -MQ src/linux-enable-ir-emitter.p/configuration_finder.cpp.o -MF src/linux-enable-ir-emitter.p/configuration_finder.cpp.o.d -o src/linux-enable-ir-emitter.p/configuration_finder.cpp.o -c ../linux-enable-ir-emitter-6.0.3/src/configuration/finder.cpp
In file included from ../linux-enable-ir-emitter-6.0.3/src/configuration/finder.cpp:10:
src/logger.hpp: In instantiation of ‘void logger::debug(std::string_view, Args&& ...) [with Args = {std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >}; std::string_view = std::basic_string_view<char>]’:
../linux-enable-ir-emitter-6.0.3/src/configuration/finder.cpp:44:20:   required from here
   44 |       logger::debug("Disable instruction skipped: {}.", to_string(instruction));
      |       ~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/logger.hpp:50:16: error: call to consteval function ‘fmt::v10::basic_format_string<char, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >(fmt)’ is not a constant expression
   50 |   spdlog::debug(fmt, std::forward<Args>(args)...);
      |   ~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/spdlog/fmt/fmt.h:28,
                 from /usr/include/spdlog/common.h:50,
                 from /usr/include/spdlog/sinks/daily_file_sink.h:6,
                 from src/logger.hpp:7:
src/logger.hpp:50:16:   in ‘constexpr’ expansion of ‘fmt::v10::basic_format_string<char, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >(fmt)’
/usr/include/fmt/core.h:2776:62: error: ‘fmt’ is not a constant expression
 2776 |   FMT_CONSTEVAL FMT_INLINE basic_format_string(const S& s) : str_(s) {
      |                                                              ^~~~~~~
cc1plus: note: unrecognized command-line option ‘-Wno-unknown-warning-option’ may have been intended to silence earlier diagnostics
[3/13] Compiling C++ object src/linux-enable-ir-emitter.p/command_tweak.cpp.o
FAILED: src/linux-enable-ir-emitter.p/command_tweak.cpp.o 
c++ -Isrc/linux-enable-ir-emitter.p -Isrc -I../linux-enable-ir-emitter-6.0.3/src -I/usr/include/gtk-3.0 -I/usr/include/pango-1.0 -I/usr/include/cairo -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/atk-1.0 -I/usr/include/freetype2 -I/usr/include/glib-2.0 -I/usr/lib/glib-2.0/include -I/usr/include/harfbuzz -I/usr/include/libmount -I/usr/include/blkid -I/usr/include/libpng16 -I/usr/include/pixman-1 -I/usr/include/cloudproviders -I/usr/include/at-spi2-atk/2.0 -I/usr/include/at-spi-2.0 -I/usr/include/dbus-1.0 -I/usr/lib/dbus-1.0/include -I/usr/include/fribidi -I/usr/include/sysprof-6 -I/usr/include/gio-unix-2.0 -fdiagnostics-color=always -D_GLIBCXX_ASSERTIONS=1 -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -Wpedantic -Wcast-qual -Wconversion -Wfloat-equal -Wformat=2 -Winline -Wmissing-declarations -Wredundant-decls -Wshadow -Wundef -Wuninitialized -Wwrite-strings -Wdisabled-optimization -Wpacked -Wpadded -Wmultichar -Wswitch-default -Wswitch-enum -Wunused-macros -Wmissing-include-dirs -Wunsafe-loop-optimizations -Wstack-protector -Wstrict-overflow=5 -Warray-bounds=2 -Wlogical-op -Wstrict-aliasing=3 -Wvla -Wdouble-promotion -Wsuggest-attribute=const -Wsuggest-attribute=noreturn -Wsuggest-attribute=pure -Wtrampolines -Wvector-operation-performance -Wsuggest-attribute=format -Wdate-time -Wformat-signedness -Wnormalized=nfc -Wduplicated-cond -Wnull-dereference [... cut ...]
[11/13] Compiling C++ object src/linux-enable-ir-emitter.p/camera_camera.cpp.o
[12/13] Compiling C++ object src/linux-enable-ir-emitter.p/_build_linux-enable-ir-emitter_src_build_src_linux-enable-ir-emitter.cpp.o
ninja: build stopped: subcommand failed.
INFO: autodetecting backend as ninja
INFO: calculating backend command to run: /usr/bin/ninja -C /build/linux-enable-ir-emitter/src/build
```

I also tried with GCC 13 just in case, but building still fails.

**Additional info**
 - Linux distribution: Arch Linux
 - Version of linux-enable-ir-emitter: 6.0.3

EmixamPP, 2024-09-06:
That's because I use spdlog with C++20 std format lib instead of fmt: https://github.com/EmixamPP/linux-enable-ir-emitter/blob/b8ee83df28f0a74e66df0dd143f8a90a4e427900/.github/workflows/deps/CMakeLists.txt#L94-L103

This allows me to ship one less library in the standalone binary I provide here.

Honestly, I'm not really in favor of suppoting package-format distros here. But I understand that in this case, you're really in a cul-de-sac. So what I propose: I can add an option to the meson config in order to enable or disable a macro that will use the fmt or std library for that.

I will try to find some time next week to do that, otherwise feel free to PR anything

iyanmv, 2024-09-06:
Oh, I see. But why do you jump directly to blame PKGBUILD? I'm not asking for you to support any package-format distros, let me do that for you. I saw there was a new version, I read the [changelog](https://github.com/EmixamPP/linux-enable-ir-emitter/blob/master/CHANGELOG.md), and nothing there made me thing there was a relevant change that I would have to change the packaging.

By the way, forget about the PKGBUILD, I'm following your [manual-build.md](https://github.com/EmixamPP/linux-enable-ir-emitter/blob/master/docs/manual-build.md) instructions and I get exactly the same error.

I will check your GitHub workflows and adapt the PKGBUILD.

nikicat, 2024-09-15:
Hello, is there any workaround?

EmixamPP, 2024-09-28:
I'll do it on Monday at the latest, sorry for the long wait

EmixamPP, 2024-09-30:
I think, I found an easy fix, please check that it works for you: https://github.com/EmixamPP/linux-enable-ir-emitter/pull/197

iyanmv, 2024-09-30:
Unfortunately, that fix does not solve the issue. I tried with 6.0.4 and applying the patch #197 on top. I copy here the output:

<details>

```
+ exec meson setup --prefix /usr --libexecdir lib --sbindir bin --buildtype plain --auto-features enabled --wrap-mode nodownload -D b_pie=true -D python.bytecompile=1 linux-enable-ir-emitter-6.0.4 build
The Meson build system
Version: 1.5.2
Source dir: /build/linux-enable-ir-emitter/src/linux-enable-ir-emitter-6.0.4
Build dir: /build/linux-enable-ir-emitter/src/build
Build type: native build
Project name: linux-enable-ir-emitter
Project version: 6.0.4
C++ compiler for the host machine: c++ (gcc 14.2.1 "c++ (GCC) 14.2.1 20240910")
C++ linker for the host machine: c++ ld.bfd 2.43.0
Host machine cpu family: x86_64
Host machine cpu: x86_64
Found pkg-config: YES (/usr/bin/pkg-config) 2.1.1
Run-time dependency opencv4 found: YES 4.10.0
Run-time dependency gtk+-3.0 found: YES 3.24.43
Run-time dependency yaml-cpp found: YES 0.8.0
Run-time dependency argparse found: YES 3.1.0
Run-time dependency spdlog found: YES 1.14.1
Configuring linux-enable-ir-emitter.cpp using configuration
Configuring configuration.hpp using configuration
Configuring logger.hpp using configuration
Build targets in project: 1

linux-enable-ir-emitter 6.0.4

  User defined options
    auto_features     : enabled
    buildtype         : plain
    libexecdir        : lib
    prefix            : /usr
    sbindir           : bin
    wrap_mode         : nodownload
    python.bytecompile: 1
    b_pie             : true

Found ninja-1.12.1 at /usr/bin/ninja
ninja: Entering directory `/build/linux-enable-ir-emitter/src/build'
[1/13] Compiling C++ object src/linux-enable-ir-emitter.p/camera_autocamera.cpp.o
[2/13] Compiling C++ object src/linux-enable-ir-emitter.p/configuration_scanner.cpp.o
FAILED: src/linux-enable-ir-emitter.p/configuration_scanner.cpp.o 
c++ -Isrc/linux-enable-ir-emitter.p -Isrc -I../linux-enable-ir-emitter-6.0.4/src -I/usr/include/gtk-3.0 -I/usr/include/pango-1.0 -I/usr/include/cairo -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/atk-1.0 -I/usr/include/freetype2 -I/usr/include/glib-2.0 -I/usr/lib/glib-2.0/include -I/usr/include/harfbuzz -I/usr/include/libmount -I/usr/include/blkid -I/usr/include/libpng16 -I/usr/include/pixman-1 -I/usr/include/cloudproviders -I/usr/include/at-spi2-atk/2.0 -I/usr/include/at-spi-2.0 -I/usr/include/dbus-1.0 -I/usr/lib/dbus-1.0/include -I/usr/include/fribidi -I/usr/include/sysprof-6 -I/usr/include/gio-unix-2.0 -fdiagnostics-color=always -D_GLIBCXX_ASSERTIONS=1 -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -Wpedantic -Wcast-qual -Wconversion -Wfloat-equal -Wformat=2 -Winline -Wmissing-declarations -Wredundant-decls -Wshadow -Wundef -Wuninitialized -Wwrite-strings -Wdisabled-optimization -Wpacked -Wpadded -Wmultichar -Wswitch-default -Wswitch-enum -Wunused-macros -Wmissing-include-dirs -Wunsafe-loop-optimizations -Wstack-protector -Wstrict-overflow=5 -Warray-bounds=2 -Wlogical-op -Wstrict-aliasing=3 -Wvla -Wdouble-promotion -Wsuggest-attribute=const -Wsuggest-attribute=noreturn -Wsuggest-attribute=pure -Wtrampolines -Wvector-operation-performance -Wsuggest-attribute=format -Wdate-time -Wformat-signedness -Wnormalized=nfc -Wduplicated-cond -Wnull-dereference -Wshift-negative-value -Wshift-overflow=2 -Wunused-const-variable=2 -Walloca -Walloc-zero -Wformat-overflow=2 -Wformat-truncation=2 -Wstringop-overflow=3 -Wduplicated-branches -Wcast-align=strict -Wsuggest-attribute=cold -Wsuggest-attribute=malloc -Wattribute-alias=2 -Wanalyzer-too-complex -Warith-conversion -Wbidi-chars=ucn -Wopenacc-parallelism -Wtrivial-auto-var-init -Wctor-dtor-privacy -Weffc++ -Wnon-virtual-dtor -Wold-style-cast -Woverloaded-virtual -Wsign-promo -Wstrict-null-sentinel -Wnoexcept -Wzero-as-null-pointer-constant -Wabi-tag -Wuseless-cast -Wconditionally-supported -Wsuggest-final-methods -Wsuggest-final-types -Wsuggest-override -Wmultiple-inheritance -Wplacement-new=2 -Wvirtual-inheritance -Waligned-new=all -Wnoexcept-type -Wregister -Wcatch-value=3 -Wextra-semi -Wdeprecated-copy-dtor -Wredundant-move -Wcomma-subscript -Wmismatched-tags -Wredundant-tags -Wvolatile -Wdeprecated-enum-enum-conversion -Wdeprecated-enum-float-conversion -Winvalid-imported-macros -Werror -std=c++20 -Wno-unknown-warning-option -Wno-padded -Wno-shadow -Wno-abi-tag -Wno-effc++ -Wno-unused-const-variable -Wno-suggest-attribute=const -Wno-inline -Wno-suggest-attribute=pure -Wno-suggest-final-types -Wno-suggest-final-methods -march=x86-64 -mtune=generic -O2 -pipe -fno-plt -fexceptions -Wp,-D_FORTIFY_SOURCE=3 -Wformat -Werror=format-security -fstack-clash-protection -fcf-protection -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -Wp,-D_GLIBCXX_ASSERTIONS -g -ffile-prefix-map=/build/linux-enable-ir-emitter/src=/usr/src/debug/linux-enable-ir-emitter -flto=auto -fPIE -DSPDLOG_SHARED_LIB -DSPDLOG_COMPILED_LIB -DSPDLOG_FMT_EXTERNAL -pthread -isystem/usr/include/opencv4 -MD -MQ src/linux-enable-ir-emitter.p/configuration_scanner.cpp.o -MF src/linux-enable-ir-emitter.p/configuration_scanner.cpp.o.d -o src/linux-enable-ir-emitter.p/configuration_scanner.cpp.o -c ../linux-enable-ir-emitter-6.0.4/src/configuration/scanner.cpp
In file included from /usr/include/fmt/core.h:5,
                 from /usr/include/spdlog/fmt/fmt.h:28,
                 from /usr/include/spdlog/common.h:50,
                 from /usr/include/spdlog/sinks/daily_file_sink.h:6,
                 from src/logger.hpp:7,
[... cut ...]
   76 |   spdlog::critical(FMT_STRING(fmt), std::forward<Args>(args)...);
      |                               ^~~
src/logger.hpp:75:32: note: ‘std::string_view fmt’ declared here
   75 | void critical(std::string_view fmt, Args &&...args) {
      |               ~~~~~~~~~~~~~~~~~^~~
At global scope:
cc1plus: note: unrecognized command-line option ‘-Wno-unknown-warning-option’ may have been intended to silence earlier diagnostics
[12/13] Compiling C++ object src/linux-enable-ir-emitter.p/camera_camera.cpp.o
ninja: build stopped: subcommand failed.
```

</details>

EmixamPP, 2024-10-02:
Ok, this time this should be good! And rebased on 6.0.4; it will be easier for you

iyanmv, 2024-10-02:
Now yes, thank you! :)

EmixamPP, 2024-10-02:
Cool! 6.0.5 released https://github.com/EmixamPP/linux-enable-ir-emitter/releases/tag/6.0.5

closed 2024-10-02 by EmixamPP