EmixamPP/linux-enable-ir-emitter #195: IR camera flashes a couple times and stops

madesho, 2024-09-28:
**Ouput of `cat /var/local/log/linux-enable-ir-emitter/linux-enable-ir-emitter.log`**
```
[17:56:55] [debug] Executing configure command.
[17:56:55] [info] Stand in front of and close to the camera and make sure the room is well lit.
[17:56:55] [info] Ensure to not use the camera during the execution.
[17:56:55] [debug] Checking if /dev/video0 is a grayscale camera.
[17:56:57] [debug] Checking if /dev/video1 is a grayscale camera.
[17:56:57] [debug] Checking if /dev/video2 is a grayscale camera.
[17:56:58] [debug] /dev/video2 is a grayscale camera.
[17:56:58] [info] Configuring the camera /dev/video2.
[17:56:58] [debug] yaml error: bad file: /etc/linux-enable-ir-emitter/pci-0000:04:00.3-usb-0:4:1.2-video-index0.ini.
[17:56:58] [debug] No previous configuration found.
[17:56:59] [debug] Configuration for /dev/video2 saved here: /etc/linux-enable-ir-emitter/pci-0000:04:00.3-usb-0:4:1.2-video-index0.ini.
[17:57:07] [error] The emitter is already working, skipping the configuration.
[18:57:12] [debug] Executing configure command.
[18:57:12] [info] Stand in front of and close to the camera and make sure the room is well lit.
[18:57:12] [info] Ensure to not use the camera during the execution.
[18:57:12] [debug] Checking if /dev/video0 is a grayscale camera.
[18:57:14] [debug] Checking if /dev/video1 is a grayscale camera.
[18:57:14] [debug] Checking if /dev/video2 is a grayscale camera.
[18:57:16] [debug] /dev/video2 is a grayscale camera.
[18:57:16] [info] Configuring the camera /dev/video2.
[18:57:16] [debug] Previous configuration found.
[18:57:28] [debug] Instruction applied: unit: 14, selector: 6, control: 1 3 2 0 0 0 0 0 0.
[18:57:30] [debug] The instruction is not valid.
[18:57:30] [debug] Instruction applied: unit: 14, selector: 6, control: 1 3 3 0 0 0 0 0 0.
[18:57:31] [debug] The instruction is not valid.
[18:57:31] [debug] Reseting to the instruction: unit: 14, selector: 6, control: 1 3 1 0 0 0 0 0 0.
[18:57:31] [debug] Instruction applied: unit: 14, selector: 9, control: 1 0 0 0.
[18:57:32] [debug] The instruction is not valid.
[18:57:32] [debug] Reseting to the instruction: unit: 14, selector: 9, control: 1 0 0 0.
[18:57:32] [error] The configuration failed.
[18:57:32] [info] Please retry in manual mode by adding the '-m' option.
[18:57:32] [info] Do not hesitate to visit the GitHub!
[18:57:32] [info] https://github.com/EmixamPP/linux-enable-ir-emitter/blob/master/docs/README.md
[19:01:46] [debug] Executing configure command.
[19:01:46] [info] Stand in front of and close to the camera and make sure the room is well lit.
[19:01:46] [info] Ensure to not use the camera during the execution.
[19:01:46] [debug] Checking if /dev/video0 is a grayscale camera.
[19:01:47] [debug] Checking if /dev/video1 is a grayscale camera.
[19:01:47] [debug] Checking if /dev/video2 is a grayscale camera.
[19:01:49] [debug] /dev/video2 is a grayscale camera.
[19:01:49] [info] Configuring the camera /dev/video2.
[19:01:49] [debug] Previous configuration found.
[19:01:58] [debug] Instruction applied: unit: 14, selector: 6, control: 1 3 2 0 0 0 0 0 0.
[19:01:59] [debug] The instruction is not valid.
[19:01:59] [debug] Instruction applied: unit: 14, selector: 6, control: 1 3 3 0 0 0 0 0 0.
[19:02:01] [debug] The instruction is not valid.
[19:02:01] [debug] Reseting to the instruction: unit: 14, selector: 6, control: 1 3 1 0 0 0 0 0 0.
[19:02:01] [debug] Instruction applied: unit: 14, selector: 9, control: 1 0 0 0.
[19:02:02] [debug] The instruction is not valid.
[19:02:02] [debug] Reseting to the instruction: unit: 14, selector: 9, control: 1 0 0 0.
[19:02:02] [error] The configuration failed.
[19:02:02] [info] Please retry in manual mode by adding the '-m' option.
[19:02:02] [info] Do not hesitate to visit the GitHub!
[19:02:02] [info] https://github.com/EmixamPP/linux-enable-ir-emitter/blob/master/docs/README.md
[19:02:09] [debug] Executing configure command.
[19:02:09] [info] Stand in front of and close to the camera and make sure the room is well lit.
[19:02:09] [info] Ensure to not use the camera during the execution.
[19:02:09] [debug] Checking if /dev/video0 is a grayscale camera.
[19:02:09] [debug] Checking if /dev/video1 is a grayscale camera.
[19:02:09] [debug] Checking if /dev/video2 is a grayscale camera.
[19:02:10] [debug] /dev/video2 is a grayscale camera.
[19:02:10] [info] Configuring the camera /dev/video2.
[19:02:10] [debug] Previous configuration found.
[19:02:14] [debug] Instruction applied: unit: 14, selector: 6, control: 1 3 2 0 0 0 0 0 0.
[19:02:17] [debug] The instruction is not valid.
[19:02:17] [debug] Instruction applied: unit: 14, selector: 6, control: 1 3 3 0 0 0 0 0 0.
[19:02:20] [debug] The instruction is not valid.
[19:02:20] [debug] Reseting to the instruction: unit: 14, selector: 6, control: 1 3 1 0 0 0 0 0 0.
[... 295 lines cut ...]
	Type: Video Capture
```

**Output of `ls -l /dev/v4l/by-path`**
```
totale 0
lrwxrwxrwx. 1 root root 12 28 set 03.29 pci-0000:04:00.3-usb-0:4:1.0-video-index0 -> ../../video0
lrwxrwxrwx. 1 root root 12 28 set 03.29 pci-0000:04:00.3-usb-0:4:1.0-video-index1 -> ../../video1
lrwxrwxrwx. 1 root root 12 28 set 03.29 pci-0000:04:00.3-usb-0:4:1.2-video-index0 -> ../../video2
lrwxrwxrwx. 1 root root 12 28 set 03.29 pci-0000:04:00.3-usb-0:4:1.2-video-index1 -> ../../video3
lrwxrwxrwx. 1 root root 12 28 set 03.29 pci-0000:04:00.3-usbv2-0:4:1.0-video-index0 -> ../../video0
lrwxrwxrwx. 1 root root 12 28 set 03.29 pci-0000:04:00.3-usbv2-0:4:1.0-video-index1 -> ../../video1
lrwxrwxrwx. 1 root root 12 28 set 03.29 pci-0000:04:00.3-usbv2-0:4:1.2-video-index0 -> ../../video2
lrwxrwxrwx. 1 root root 12 28 set 03.29 pci-0000:04:00.3-usbv2-0:4:1.2-video-index1 -> ../../video3
```

**Give more information if you have**
IR camera (/dev/video2) flashes briefly for about half a second and then stops working showing only a black screen in the preview. This happens everytime it is llaunched. 
One time I forgot to close a camera app that was using the normal webcam, on that occasion the ir camera was warking fine and can consistently reproduce this. 
I'm new-ish to Linux so my apologies if something is missing, let me know if you need anything else :)

**Additional info**
 - Computer (or camera) model: HP Envy x360 15-fh0002nl, sometimes known as fh0000 or fh0xxx
 - Linux distribution: Ultramarine Linux 40, based on Fedora 40
 - Version of linux-enable-ir-emitter:  6.0.3

EmixamPP, 2024-09-28:
Execute 
1. sudo linux-enable-ir-emitter -v tweak
2. input 2
3. input 1 3 2 0 0 0 0 0 0
4. look at the video feedback and the ir emitter, does it work?
5. if not retry with 1 3 3 0 0 0 0 0 0
6. does it work?
7. if not paste output here

madesho, 2024-09-28:
```
Executing tweak command.
Checking if /dev/video0 is a grayscale camera.
Checking if /dev/video1 is a grayscale camera.
Checking if /dev/video2 is a grayscale camera.
/dev/video2 is a grayscale camera.
Tweaking the camera /dev/video2
Caution, you could break the camera.
Previous configuration found.
0) unit: 14, selector: 6, control: 1 3 1 0 0 0 0 0 0
1) unit: 14, selector: 9, control: 1 0 0 0
2) exit
Choose an instruction to tweak: 0
minimum:
maximum: 1 3 3 0 0 0 0 0 0
initial: 1 3 1 0 0 0 0 0 0
current: 1 3 1 0 0 0 0 0 0
status: enable
new current or status:  1 3 2 0 0 0 0 0 0
Invalid value for the instruction.
0) unit: 14, selector: 6, control: 1 3 1 0 0 0 0 0 0
1) unit: 14, selector: 9, control: 1 0 0 0
2) exit
Choose an instruction to tweak: terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.10.0) /home/runner/work/linux-enable-ir-emitter/linux-enable-ir-emitter/.github/workflows/deps/build/opencv/src/opencv/modules/highgui/src/window.cpp:973: error: (-215:Assertion failed) size.width>0 && size.height>0 in function 'imshow'

zsh: IOT instruction  sudo linux-enable-ir-emitter -v tweak
```

Seems that I can't change the configuration, am I doing something wrong?

EmixamPP, 2024-09-30:
Could you retry with the new v6.0.4? I think it should work now; I fixed a bug that you reported here!

madesho, 2024-09-30:
I tried to uninstall the old version and installed the new one and the camera was working correctly without changing any setting for some reason? Then I installed Howdy and it did the same thing that I described the other day with a few flashes and then the black preview. 
Went back to try with your solution of the other day and now it changes the settings correctly, but it still doesn't work properly. 
Also tried to uninstall Howdy and its dependencies to see if was some kind of conflict with something, but the problem is still there

EmixamPP, 2024-10-02:
First using the tweak command assign the value `1 3 2 0 0 0 0 0 0` to the instruction 0
Then exit and:
1. execute `linux-enable-ir-emitter -d /dev/video2 test` does the emitter flash?
2. execute `linux-enable-ir-emitter -d /dev/video2 test` does the emitter flash?
3. execute `linux-enable-ir-emitter run`
4. execute `linux-enable-ir-emitter -d /dev/video2 test` does the emitter flash?
5. execute `linux-enable-ir-emitter -d /dev/video2 test` does the emitter flash?

We will determine if the applying of the instruction is permanent or not

madesho, 2024-10-07:
Sorry for the late answer, followed your instructions and it's the same thing as always, some flashes and then black screen

EmixamPP, 2024-11-26:
I just want to let you know that I am working on a kernel space solution (instead of user space as currently), this should solve this kind of issue (I think).

EmixamPP, 2025-02-16:
@zkurdi45 could be something that may disable using the `tweak` command, see [readme](https://github.com/EmixamPP/linux-enable-ir-emitter?tab=readme-ov-file#how-do-i-tweak-my-camera)