EmixamPP/linux-enable-ir-emitter #198: Doesnot work on Zorin 17.2

ebeneezer, 2024-10-05:
```
oc@Citong:~/Downloads/usr/local/bin$ ./linux-enable-ir-emitter 
./linux-enable-ir-emitter: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by ./linux-enable-ir-emitter)
./linux-enable-ir-emitter: /lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.32' not found (required by ./linux-enable-ir-emitter)
./linux-enable-ir-emitter: /lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.31' not found (required by ./linux-enable-ir-emitter)
idoc@Citong:~/Downloads/usr/local/bin$ 
```

System is a Dell XPS 15 9500

Seems base system of Zorin uses old Clib.

Tried to compile source. Does not work either. Misses format.h header.

EmixamPP, 2024-10-05:
Is it possible for you to install gcc13 on ZorinOs?

ebeneezer, 2024-10-05:
Tnx for looking into my problem. I will try and post results here.

ebeneezer, 2024-10-06:
I decided to ditch Zorin in favour of Ubuntu 24.4.1 as I expect more trouble ahead using an outdated distro base - especially using Linux on Notebooks.

Your app installs and runs perfectly on Ubuntu 24.4.1

I use and recommend "Slimbook-Face" as GUI to register faces with Howdy and using your driver the two emitters flash every time Howdy activates tthe IR cam.

Hope to see your driver integrated in Howdy in the future.

Thank you for your support and i will close the issue now.

closed 2024-10-06 by ebeneezer