EmixamPP/linux-enable-ir-emitter #202: build() Failed.

derFelix244, 2024-10-17:
**Describe the bug**
Installation via yay and  pkgbuild failes with an error.

Im globalen Gültigkeitsbereich:
cc1plus: Anmerkung: Unbekannte Kommandozeilenoption »-Wno-unknown-warning-option« könnte den Zweck haben, bisherige Diagnosemeldungen zu unterdrücken
cc1plus: Alle Warnungen werden als Fehler behandelt
ninja: build stopped: subcommand failed.

Translation:
In the global scope 
cc1plus: Unknown command line argument: "-Wno-unknown-warning-option" could be used to suppress diagnostic warnings.
cc1plus: All warnings are seen as errors
ninja: build stopped: subcommand failed.

**How to reproduce**
install - I guess

**Additional info**
 - Computer (or camera) model: Thinpad T490
 - Linux distribution: Endeavor OS
 - Version of linux-enable-ir-emitter: 6.0.6

EmixamPP, 2024-10-17:
Could you copy paste all the building logs ?

derFelix244, 2024-10-17:
Here is the Pastebin: https://pastebin.com/XZK1rnfw
Unfortunately, a big part of it is in German. I can translate, if needed.

closed 2024-10-21 by EmixamPP