EmixamPP/linux-enable-ir-emitter #203: The webcam continues forgetting the configuration

elegos, 2024-10-24:
**Describe the bug**
I've got an IR webcam connected to the computer via a KVM. I've noticed that pretty much every time I restart the computer, the IR emitter doesn't emit light. If I configure it again via `sudo linux-enable-ir-emitter configure -m` it works again, until the next restart (or two). I've noticed that in `/etc/linux-enable-ir-emitter` I have different configuration files, even if the webcam is one.

I'm in a dual-boot system (Windows / Linux)

```bash
  /etc/linux-enable-ir-emitter                                                                                                            at  13:00:23
❯ ll
totale 16K
-rw-r--r--. 1 root root 460 24 ott 13.00 pci-0000:02:00.0-usb-0:1.4.1:1.2-video-index0
-rw-r--r--. 1 root root 377  9 ott 11.26 pci-0000:02:00.0-usb-0:1.4.1:1.2-video-index0.ini
-rw-r--r--. 1 root root 460 17 set 00.17 pci-0000:0c:00.3-usb-0:2.4.1:1.2-video-index0
-rw-r--r--. 1 root root 377 15 set 17.00 pci-0000:0c:00.3-usb-0:2.4.1:1.2-video-index0.ini
```

**How to reproduce**
- (possibly?) use a KVM
- Configure the emitter manually (the automatic configuration fails to let the emitter work correctly)
- The emitter works correctly
- Restart the computer (once or twice, or just start another OS and turn back to Linux)

**Additional info**
 - Computer (or camera) model: Nexight INC NexiGo HelloCam N930W Camera
 - Linux distribution: Fedora Linux 40 (KDE flavour)
 - Version of linux-enable-ir-emitter: Command fails (too few arguments), though I've downloaded and installed the latest version available as of 2nd of September, 2024

EmixamPP, 2024-11-23:
Could you show me the output of  `systemctl status linux-enable-ir-emitter`?

By the way you, only need to configure once, then just execute `linux-enable-ir-emitter run` to re-apply the last configuration

EmixamPP, 2024-11-26:
I also just want to let you know that I am working on a kernel space solution (instead of user space as currently), this should solve this kind of issue (I think).

elegos, 2024-11-27:
Sorry, I totally missed the notifications somehow!!

The linux-enable-ir-emitter.service effectively failed (core-dumped).

```
nov 27 23:07:29 plungedcopper linux-enable-ir-emitter[1628]: Executing run command.
nov 27 23:07:29 plungedcopper linux-enable-ir-emitter[1628]: filesystem error: cannot make canonical path: No such file or directory [/dev/v4l/by-path/pci-0000:0c:00.3-usb-0:2.4.1:1.2-video-index0]
nov 27 23:07:29 plungedcopper linux-enable-ir-emitter[1628]: terminate called without an active exception
```

```
❯ ls -l /dev/v4l/by-path
totale 0
lrwxrwxrwx. 1 root root 12 27 nov 23.07 pci-0000:02:00.0-usb-0:1.4.1:1.0-video-index0 -> ../../video0
lrwxrwxrwx. 1 root root 12 27 nov 23.07 pci-0000:02:00.0-usb-0:1.4.1:1.0-video-index1 -> ../../video1
lrwxrwxrwx. 1 root root 12 27 nov 23.07 pci-0000:02:00.0-usb-0:1.4.1:1.2-video-index0 -> ../../video2
lrwxrwxrwx. 1 root root 12 27 nov 23.07 pci-0000:02:00.0-usb-0:1.4.1:1.2-video-index1 -> ../../video3
lrwxrwxrwx. 1 root root 12 27 nov 23.07 pci-0000:02:00.0-usbv2-0:1.4.1:1.0-video-index0 -> ../../video0
lrwxrwxrwx. 1 root root 12 27 nov 23.07 pci-0000:02:00.0-usbv2-0:1.4.1:1.0-video-index1 -> ../../video1
lrwxrwxrwx. 1 root root 12 27 nov 23.07 pci-0000:02:00.0-usbv2-0:1.4.1:1.2-video-index0 -> ../../video2
lrwxrwxrwx. 1 root root 12 27 nov 23.07 pci-0000:02:00.0-usbv2-0:1.4.1:1.2-video-index1 -> ../../video3
```

It seems that the KVM switches the identifiers, apparently

elegos, 2024-11-28:
For now I made a (very crappy) hack support service, like follows (as the configuration never changes, only (part of the) identifier:

```
[Unit]
Description = COPY infrared configuration (hack)
Before = linux-enable-ir-emitter.service 

[Service]
Type = oneshot
ExecStartPre = /sbin/modprobe uvcvideo
ExecStart = /usr/local/eremitter-helper/helper.sh

[Install]
WantedBy = multi-user.target suspend.target sleep.target hybrid-sleep.target hibernate.target suspend-then-hibernate.target
```

The script:
```bash
#!/usr/bin/env bash

config_dir="/etc/linux-enable-ir-emitter"

v_device=$(ls -l /dev/v4l/by-path/ | grep ":1.2-video-index0" | grep -v usbv | awk '{print $9}')

is_registered=$(ls -l "${config_dir}" | grep "${v_device}" | grep -v ".ini")
is_registered_ini=$(ls -l "${config_dir}" | grep "${v_device}.ini")

if [ "${is_registered}" = "" ]; then
        cp_file=$(ls -1 /etc/linux-enable-ir-emitter | grep -v ".ini" | head -n 2 | tail -n 1)
        cp "${config_dir}/${cp_file}" "${config_dir}/${v_device}"
fi

if [ "${is_registered_ini}" = "" ]; then
        cp_file=$(ls -1 /etc/linux-enable-ir-emitter | grep ".ini" | head -n 2 | tail -n 1)
        cp "${config_dir}/${cp_file}" "${config_dir}/${v_device}.ini"
fi
```

EmixamPP, 2024-11-28:
So if I understand correctly, you think that your camera is changing its v4l path ? Hum, sounds strange because that's the point of such path; be well determined. 

Maybe you could try to delete all the files in /etc/linux-enable-ir-emitter, restart and reconfigure to have a fresh configuration for the current path (if you know its actual /dev/video path you can specify it with -d) and see if it works for the next restarts? In the next update, I plan to release soon(TM), I've made the tool more resilient to error if one configuration fails to apply on the linked camera.

elegos, 2024-11-28:
For some reason, I can tell you that it changes somehow. I don't know if this depends on system updates or KVM, but it definitely happens. I'll monitor the path to understand if it always changes or not.

I'm not into hardware like you are at Linux level, though couldn't you add the option to recognize the webcam via vendor:product instead? That should never change by definition

EmixamPP, 2024-11-30:
This version https://github.com/EmixamPP/linux-enable-ir-emitter/pull/211 will just ignore configuration that it cannot apply, so you could have a bunch of configuration in case of device path change.
There is a bot comment explaining how to install it

elegos, 2025-01-08:
Hello @EmixamPP - just an update on the issue.

I can confirm that - apparently - the KVM is causing the issue, changing the pci address - how cute!

Exploring udev rules for another project, I managed to sort out a better solution: I've added a udev rule which aliases the device.

This is the particular rule:

```
SUBSYSTEM=="video4linux", ATTR{name}=="NexiGo HelloCam N930W Camera: N", ATTRS{interface}=="NexiGo HelloCam N930W IR Camera", TAG+="symlink", SYMLINK+="v4l/by-path/nexigo-hellocam-n930w-camera-ir"
```

I got the udev filter by doing an attribute walk of the known device: `devadm info --attribute-walk --name=/dev/v4l/by-path/pci-0000:0a:00.3-usb-0:4.4.1:1.2-video-index0` - then I immediately saw that (probably only required) `ATTRS{interface}`.

As a result, I now have a stable `/dev/v4l/by-path/nexigo-hellocam-n930w-camera-ir` symlink to the same hardware. After configuring the emitter service, I now have the two files in `/etc/linux-enable-ir-emitter`:

```
-rw-r--r--. 1 root root 460  8 gen 02.01 nexigo-hellocam-n930w-camera-ir
-rw-r--r--. 1 root root 377  8 gen 02.01 nexigo-hellocam-n930w-camera-ir.ini
```

EmixamPP, 2025-02-16:
In the first versions of the tool I was also generating such a file at the end of the configuration, but it was not very convenient, and often not required.

However, your idea of using the `by-id` v4l path instead of `by-path` should solve the issue. I will include that in the next release, but this will be a breaking change; I am afraid.

zayn7lie, 2025-11-25:
@MixSens I think I have successfully figured out for my laptop (ROG Flow X13 with `Sonix_Technology_Co.__Ltd._USB2.0_FHD_UVC_WebCam`). First, the `-d --device` option is preferred to pass in a `/dev/v4l/by-path/*` dir instead of `/dev/v4l/by-id/*` since the latter one would failed in my laptop. So second what you need to do is to figure out which one in `/dev/v4l/by-path/` link to your ir camera and then pass it to `-d --device` as you configure and run. In this issue @elegos 's device is not stably linked to `/dev/v4l/by-path/` so the manual stable link is required (achieved by `udev` rule). To fix the issue, you just need to figure out which link in `/dev/v4l/by-path/` link to your ir camera, if it is unstable then manually create a stable link.

EmixamPP, 2025-12-16:
The v7.0.0-beta is out and is using `/dev/v4l/by-id/` as you proposed! Thanks!