EmixamPP/linux-enable-ir-emitter #210: Upon running config command it randomly exits

WX580, 2024-11-27:
**Ouput of `cat /var/local/log/linux-enable-ir-emitter/linux-enable-ir-emitter.log`**
```
```

**Output of `sudo linux-enable-ir-emitter tweak`**
```
Tweaking the camera /dev/video2
Caution, you could break the camera.

terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
Aborted

```

**Output of `v4l2-ctl --list-devices`**
```
Logitech BRIO (usb-0000:04:00.0-1):
        /dev/video0
        /dev/video1
        /dev/video2
        /dev/video3
        /dev/media0

```

**Ouput of `for dev in /dev/video*; do echo $dev && v4l2-ctl -d $dev --list-formats-ext; done`**
```
/dev/video0
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'YUYV' (YUYV 4:2:2)
                Size: Discrete 640x480
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.042s (24.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.133s (7.500 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 160x120
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.042s (24.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.133s (7.500 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 176x144
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.042s (24.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.133s (7.500 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 320x180
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.042s (24.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.133s (7.500 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 320x240
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.042s (24.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
[... 233 lines cut ...]

```

**Output of `ls -l /dev/v4l/by-path`**
```
total 0
lrwxrwxrwx 1 root root 12 Nov 26 20:07 pci-0000:04:00.0-usb-0:1:1.0-video-index0 -> ../../video0
lrwxrwxrwx 1 root root 12 Nov 26 20:07 pci-0000:04:00.0-usb-0:1:1.0-video-index1 -> ../../video1
lrwxrwxrwx 1 root root 12 Nov 26 20:07 pci-0000:04:00.0-usb-0:1:1.0-video-index2 -> ../../video2
lrwxrwxrwx 1 root root 12 Nov 26 20:07 pci-0000:04:00.0-usb-0:1:1.0-video-index3 -> ../../video3
lrwxrwxrwx 1 root root 12 Nov 26 20:07 pci-0000:04:00.0-usbv2-0:1:1.0-video-index0 -> ../../video0
lrwxrwxrwx 1 root root 12 Nov 26 20:07 pci-0000:04:00.0-usbv2-0:1:1.0-video-index1 -> ../../video1
lrwxrwxrwx 1 root root 12 Nov 26 20:07 pci-0000:04:00.0-usbv2-0:1:1.0-video-index2 -> ../../video2
lrwxrwxrwx 1 root root 12 Nov 26 20:07 pci-0000:04:00.0-usbv2-0:1:1.0-video-index3 -> ../../video3

```

**Give more information if you have**
IR camera is video2 along with the verbose output of the configure command
[image]

**Additional info**
 - Computer (or camera) model: Logitech BRIO
 - Linux distribution: Arch
 - Version of linux-enable-ir-emitter: Latest ((I assume) --version and -V were unknown arguments)

EmixamPP, 2024-11-30:
Could you try this new version: https://github.com/EmixamPP/linux-enable-ir-emitter/pull/211
There is a bot comment explaining how to try it (uninstall your current version if installed via your package manager)

EmixamPP, 2025-01-29:
Feel free to comment to reopen

closed 2025-01-29 by EmixamPP