EmixamPP/linux-enable-ir-emitter #212: Impossible to obtain the v4l name while configuring

simonsd054, 2024-12-05:
**Ouput of `cat /var/local/log/linux-enable-ir-emitter/linux-enable-ir-emitter.log`**
```
[05:20:21] [INFO] Stand in front of and close to the camera and make sure the room is well lit.
[05:20:21] [INFO] Ensure to not use the camera during the execution.
[05:20:23] [INFO] Configuring the camera /dev/video0.
[05:20:34] [ERROR] Impossible to obtain the v4l name of /dev/video0.
[05:20:34] [ERROR] The configuration failed.
[05:20:34] [INFO] Do not hesitate to visit the GitHub!
[05:20:34] [INFO] https://github.com/EmixamPP/linux-enable-ir-emitter/blob/master/docs/README.md
[05:31:13] [INFO] Stand in front of and close to the camera and make sure the room is well lit.
[05:31:13] [INFO] Ensure to not use the camera during the execution.
[05:31:14] [INFO] Configuring the camera /dev/video0.
[05:31:19] [ERROR] Impossible to obtain the v4l name of /dev/video0.
[05:31:19] [ERROR] The configuration failed.
[05:31:19] [INFO] Do not hesitate to visit the GitHub!
[05:31:19] [INFO] https://github.com/EmixamPP/linux-enable-ir-emitter/blob/master/docs/README.md
[05:31:35] [INFO] Stand in front of and close to the camera and make sure the room is well lit.
[05:31:35] [INFO] Ensure to not use the camera during the execution.
[05:31:36] [INFO] Configuring the camera /dev/video0.
[05:31:44] [ERROR] Impossible to obtain the v4l name of /dev/video0.
[05:31:44] [ERROR] The configuration failed.
[05:31:44] [INFO] Do not hesitate to visit the GitHub!
[05:31:44] [INFO] https://github.com/EmixamPP/linux-enable-ir-emitter/blob/master/docs/README.md
[05:32:07] [INFO] Stand in front of and close to the camera and make sure the room is well lit.
[05:32:07] [INFO] Ensure to not use the camera during the execution.
[05:32:09] [INFO] Configuring the camera /dev/video0.
[05:32:12] [ERROR] Impossible to obtain the v4l name of /dev/video0.
[05:32:12] [ERROR] The configuration failed.
[05:32:12] [INFO] Do not hesitate to visit the GitHub!
[05:32:12] [INFO] https://github.com/EmixamPP/linux-enable-ir-emitter/blob/master/docs/README.md
[05:35:39] [INFO] Stand in front of and close to the camera and make sure the room is well lit.
[05:35:39] [INFO] Ensure to not use the camera during the execution.
[05:35:40] [INFO] Configuring the camera /dev/video0.
[05:35:44] [ERROR] Impossible to obtain the v4l name of /dev/video0.
[05:35:44] [ERROR] The configuration failed.
[05:35:44] [INFO] Do not hesitate to visit the GitHub!
[05:35:44] [INFO] https://github.com/EmixamPP/linux-enable-ir-emitter/blob/master/docs/README.md
```

**Output of `sudo linux-enable-ir-emitter tweak`**
```
[INFO] Tweaking the camera /dev/video15
[INFO] Caution, you could break the camera.
0) exit
Choose an instruction to tweak: 0
```

**Output of `v4l2-ctl --list-devices`**
```
ipu6 (PCI:pci:pci0000:00):
	/dev/video8
	/dev/video9
	/dev/video10
	/dev/video11
	/dev/video12
	/dev/video13
	/dev/video14
	/dev/video15

ipu6 (pci:pci0000:00):
	/dev/media0

Intel MIPI Camera (platform:v4l2loopback-000):
	/dev/video0
```

**Ouput of `for dev in /dev/video*; do echo $dev && v4l2-ctl -d $dev --list-formats-ext; done`**
```
/dev/video0
ioctl: VIDIOC_ENUM_FMT
[... 194 lines cut ...]
		Size: Discrete 1280x720
			Interval: Discrete 0.033s (30.000 fps)
```

**Output of `ls -l /dev/v4l/by-path`**
```
total 0
lrwxrwxrwx. 1 root root 12 Dec  5 04:13 pci-0000:00:05.0-video-index0 -> ../../video8
lrwxrwxrwx. 1 root root 12 Dec  5 04:13 pci-0000:00:05.0-video-index1 -> ../../video9
lrwxrwxrwx. 1 root root 13 Dec  5 04:13 pci-0000:00:05.0-video-index2 -> ../../video10
lrwxrwxrwx. 1 root root 13 Dec  5 04:13 pci-0000:00:05.0-video-index3 -> ../../video11
lrwxrwxrwx. 1 root root 13 Dec  5 04:13 pci-0000:00:05.0-video-index4 -> ../../video12
lrwxrwxrwx. 1 root root 13 Dec  5 04:13 pci-0000:00:05.0-video-index5 -> ../../video13
lrwxrwxrwx. 1 root root 13 Dec  5 04:13 pci-0000:00:05.0-video-index6 -> ../../video14
lrwxrwxrwx. 1 root root 13 Dec  5 04:13 pci-0000:00:05.0-video-index7 -> ../../video15
```

**Give more information if you have**
I get this error when I try to configure as shown in the image below: `Impossible to obtain the v4l name of /dev/video0`
[image]

**Additional info**
 - Computer (or camera) model: Dell XPS 13 9340
 - Linux distribution: Fedora
 - Version of linux-enable-ir-emitter: 6.1.0 but tried 6.0.6 and 6.0.5 - same error

EmixamPP, 2024-12-16:
If you execute `linux-enable-ir-emitter -d /dev/video<X> test`

Where you replace <X> by the number for each of the following device entry:
	/dev/video8
	/dev/video9
	/dev/video10
	/dev/video11
	/dev/video12
	/dev/video13
	/dev/video14
	/dev/video15
	
Which one does show a grayscale image (or very dark if your room is not lighted enough)?

simonsd054, 2024-12-17:
None of them work and I get the error as seen in the image. It is the same with and without sudo.
[image]

EmixamPP, 2024-12-18:
Try also with the ffplay utils: ffplay /dev/video<X> (ffmpeg package on fedora) to see if you can get a video feedback and let me know

EmixamPP, 2025-01-29:
Feel free to comment to reopen

closed 2025-01-29 by EmixamPP

liquidmoon09, 2025-06-08:
I'm having this same issue on a Dell Latitude 7450 on Mint 22.1. I'm going to keep trying, and I'll let you know if I find a solution.