EmixamPP/linux-enable-ir-emitter #213: Configure crashes with a segmentation fault

KirrimK, 2024-12-14:
**Ouput of `cat /var/local/log/linux-enable-ir-emitter/linux-enable-ir-emitter.log`**
```
[14:57:17] [debug] Executing configure command.
[14:57:17] [info] Stand in front of and close to the camera and make sure the room is well lit.
[14:57:17] [info] Ensure to not use the camera during the execution.
[14:57:17] [debug] Checking if /dev/video0 is a grayscale camera.
[14:57:19] [debug] Checking if /dev/video1 is a grayscale camera.
[14:57:19] [debug] Checking if /dev/video2 is a grayscale camera.
[14:57:22] [debug] /dev/video2 is a grayscale camera.
[14:57:22] [info] Configuring the camera /dev/video2.
[14:57:22] [debug] yaml error: bad file: /etc/linux-enable-ir-emitter/pci-0000:65:00.4-usb-0:1:1.2-video-index0.ini.
[14:57:22] [debug] No previous configuration found.
[14:57:23] [debug] Configuration for /dev/video2 saved here: /etc/linux-enable-ir-emitter/pci-0000:65:00.4-usb-0:1:1.2-video-index0.ini.
[14:57:28] [debug] Instruction applied: unit: 4, selector: 2, control: 255 255.
[14:57:30] [debug] The instruction does not enable the emitter.
[14:57:30] [debug] Reseting to the instruction: unit: 4, selector: 2, control: 0 32.
[14:57:30] [debug] Instruction applied: unit: 4, selector: 3, control: 255.
[14:57:30] [debug] The instruction does not enable the emitter.
[14:57:30] [debug] Reseting to the instruction: unit: 4, selector: 3, control: 0.
[14:57:30] [debug] Instruction applied: unit: 4, selector: 4, control: 255 255.
[14:57:30] [debug] The instruction does not enable the emitter.
[14:57:30] [debug] Reseting to the instruction: unit: 4, selector: 4, control: 0 0.
[14:57:30] [debug] Instruction applied: unit: 4, selector: 5, control: 255 255.
[14:57:30] [debug] The instruction does not enable the emitter.
[14:57:30] [debug] Reseting to the instruction: unit: 4, selector: 5, control: 255 0.
[14:57:30] [debug] Instruction applied: unit: 4, selector: 6, control: 1.
[14:57:30] [debug] The instruction does not enable the emitter.
[14:57:30] [debug] Reseting to the instruction: unit: 4, selector: 6, control: 1.
[14:57:30] [debug] Instruction applied: unit: 4, selector: 7, control: 13.
[14:57:30] [debug] The instruction does not enable the emitter.
[14:57:30] [debug] Reseting to the instruction: unit: 4, selector: 7, control: 1.
[14:57:30] [debug] Instruction applied: unit: 4, selector: 8, control: 255.
[14:57:30] [debug] The instruction does not enable the emitter.
[14:57:30] [debug] Reseting to the instruction: unit: 4, selector: 8, control: 3.
[14:57:30] [debug] Instruction applied: unit: 4, selector: 9, control: 1.
[14:57:30] [debug] The instruction does not enable the emitter.
[14:57:30] [debug] Reseting to the instruction: unit: 4, selector: 9, control: 0.
[14:57:30] [debug] Instruction applied: unit: 4, selector: 12, control: 1.
[14:57:30] [debug] The instruction does not enable the emitter.
[14:57:30] [debug] Reseting to the instruction: unit: 4, selector: 12, control: 0.
[14:57:30] [debug] Instruction applied: unit: 4, selector: 13, control: 255 255 255 0.
[14:57:30] [debug] The instruction does not enable the emitter.
[14:57:30] [debug] Reseting to the instruction: unit: 4, selector: 13, control: 0 0 0 0.
[14:57:30] [debug] Instruction applied: unit: 4, selector: 14, control: 255 255 255 0.
[14:57:30] [debug] The instruction does not enable the emitter.
[14:57:30] [debug] Reseting to the instruction: unit: 4, selector: 14, control: 0 0 0 0.
[14:57:30] [debug] Reseting to the instruction: unit: 4, selector: 15, control: 149 40 48 201 0 194 2 0 255 255 65 255 0 0 0 0 0 0 16 0 0 0 0 1 234 233 63 80 123 217 224 110 10 194 39 92 33 153 133 125 36 167 16 220 158 226 59 6 250 32 23 245 210 24 49 136 0 0 0 1 0 3 240 0 0 0 0 1 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255.
[14:57:30] [debug] Instruction applied: unit: 4, selector: 16, control: 3.
[14:57:30] [debug] The instruction does not enable the emitter.
[14:57:30] [debug] Reseting to the instruction: unit: 4, selector: 16, control: 0.
[14:57:30] [debug] Instruction applied: unit: 4, selector: 17, control: 255 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0.
[14:57:30] [debug] The instruction does not enable the emitter.
[14:57:30] [debug] Reseting to the instruction: unit: 4, selector: 17, control: 0 8 0 143 170 41 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0.
[14:57:30] [debug] Instruction applied: unit: 4, selector: 21, control: 1.
[14:57:30] [debug] The instruction does not enable the emitter.
[14:57:30] [debug] Reseting to the instruction: unit: 4, selector: 21, control: 0.
[... cut ...]
**Give more information if you have**

The installation was done using the tarball provided on GitHub.

The IR emitter never lit up even once, and the configuration crashed with a segmentation fault.

A shutdown + boot is required to be able to access the camera again; after that if I run the configure command again, the outcome is the same.

**Additional info**
 - Computer (or camera) model: Lenovo Yoga 7 2-in-1 14AHP9
 - Linux distribution: Fedora 41 Kinoite
 - Version of linux-enable-ir-emitter: 6.0.6

EmixamPP, 2024-12-16:
Could you try this new version: https://github.com/EmixamPP/linux-enable-ir-emitter/pull/211
There is a bot comment explaining how to try it (uninstall your current version if installed via your package manager)

KirrimK, 2024-12-16:
Hello,

I have tried the version you mentioned.

I tried to configure once, and it failed (was told to shutdown and boot again).

I did so, then tried to configure again in manual mode. I managed to see my IR emitter turn on at various intensity levels, but it never blinked, and so the configuration failed.

I did not encounter a segmentation fault with this new version.

EmixamPP, 2024-12-16:
Try to execute `sudo linux-enable-ir-emitter tweak` and input `24` then `1 3 2 0 0 0 0 0 0`

Do you see your emitter blinking?

KirrimK, 2024-12-16:
I did as you suggested, but my IR emitter doesn't blink and doesn't turn on.

Here is the output of my `tweak` command if that helps:

```
[INFO] Tweaking the camera /dev/video2
[INFO] Caution, you could break the camera.
0) unit: 4, selector: 2, control: 0 32 [idle]
1) unit: 4, selector: 3, control: 0 [disable]
2) unit: 4, selector: 4, control: 0 0 [idle]
3) unit: 4, selector: 5, control: 0 0 [idle]
4) unit: 4, selector: 6, control: 1 [idle]
5) unit: 4, selector: 7, control: 1 [idle]
6) unit: 4, selector: 8, control: 3 [idle]
7) unit: 4, selector: 9, control: 0 [idle]
8) unit: 4, selector: 12, control: 0 [idle]
9) unit: 4, selector: 13, control: 0 0 0 0 [idle]
10) unit: 4, selector: 14, control: 0 0 0 0 [idle]
11) unit: 4, selector: 15, control: 149 40 48 201 0 194 2 0 255 255 65 255 0 0 0 0 0 0 16 0 0 0 0 1 234 233 63 80 123 217 224 110 10 194 39 92 33 153 133 125 36 167 16 220 158 226 59 6 250 32 23 245 210 24 49 136 0 0 0 1 0 3 240 0 0 0 0 1 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 255 [idle]
12) unit: 4, selector: 16, control: 0 [idle]
13) unit: 4, selector: 17, control: 0 8 0 143 170 41 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 [idle]
14) unit: 4, selector: 21, control: 0 [idle]
15) unit: 4, selector: 22, control: 149 40 48 201 0 194 2 0 255 255 65 255 0 0 0 0 0 0 16 0 0 0 0 1 234 233 63 80 123 217 224 110 10 194 39 92 33 153 133 125 36 167 16 220 158 226 59 6 250 32 23 245 210 24 49 136 0 0 0 1 0 3 240 0 [idle]
16) unit: 4, selector: 23, control: 0 0 0 [idle]
17) unit: 4, selector: 25, control: 0 [idle]
18) unit: 4, selector: 26, control: 0 0 0 0 [idle]
19) unit: 4, selector: 27, control: 0 0 0 0 [idle]
20) unit: 5, selector: 2, control: 0 [idle]
21) unit: 5, selector: 4, control: 114 0 [idle]
22) unit: 7, selector: 2, control: 1 0 0 0 0 0 0 96 97 2 0 0 0 0 0 [idle]
23) unit: 7, selector: 3, control: 16 0 0 0 0 0 0 0 0 0 0 [idle]
24) unit: 7, selector: 6, control: 1 3 2 0 0 0 0 0 0 [idle]
25) unit: 7, selector: 9, control: 0 0 0 0 [idle]
26) exit
```

EmixamPP, 2024-12-17:
Try with the `ffplay` utils: `ffplay /dev/video<X>` (`ffmpeg` package on fedora)

EmixamPP, 2024-12-17:
But you can see the camera feedback?

KirrimK, 2024-12-17:
Yes, I can see the camera feedback.

EmixamPP, 2024-12-18:
Oops I responded the wrong discussion. 

Try to execute `sudo linux-enable-ir-emitter tweak` and input `24` then `start`.

After that, execute  `linux-enable-ir-emitter run` followed by  `linux-enable-ir-emitter test`, do you see your emitter blinking?

KirrimK, 2024-12-19:
I just tried as you suggested, and I managed to get the emitter blinking.

However, I have to run `linux-enable-ir-emitter run` every time I want to use my camera to see the emitter blinking, am I missing something?

KirrimK, 2024-12-19:
Nevermind, I just saw that my problem is related to other ongoing issues, and since my initial issue has been solved, there's no use in keeping this thread open.

Thank you for all the help you provided.

closed 2024-12-19 by KirrimK