EmixamPP/linux-enable-ir-emitter #218: Dead lock detected

Frigyes06, 2025-02-09:
**Describe the bug**

When running in -m manual mode, it pops up one test window, but when I answer no, it starts logging
QMetaMethod::invoke: Dead lock detected in BlockingQueuedConnection: Reciever is GuiReciever(0x70cc704383f0)
**How to reproduce**

run the program in manual mode
answer first prompt
observe

**Additional info**
 - Computer (or camera) model:
 - ThinkPad T14 Gen 4
 - Linux distribution:
 - Arch Linux
 - Version of linux-enable-ir-emitter: 6.1.1

juanCoder64, 2025-02-14:
I had the same issue, seems to be related with running the command from Wayland (using xhost), I was able to configure the IR emitter using Xorg (i3)

Frigyes06, 2025-02-14:
Ah interesting. I will try that soon then

juanCoder64, 2025-02-14:
I was able to replicate the bug on Xorg :/

EmixamPP, 2025-02-16:
Did you try with the binary I provide here on GitHub?

juanCoder64, 2025-02-17:
The issue seems to be fixed on the binary from here

closed 2025-03-10 by EmixamPP