EmixamPP/linux-enable-ir-emitter #219: How do I use the tweak function?

NotMifa, 2025-02-15:
I have used the --verbose function with this tool and found out that the command ("Instruction applied: unit: 6, selector: 14, control: 2 100.") lights up my emitters. How do I use the tweak function to get this working? Sorry, a BIG noob at this. Also, how can I integrate this with howdy?

NotMifa, 2025-02-15:
nvm, got this working. But they are not as bright as they are supposed to be, no matter how much I change the control. During the howdy setup, it worked and it blinked. But after pressing Y, and testing it with howdy, it never worked..

EmixamPP, 2025-02-16:
First, let's see what you did; please show me the content of your emitter configuration located in `/etc/linux-enable-ir-emitter/`

Second, give me the output of : `systemctl status linux-enable-ir-emitter`

EmixamPP, 2025-03-29:
Feel free to comment to reopen.

closed 2025-03-29 by EmixamPP