EmixamPP/linux-enable-ir-emitter #220: IR emitter stops working after 1 or 2 boots

omjee-pandey, 2025-02-24:
**Output of `journalctl -b -u linux-enable-ir-emitter`**
```
-- No entries --
```

**Output of `sudo linux-enable-ir-emitter tweak`**
```
[ WARN:0@0.001] global cap_v4l.cpp:913 open VIDEOIO(V4L2:/dev/video3): can't open camera by index
[ WARN:0@0.001] global cap.cpp:478 open VIDEOIO(V4L2): backend is generally available but can't be used to capture by index
[INFO] Tweaking the camera /dev/video2
[INFO] Caution, you could break the camera.
0) unit: 13, selector: 14, control: 0 0 [idle]
1) unit: 13, selector: 16, control: 0 [idle]
2) unit: 14, selector: 2, control: 1 0 0 0 0 0 0 16 39 0 0 0 0 0 0 [idle]
3) unit: 14, selector: 3, control: 16 0 0 0 0 0 0 0 0 0 0 [idle]
4) unit: 14, selector: 6, control: 1 3 1 0 0 0 0 0 0 [idle]
5) unit: 14, selector: 9, control: 1 0 0 0 [idle]
6) exit
Choose an instruction to tweak: QSettings::value: Empty key passed
```

**Give more information if you have**
When I try to configure it, it keeps blinking so I close the terminal and it start working and it works for some time but after 1 or 2 reboots it stops working

**Additional info**
 - Computer (or camera) model: Syntek Integrated Camera
 - Linux distribution: Arch Linux
 - Version of linux-enable-ir-emitter:

EmixamPP, 2025-03-10:
1. sudo linux-enable-ir-emitter tweak
2. enter: 4
3. enter:  1 3 2 0 0 0 0 0 0
4. See if the emitter is blinking?
5. enter: 4
6. enter: start
7. enter 6

Now it should be configured

Restart  and show me: systemctl status linux-enable-ir-emitter

omjee-pandey, 2025-03-22:
> 1. sudo linux-enable-ir-emitter tweak
> 2. enter: 4
> 3. enter:  1 3 2 0 0 0 0 0 0
> 4. See if the emitter is blinking?
> 5. enter: 4
> 6. enter: start
> 7. enter 6
> 
> Now it should be configured
> 
> Restart  and show me: systemctl status linux-enable-ir-emitter

It works now
Thanks :)

closed 2025-03-22 by omjee-pandey