EmixamPP/linux-enable-ir-emitter #221: Please support aarch64

drowningblue, 2025-03-07:
I would like to be able to run this on a Raspberry pi. Thank you.

EmixamPP, 2025-03-10:
My code does not use any x86 specific feature.
You have to compile it yourself on your Raspberry pi: https://github.com/EmixamPP/linux-enable-ir-emitter/blob/master/docs/manual-build.md

drowningblue, 2025-03-13:
Thank you for the response. I have been trying to get this to compile for days. I am kinda in over my head here. I can't complie on Raspbian because it doesn't have gcc 13 package. I tried compiling gcc from source but it keeps failing from ram constraints. 

I was looking over some of your previous comments on some of the other projects and this may not even work for my usecase. I just need something to set the IR torch from every other frame to on, to use it for night vision. So basically set the KS_EXTENDEDPROP_TORCHMOFE_ALWAYS_ON in the usb driver somehow. Is this how your program works?

EmixamPP, 2025-03-23:
Sorry for the long response delay!

> Is this how your program works?

No, not at all, I rely on the linux interface for the UVC standard.

I think we should go for the aarch64 cross compilation on x86 approach. I don't currently have the will to do it myself, but I'd be happy to help you if you'd like.

And if you feel like it you can contribute to this project by creating a PR to extend the current build and release pipelines to support this cross compilation and share this aarch64 tarball. https://github.com/EmixamPP/linux-enable-ir-emitter/blob/master/.github/workflows/build.yml and https://github.com/EmixamPP/linux-enable-ir-emitter/blob/master/.github/workflows/release.yml

But of course there's no obligation. Even if you manage to do it and just share with me what package you installed and what steps you had to take, it could help me if one day I want to do it myself.

drowningblue, 2025-03-24:
I figured as much. Thank you. I am new here and would love to contribute in some way, but I feel it is too advanced for me at my current level of knowledge. 

I looked at cross compiling but I don't have the time right now to go down the rabbit hole. I will attempt at a later date. If I am successful I will definitely share my fruits and findings.

closed 2025-03-24 by drowningblue

EmixamPP, 2026-11-04:
I just added support in the last merge on master; I should release tomorrow the v7 with the binary for it.