EmixamPP/linux-enable-ir-emitter #222: configure finds IR LED and then hangs

jwagnerhki, 2025-03-10:
**Describe the bug**

Built from git master. The "linux-enable-ir-emitter configure" step hangs indefinitely with a blinking IR LED after responding with Yes.

**How to reproduce**

"linux-enable-ir-emitter configure", first respond No (as  IR not blinking), second time respond Yes.

**Additional info**
 - Computer (or camera) model: ThinkPad X13 Yoga Gen 1
 - Linux distribution: Ubuntu 24.04.02
 - Version of linux-enable-ir-emitter: 6.1.1 (git master)

root@x13yoga~# linux-enable-ir-emitter -V
linux-enable-ir-emitter 6.1.1
Developed by Maxime Dirksen - EmixamPP
MIT License

root@x13yoga:~# gdb -p 67817
(gdb) bt
#0  0x000079d1b9a98d71 in __futex_abstimed_wait_common64
    (private=128, cancel=true, abstime=0x0, op=265, expected=67860, futex_word=0x79d1659f6990)
    at ./nptl/futex-internal.c:57
#1  __futex_abstimed_wait_common
    (cancel=true, private=128, abstime=0x0, clockid=0, expected=67860, futex_word=0x79d1659f6990)
    at ./nptl/futex-internal.c:87
#2  __GI___futex_abstimed_wait_cancelable64
    (futex_word=futex_word@entry=0x79d1659f6990, expected=67860, clockid=clockid@entry=0, abstime=abstime@entry=0x0, private=private@entry=128) at ./nptl/futex-internal.c:139
#3  0x000079d1b9a9e7a3 in __pthread_clockjoin_ex
    (threadid=133940260071104, thread_return=0x0, clockid=0, abstime=0x0, block=<optimized out>)
    at ./nptl/pthread_join_common.c:102
#4  0x000079d1b9eece33 in std::thread::join() () at /lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00005d0d71b5e949 in Camera::is_emitter_working() ()
#6  0x00005d0d71b8344a in Tools::Find(Configuration&, unsigned int, unsigned int) ()
#7  0x00005d0d71b7abf4 in configure(std::optional<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > const&, int, int, bool, unsigned int, unsigned int, bool) ()
#8  0x00005d0d71b3c44f in main ()

(gdb) info threads
  Id   Target Id                                           Frame 
* 1    Thread 0x79d1a3a43d80 (LWP 67817) "linux-enable-ir" 0x000079d1b9a98d71 in __futex_abstimed_wait_common64 (private=128, cancel=true, abstime=0x0, op=265, expected=67860, 
    futex_word=0x79d1659f6990) at ./nptl/futex-internal.c:57
  2    Thread 0x79d1659f66c0 (LWP 67860) "linux-enable-ir" syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
  3    Thread 0x79d180dff6c0 (LWP 67838) "linux-enab:sh0"  0x000079d1b9a98d71 in __futex_abstimed_wait_common64 (private=0, cancel=true, abstime=0x0, op=393, expected=0, 
    futex_word=0x79d134421d58) at ./nptl/futex-internal.c:57
  4    Thread 0x79d1783ff6c0 (LWP 67837) "linux-e:disk$0"  0x000079d1b9a98d71 in __futex_abstimed_wait_common64 (private=0, cancel=true, abstime=0x0, op=393, expected=0, 
    futex_word=0x79d1342781b8) at ./nptl/futex-internal.c:57
  5    Thread 0x79d15eefe6c0 (LWP 67835) "QDBusConnection" 0x000079d1b9b1b4cd in __GI___poll (fds=0x79d128001350, nfds=1, timeout=-1) at ../sysdeps/unix/sysv/linux/poll.c:29
  6    Thread 0x79d1661f86c0 (LWP 67831) "linux-enable-ir" syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
  7    Thread 0x79d165df76c0 (LWP 67830) "linux-enable-ir" syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
  8    Thread 0x79d1669fa6c0 (LWP 67829) "linux-enable-ir" syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
  9    Thread 0x79d1665f96c0 (LWP 67828) "linux-enable-ir" syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
  10   Thread 0x79d166dfb6c0 (LWP 67827) "linux-enable-ir" syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
  11   Thread 0x79d1671fc6c0 (LWP 67826) "linux-enable-ir" syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
  12   Thread 0x79d1675fd6c0 (LWP 67825) "linux-enable-ir" syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38

root@x13yoga:~# systemctl status linux-enable-ir-emitter
○ linux-enable-ir-emitter.service - enable the infrared emitter
     Loaded: loaded (/etc/systemd/system/linux-enable-ir-emitter.service; enabled; preset: enabled)
     Active: inactive (dead) since Mon 2025-03-10 09:04:08 CET; 43min ago
    Process: 66935 ExecStartPre=modprobe uvcvideo (code=exited, status=0/SUCCESS)
    Process: 66937 ExecStartPre=sleep 1 (code=exited, status=0/SUCCESS)
    Process: 67396 ExecStart=linux-enable-ir-emitter --verbose run (code=exited, status=0/SUCCESS)
   Main PID: 67396 (code=exited, status=0/SUCCESS)
        CPU: 558ms

Mar 10 09:04:07 x13yoga systemd[1]: Starting linux-enable-ir-emitter.service - enable the infrared emitter...
Mar 10 09:04:08 x13yoga linux-enable-ir-emitter[67396]: [DEBUG] Executing run command.
Mar 10 09:04:08 x13yoga linux-enable-ir-emitter[67396]: [WARN] filesystem error: directory iterator cannot open directory: No such file or directory [etc/linux-enable-ir-emitter/]
Mar 10 09:04:08 x13yoga linux-enable-ir-emitter[67396]: [WARN] No device has been configured.
Mar 10 09:04:08 x13yoga systemd[1]: linux-enable-ir-emitter.service: Deactivated successfully.
Mar 10 09:04:08 x13yoga systemd[1]: Finished linux-enable-ir-emitter.service - enable the infrared emitter.

jwagnerhki, 2025-03-10:
The non verbose output is

```
root@x13yoga~# linux-enable-ir-emitter configure
[INFO] Stand in front of and close to the camera and make sure the room is well lit.
[INFO] Ensure to not use the camera during the execution.
[INFO] Configuring the camera /dev/video2.
Is the ir emitter flashing (not just turn on)? Yes/No? QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
no
Is the ir emitter flashing (not just turn on)? Yes/No? Yes
```

During the first question the camera feed preview window is still showing. After entering "no", the feed closes. And in the next stage while the program looks for the next emitter, the preview windows does not re-open, but the IR LED is flashing. I guess after "yes" the program hangs while waiting for the not opened preview windows to "close".

jwagnerhki, 2025-03-10:
With an explicitly specified device and immediately replying Yes there is no hang. But curiously configure then skips the configuration.

```
root@x13yoga:/# systemctl stop linux-enable-ir-emitter
root@x13yoga:/# linux-enable-ir-emitter -v -d /dev/video2 configure
[DEBUG] Executing configure command.
[INFO] Stand in front of and close to the camera and make sure the room is well lit.
[INFO] Ensure to not use the camera during the execution.
[INFO] Configuring the camera /dev/video2.
Is the ir emitter flashing (not just turn on)? Yes/No? QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
Yes
[ERROR] The emitter is already working, skipping the configuration.
QObject::killTimer: Timers cannot be stopped from another thread
QObject::~QObject: Timers cannot be stopped from another thread

root@x13yoga:/# systemctl start linux-enable-ir-emitter
```

with an empty config dir, contrary to [documentation](https://github.com/EmixamPP/linux-enable-ir-emitter/blob/master/docs/configurations.md),

```
root@x13yoga:/# ll /etc/linux-enable-ir-emitter/
total 16
drwxr-xr-x   2 root root  4096 Mar 10 09:20 ./
drwxr-xr-x 149 root root 12288 Mar 10 09:20 ../
root@x13yoga:/# 
```

Nevertheless, a config somewhere must've changed, as Howdy accessing /dev/video2 now turns on the IR light.

The bug in the original post persists (and'd be nice to get a fix for it). Meanwhile an  explicit "-d /dev/video2 configure"  might work for others too as a workaround.

EmixamPP, 2025-03-10:
Looks like your config folder is wrongly installed
> Mar 10 09:04:08 x13yoga linux-enable-ir-emitter[67396]: [WARN] filesystem error: directory iterator cannot open directory: No such file or directory [etc/linux-enable-ir-emitter/]

I guess you followed step by step: https://github.com/EmixamPP/linux-enable-ir-emitter/blob/master/docs/manual-build.md ?

EmixamPP, 2025-05-29:
Feel free to comment if you need further help

closed 2025-05-29 by EmixamPP