EmixamPP/linux-enable-ir-emitter #224: Doesnt work after suspend

dark-wizz, 2025-03-24:
**Describe the bug**
Does not work after suspending the system. Even works on reboot, but suspend casing the IR no to emit. After resuming I had to do `sudo linux-enable-ir-emitter configure`, running just `sudo linux-enable-ir-emitter run` after the suspension also not working.

**How to reproduce**
1. configure `sudo linux-enable-ir-emitter configure`
2. access IR camera, IR would emit
3. suspend and try access again, it wont work

**Additional info**
 - Computer (or camera) model: lenovo T490
 - Linux distribution: Arch(gnome)
 - Version of linux-enable-ir-emitter: 6.1.1

dark-wizz, 2025-03-24:
`journel -f` gives the below output while suspending

Mar 25 03:51:16 arch linux-enable-ir-emitter[6399]: [DEBUG] Executing run command.
Mar 25 03:51:16 arch linux-enable-ir-emitter[6399]: [ WARN:0@0.001] global cap_v4l.cpp:913 open VIDEOIO(V4L2:/dev/video2): can't open camera by index
Mar 25 03:51:16 arch linux-enable-ir-emitter[6399]: [ WARN:0@0.001] global cap.cpp:478 open VIDEOIO(V4L2): backend is generally available but can't be used to capture by index
Mar 25 03:51:16 arch linux-enable-ir-emitter[6399]: [DEBUG] Configuration for /dev/video2 loaded from /etc/linux-enable-ir-emitter/pci-0000:00:14.0-usb-0:8:1.2-video-index0.

EmixamPP, 2025-03-29:
Please try with the binary provided in the release section here.

dark-wizz, 2025-03-29:
Thanks it worked.

closed 2025-03-29 by dark-wizz