EmixamPP/linux-enable-ir-emitter #226: COPR Questions

Cwavs, 2025-04-05:
Hi there!

I haven't used a template because this didn't seem to really fit any of them. If you'd like me to I can go back and edit it into one though!

I've been setting a new laptop up and found this project as part of the process, I notice there's a copr repo, however it's deprecated and hasn't received an update in a few years. I've had a look round this github repo in the issues and such, but I couldn't find any obvious announcement of it's deprecation and the reasoning behind it.

I generally prefer to install things through my package manager, but I'm not opposed to installing this manually. I was mainly wondering, if you don't mind me asking, if you could share the reasoning for discontinuing the repo? I totally get it if it was getting too much to maintain. I'm mostly just curious!

Additionally what's the update flow like? Does it auto-update in anyway or notify me so I can come here and install it manually?

EmixamPP, 2025-04-05:
Hi!
The main reason is there is no point in updating this tool if that version works for you. And I also like to do breaking change when I want to.
I also understand that it is not the best to install tar balls, but I provide the uninstallation command.

Eventually, when flatpak will have proper systemd support, I could consider an officially supported package.

Cwavs, 2025-04-06:
Thanks for replying! That makes a lot of sense to be fair. I did see the uninstallation commands, and that is helpful. I wonder if it might be worth distributing packages as rpm files or something through GitHub actions? Then they don't update but installation/installation is a little cleaner for the user. Either way the current tool works really well for me so thanks for creating it! I'll close the issue because that answered my question.

closed 2025-04-06 by Cwavs