EmixamPP/linux-enable-ir-emitter #229: Configuration fails on Dell Precision 5540

mynameisdaniel32, 2025-06-09:
**Before opening the issue**
- [Y ] I looked at the [docs](https://github.com/EmixamPP/linux-enable-ir-emitter/blob/master/docs/README.md)
- [ Y] I installed the [release tarball provided here on Github](https://github.com/EmixamPP/linux-enable-ir-emitter?tab=readme-ov-file#installation)

**Ouput of `cat /var/local/log/linux-enable-ir-emitter/linux-enable-ir-emitter.log`**
```
[19:17:48] [INFO] Stand in front of and close to the camera and make sure the room is well lit.
[19:17:48] [INFO] Ensure to not use the camera during the execution.
[19:17:51] [INFO] Configuring the camera /dev/video2.
[19:18:12] [INFO] Please retry in manual mode by adding the '-m' option.
[19:18:12] [ERROR] The configuration failed.
[19:18:12] [INFO] Do not hesitate to visit the GitHub!
[19:18:12] [INFO] https://github.com/EmixamPP/linux-enable-ir-emitter/blob/master/docs/README.md
[19:21:44] [INFO] Stand in front of and close to the camera and make sure the room is well lit.
[19:21:44] [INFO] Ensure to not use the camera during the execution.
[19:21:46] [INFO] Configuring the camera /dev/video2.
[19:22:00] [INFO] Please retry in manual mode by adding the '-m' option.
[19:22:00] [ERROR] The configuration failed.
[19:22:00] [INFO] Do not hesitate to visit the GitHub!
[19:22:00] [INFO] https://github.com/EmixamPP/linux-enable-ir-emitter/blob/master/docs/README.md
[19:22:10] [INFO] Stand in front of and close to the camera and make sure the room is well lit.
[19:22:10] [INFO] Ensure to not use the camera during the execution.
[19:22:10] [INFO] Configuring the camera /dev/video2.
[19:22:21] [INFO] Please retry in manual mode by adding the '-m' option.
[19:22:21] [ERROR] The configuration failed.
[19:22:21] [INFO] Do not hesitate to visit the GitHub!
[19:22:21] [INFO] https://github.com/EmixamPP/linux-enable-ir-emitter/blob/master/docs/README.md
[19:22:57] [INFO] Tweaking the camera /dev/video2
[19:22:57] [INFO] Caution, you could break the camera.
[19:27:07] [INFO] Tweaking the camera /dev/video2
[19:27:07] [INFO] Caution, you could break the camera.
[19:27:25] [INFO] Tweaking the camera /dev/video2
[19:27:25] [INFO] Caution, you could break the camera.
[19:27:37] [INFO] Tweaking the camera /dev/video2
[19:27:37] [INFO] Caution, you could break the camera.
[19:29:31] [INFO] Stand in front of and close to the camera and make sure the room is well lit.
[19:29:31] [INFO] Ensure to not use the camera during the execution.
[19:29:33] [INFO] Configuring the camera /dev/video2.
[19:29:43] [INFO] Please retry in manual mode by adding the '-m' option.
[19:29:43] [ERROR] The configuration failed.
[19:29:43] [INFO] Do not hesitate to visit the GitHub!
[19:29:43] [INFO] https://github.com/EmixamPP/linux-enable-ir-emitter/blob/master/docs/README.md
[19:29:51] [INFO] Stand in front of and close to the camera and make sure the room is well lit.
[19:29:51] [INFO] Ensure to not use the camera during the execution.
[19:29:52] [INFO] Configuring the camera /dev/video2.
[19:30:02] [INFO] Please retry in manual mode by adding the '-m' option.
[19:30:02] [ERROR] The configuration failed.
[19:30:02] [INFO] Do not hesitate to visit the GitHub!
[19:30:02] [INFO] https://github.com/EmixamPP/linux-enable-ir-emitter/blob/master/docs/README.md
[19:47:46] [INFO] Stand in front of and close to the camera and make sure the room is well lit.
[19:47:46] [INFO] Ensure to not use the camera during the execution.
[19:47:49] [INFO] Configuring the camera /dev/video2.
[19:47:56] [ERROR] Impossible to reset the instruction: unit: 10, selector: 10, control: 0 8 0 187 187 187 187 187.
[19:47:56] [INFO] Please shutdown your computer, then boot and retry.
[19:47:56] [INFO] Please retry in manual mode by adding the '-m' option.
[19:47:56] [ERROR] The configuration failed.
[19:47:56] [INFO] Do not hesitate to visit the GitHub!
[19:47:56] [INFO] https://github.com/EmixamPP/linux-enable-ir-emitter/blob/master/docs/README.md
[19:52:42] [DEBUG] Executing configure command.
[19:52:42] [INFO] Stand in front of and close to the camera and make sure the room is well lit.
[19:52:42] [INFO] Ensure to not use the camera during the execution.
[19:52:42] [DEBUG] Checking if /dev/video0 is a grayscale camera.
[19:52:43] [DEBUG] Checking if /dev/video1 is a grayscale camera.
[19:52:43] [DEBUG] Checking if /dev/video2 is a grayscale camera.
[19:52:44] [DEBUG] /dev/video2 is a grayscale camera.
[19:52:44] [INFO] Configuring the camera /dev/video2.
[19:52:48] [DEBUG] Configuration for /dev/video2 loaded from /etc/linux-enable-ir-emitter/pci-0000:00:14.0-usb-0:12:1.2-video-index0.
[19:52:48] [DEBUG] Disable instruction skipped: unit: 10, selector: 10, control: 0 8 0 187 187 187 187 187.
[19:52:48] [INFO] Please retry in manual mode by adding the '-m' option.
[19:52:48] [DEBUG] Configuration for /dev/video2 saved here: /etc/linux-enable-ir-emitter/pci-0000:00:14.0-usb-0:12:1.2-video-index0.
[19:52:48] [ERROR] The configuration failed.
[19:52:48] [INFO] Do not hesitate to visit the GitHub!
[19:52:48] [INFO] https://github.com/EmixamPP/linux-enable-ir-emitter/blob/master/docs/README.md
```

**Output of `sudo linux-enable-ir-emitter tweak`**
```
Seems to be no effect changing all to 255.
```

**Output of `v4l2-ctl --list-devices`**
```
Integrated_Webcam_HD: Integrate (usb-0000:00:14.0-12):
	/dev/video0
	/dev/video1
	/dev/video2
	/dev/video3
	/dev/media0
	/dev/media1
```

**Ouput of `for dev in /dev/video*; do echo $dev && v4l2-ctl -d $dev --list-formats-ext; done`**
```
/dev/video0
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'MJPG' (Motion-JPEG, compressed)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 848x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 960x540
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.033s (30.000 fps)
	[1]: 'YUYV' (YUYV 4:2:2)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 424x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x180
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 160x120
			Interval: Discrete 0.033s (30.000 fps)
/dev/video1
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

/dev/video2
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'GREY' (8-bit Greyscale)
		Size: Discrete 340x340
			Interval: Discrete 0.017s (60.000 fps)
/dev/video3
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture
```

**Output of `ls -l /dev/v4l/by-path`**
```
total 0
lrwxrwxrwx 1 root root 12 Jun  9 08:10 pci-0000:00:14.0-usb-0:12:1.0-video-index0 -> ../../video0
lrwxrwxrwx 1 root root 12 Jun  9 08:10 pci-0000:00:14.0-usb-0:12:1.0-video-index1 -> ../../video1
lrwxrwxrwx 1 root root 12 Jun  9 08:10 pci-0000:00:14.0-usb-0:12:1.2-video-index0 -> ../../video2
lrwxrwxrwx 1 root root 12 Jun  9 08:10 pci-0000:00:14.0-usb-0:12:1.2-video-index1 -> ../../video3
lrwxrwxrwx 1 root root 12 Jun  9 08:10 pci-0000:00:14.0-usbv2-0:12:1.0-video-index0 -> ../../video0
lrwxrwxrwx 1 root root 12 Jun  9 08:10 pci-0000:00:14.0-usbv2-0:12:1.0-video-index1 -> ../../video1
lrwxrwxrwx 1 root root 12 Jun  9 08:10 pci-0000:00:14.0-usbv2-0:12:1.2-video-index0 -> ../../video2
lrwxrwxrwx 1 root root 12 Jun  9 08:10 pci-0000:00:14.0-usbv2-0:12:1.2-video-index1 -> ../../video3
```

**Give more information if you have**
When the camera previews the 340x340 frame, if I shine a torch at the camera it sees this, so I am fairly confident video2 is working and the one we want, I just can't get the red IR LEDs to turn on.

Currently have Howdy working with the RGB camera only.

When I try setting limit to -1 (or 256) I get this error:
'[ERROR] Impossible to reset the instruction: unit: 10, selector: 10, control: 0 8 0 187 187 187 187 187.'

**Additional info**
 - Computer (or camera) model: Dell Precision 5540
 - Linux distribution: Linux Mint 22.1
 - Version of linux-enable-ir-emitter:  linux-enable-ir-emitter 6.1.1

EmixamPP, 2025-07-16:
Please show me the outputs of the `sudo linux-enable-ir-emitter tweak` menu

EmixamPP, 2025-09-07:
Feel free to comment to re-open