EmixamPP/linux-enable-ir-emitter #232: Configuration has failed | IR light doesn't flash

Pyenb, 2025-07-22:
**Before opening the issue**
- [x] I looked at the [docs](https://github.com/EmixamPP/linux-enable-ir-emitter/blob/master/docs/README.md)
- [x] I installed the [release tarball provided here on Github](https://github.com/EmixamPP/linux-enable-ir-emitter?tab=readme-ov-file#installation)

**Ouput of `cat /var/local/log/linux-enable-ir-emitter/linux-enable-ir-emitter.log`**
```
[17:56:05] [INFO] Stand in front of and close to the camera and make sure the room is well lit.
[17:56:05] [INFO] Ensure to not use the camera during the execution.
[17:56:06] [INFO] Configuring the camera /dev/video2.
[17:57:44] [INFO] Please retry in manual mode by adding the '-m' option.
[17:57:44] [ERROR] The configuration failed.
[17:57:44] [INFO] Do not hesitate to visit the GitHub!
[17:57:44] [INFO] https://github.com/EmixamPP/linux-enable-ir-emitter/blob/master/docs/README.md
```

**Output of `sudo linux-enable-ir-emitter tweak`**
```
[ WARN:0@0.000] global cap_v4l.cpp:913 open VIDEOIO(V4L2:/dev/video3): can't open camera by index
[ WARN:0@0.000] global cap.cpp:478 open VIDEOIO(V4L2): backend is generally available but can't be used to capture by index
[INFO] Tweaking the camera /dev/video2
[INFO] Caution, you could break the camera.
0) unit: 4, selector: 2, control: 0 32 [disable]
1) unit: 4, selector: 3, control: 0 [disable]
2) unit: 4, selector: 4, control: 0 0 [disable]
3) unit: 4, selector: 5, control: 0 0 [disable]
4) unit: 4, selector: 6, control: 1 [disable]
5) unit: 4, selector: 7, control: 1 [disable]
6) unit: 4, selector: 8, control: 3 [disable]
7) unit: 4, selector: 9, control: 0 [disable]
8) unit: 5, selector: 2, control: 0 [idle]
9) unit: 5, selector: 4, control: 79 0 [idle]
10) unit: 7, selector: 2, control: 1 0 0 0 0 0 0 96 97 2 0 0 0 0 0 [idle]
11) unit: 7, selector: 3, control: 16 0 0 0 0 0 0 0 0 0 0 [idle]
12) unit: 7, selector: 6, control: 1 3 1 0 0 0 0 0 0 [idle]
13) unit: 7, selector: 9, control: 0 0 0 0 [idle]
14) exit

```

**Output of `v4l2-ctl --list-devices`**
```
Integrated RGB Camera: Integrat (usb-0000:00:14.0-6):
        /dev/video0
        /dev/video1
        /dev/video2
        /dev/video3
        /dev/media0
        /dev/media1
```

**Ouput of `for dev in /dev/video*; do echo $dev && v4l2-ctl -d $dev --list-formats-ext; done`**
```
/dev/video0
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'MJPG' (Motion-JPEG, compressed)
                Size: Discrete 1280x720
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 1920x1080
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 960x540
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 640x480
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 640x360
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 320x240
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 320x180
                        Interval: Discrete 0.033s (30.000 fps)
        [1]: 'YUYV' (YUYV 4:2:2)
                Size: Discrete 1280x720
                        Interval: Discrete 0.100s (10.000 fps)
                Size: Discrete 1920x1080
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 960x540
                        Interval: Discrete 0.067s (15.000 fps)
                Size: Discrete 640x480
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 640x360
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 320x240
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 320x180
                        Interval: Discrete 0.033s (30.000 fps)
/dev/video1
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

/dev/video2
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'GREY' (8-bit Greyscale)
                Size: Discrete 640x360
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.033s (30.000 fps)
/dev/video3
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture
```

**Output of `ls -l /dev/v4l/by-path`**
```
total 0
lrwxrwxrwx. 1 root root 12 22. Jul 17:36 pci-0000:00:14.0-usb-0:6:1.0-video-index0 -> ../../video0
lrwxrwxrwx. 1 root root 12 22. Jul 17:36 pci-0000:00:14.0-usb-0:6:1.0-video-index1 -> ../../video1
lrwxrwxrwx. 1 root root 12 22. Jul 17:36 pci-0000:00:14.0-usb-0:6:1.2-video-index0 -> ../../video2
lrwxrwxrwx. 1 root root 12 22. Jul 17:36 pci-0000:00:14.0-usb-0:6:1.2-video-index1 -> ../../video3
lrwxrwxrwx. 1 root root 12 22. Jul 17:36 pci-0000:00:14.0-usbv2-0:6:1.0-video-index0 -> ../../video0
lrwxrwxrwx. 1 root root 12 22. Jul 17:36 pci-0000:00:14.0-usbv2-0:6:1.0-video-index1 -> ../../video1
lrwxrwxrwx. 1 root root 12 22. Jul 17:36 pci-0000:00:14.0-usbv2-0:6:1.2-video-index0 -> ../../video2
lrwxrwxrwx. 1 root root 12 22. Jul 17:36 pci-0000:00:14.0-usbv2-0:6:1.2-video-index1 -> ../../video3
```

**Give more information if you have**

**Additional info**
 - Computer model: Lenovo Thinkbook 16 G7 IML
 - Linux distribution: Fedora Linux 42
 - Version of linux-enable-ir-emitter:

EmixamPP, 2025-08-01:
Try
1. `sudo linux-enable-ir-emitter tweak`
2. Input `13`
3. Input `1 3 2 0 0 0 0 0 0`
4. You should see your emitter blinking 🤞

Pyenb, 2025-08-03:
Thanks for the answer, but I am doing something wrong? I just keep getting `[WARN] Invalid value for the control.` when trying to enter the new current control.

Log:

```
sudo linux-enable-ir-emitter tweak
[ WARN:0@0.001] global cap_v4l.cpp:913 open VIDEOIO(V4L2:/dev/video3): can't open camera by index
[ WARN:0@0.001] global cap.cpp:478 open VIDEOIO(V4L2): backend is generally available but can't be used to capture by index
[INFO] Tweaking the camera /dev/video2
[INFO] Caution, you could break the camera.
0) unit: 4, selector: 2, control: 0 32 [disable]
1) unit: 4, selector: 3, control: 0 [disable]
2) unit: 4, selector: 4, control: 0 0 [disable]
3) unit: 4, selector: 5, control: 0 0 [disable]
4) unit: 4, selector: 6, control: 1 [disable]
5) unit: 4, selector: 7, control: 1 [disable]
6) unit: 4, selector: 8, control: 3 [disable]
7) unit: 4, selector: 9, control: 0 [disable]
8) unit: 5, selector: 2, control: 0 [idle]
9) unit: 5, selector: 4, control: 79 0 [idle]
10) unit: 7, selector: 2, control: 1 0 0 0 0 0 0 96 97 2 0 0 0 0 0 [idle]
11) unit: 7, selector: 3, control: 16 0 0 0 0 0 0 0 0 0 0 [idle]
12) unit: 7, selector: 6, control: 1 3 1 0 0 0 0 0 0 [idle]
13) unit: 7, selector: 9, control: 0 0 0 0 [idle]
14) exit
Choose an instruction to tweak: 13
minimum: 0 0 0 0
maximum: 1 0 0 0
initial: 0 0 0 0
current: 0 0 0 0
status: idle
Input a new current control or status (start/idle/disable) or q to return: 1 3 2 0 0 0 0 0 0
[WARN] Invalid value for the control.
```

EmixamPP, 2025-08-05:
Ah I see, it is because of the maximum value that is blocking. Hum, I should not hard block such case.

Please open the file in `/etc/linux-enable-ir-emitter` (not the .ini one) and set the value of `1 3 2 0 0 0 0 0 0` for the unit 7, selector 6 in there. The format inside should be pretty clear to understand (human readable) + switch the status from `idle` to `enable`.

After that, execute `linux-enable-ir-emitter run`

Pyenb, 2025-08-07:
Hey, sorry for the delay, I didn't really had time to test this. Funnily enough I just did a kernel update and reinstalled the program and during the configuration phase it just started working. Can I somehow make sure that the IR LED really runs at the max power for best results, or shouldn't it be to worried about that? Currently I feel like, while it is blinking, it's not at full power.

tweaks:
```
❯ sudo linux-enable-ir-emitter tweak       
[ WARN:0@0.000] global cap_v4l.cpp:913 open VIDEOIO(V4L2:/dev/video3): can't open camera by index
[ WARN:0@0.000] global cap.cpp:478 open VIDEOIO(V4L2): backend is generally available but can't be used to capture by index
[INFO] Tweaking the camera /dev/video2
[INFO] Caution, you could break the camera.
0) unit: 5, selector: 2, control: 8 [start]
1) unit: 5, selector: 4, control: 114 0 [idle]
2) unit: 7, selector: 2, control: 1 0 0 0 0 0 0 96 97 2 0 0 0 0 0 [idle]
3) unit: 7, selector: 3, control: 16 0 0 0 0 0 0 0 0 0 0 [idle]
4) unit: 7, selector: 6, control: 1 3 1 0 0 0 0 0 0 [idle]
5) unit: 7, selector: 9, control: 0 0 0 0 [idle]
6) exit
```

And thank you again for taking the time to do this.