EmixamPP/linux-enable-ir-emitter #235: Keeping LED always ON or OFF

diogoflduarte, 2025-08-05:
**Describe your feature request**
Hey Maxime. Is there a way to keep the LED always on, instead of strobbing? I'm interested in using the IR camera data with the LED always on.

**Why would it be useful or what problem can it solve ?**

**Describe your implementation ideas**
I've tried editing the status from 'idle' to 'disable', but other than that I'm pretty lost. would appreciate any pointers!!

- status: idle
  unit: 13
  selector: 16
  current:
    - 0
  maximum:
    - 1
  minimum:
    - 0
- status: disable
  unit: 14
  selector: 6
  current:
    - 1
    - 3
    - 2
    - 0
    - 0
    - 0
    - 0
    - 0
    - 0
  maximum:
    - 1
    - 3
    - 2
    - 0
    - 0
    - 0
    - 0
    - 0
    - 0
  minimum:
    - 0
    - 0
    - 0
    - 0
    - 0
    - 0
    - 0
    - 0
    - 0
- status: disable
  unit: 14
  selector: 9
  current:
    - 1
    - 0
    - 0
    - 0
  maximum:
    - 1
    - 0
    - 0
    - 0
  minimum:
    - 0
    - 0
    - 0
    - 0

EmixamPP, 2025-08-05:
Why are you trying to do that, exactly? I suspect it won’t have the effect you're expecting.
If your emitter isn’t flashing, then it’s only emitting infrared once, meaning your camera will only capture your face clearly a single time.

But if you want to experiment, I would bet that you need to set the value of `1` for the unit 13, selector 16.
The `status` is simply driving what you want to execute when the service is executing at boot, I should better document that at some point.

`enable` = execute that instruction on boot, wake-up from sleep, etc.
`disable` = the `configure` command will ignore it (used when some instruction are breaking the camera and requires a reboot)
`idle` = nothing special will be done with that instruction

```
status: enable
unit: 13
selector: 16
current:
1
maximum:
1
minimum:
0
```

EmixamPP, 2025-09-07:
I am not sure if I will be able to help you further. But if you need more explanations / help, please feel free to comment again, I will give my best!

closed 2025-09-07 by EmixamPP