EmixamPP/linux-enable-ir-emitter #239: Configuration fails on Dell Inspiron 13 (7378)

izayazaya, 2025-08-24:
**Before opening the issue**
- [x] I looked at the [docs](https://github.com/EmixamPP/linux-enable-ir-emitter/blob/master/docs/README.md)
- [x] I installed the [release tarball provided here on Github](https://github.com/EmixamPP/linux-enable-ir-emitter?tab=readme-ov-file#installation)

**Ouput of `cat /var/local/log/linux-enable-ir-emitter/linux-enable-ir-emitter.log`**
```
cat /var/local/log/linux-enable-ir-emitter/linux-enable-ir-emitter.log
[02:15:20] [INFO] Stand in front of and close to the camera and make sure the room is well lit.
[02:15:20] [INFO] Ensure to not use the camera during the execution.
[02:15:22] [INFO] Configuring the camera /dev/video0.
[02:15:28] [INFO] Stand in front of and close to the camera and make sure the room is well lit.
[02:15:28] [INFO] Ensure to not use the camera during the execution.
[02:15:30] [INFO] Configuring the camera /dev/video0.
[02:15:56] [INFO] Please retry in manual mode by adding the '-m' option.
[02:15:56] [ERROR] The configuration failed.
[02:15:56] [INFO] Do not hesitate to visit the GitHub!
[02:15:56] [INFO] https://github.com/EmixamPP/linux-enable-ir-emitter/blob/master/docs/README.md
[02:17:11] [INFO] Stand in front of and close to the camera and make sure the room is well lit.
[02:17:11] [INFO] Ensure to not use the camera during the execution.
[02:17:13] [INFO] Configuring the camera /dev/video0.
[02:17:41] [INFO] Please retry in manual mode by adding the '-m' option.
[02:17:41] [ERROR] The configuration failed.
[02:17:41] [INFO] Do not hesitate to visit the GitHub!
[02:17:41] [INFO] https://github.com/EmixamPP/linux-enable-ir-emitter/blob/master/docs/README.md
```

**Output of `sudo linux-enable-ir-emitter tweak`**
```
[ WARN:0@0.001] global cap_v4l.cpp:913 open VIDEOIO(V4L2:/dev/video1): can't open camera by index
[ WARN:0@0.001] global cap.cpp:478 open VIDEOIO(V4L2): backend is generally available but can't be used to capture by index
[INFO] Tweaking the camera /dev/video0
[INFO] Caution, you could break the camera.
0) unit: 4, selector: 10, control: 0 8 0 187 187 187 187 187 [idle]
1) unit: 7, selector: 2, control: 1 [idle]
2) unit: 7, selector: 3, control: 0 [idle]
3) unit: 7, selector: 4, control: 0 0 [idle]
4) exit
Choose an instruction to tweak: Authorization required, but no authorization protocol specified

terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.11.0) /home/runner/work/linux-enable-ir-emitter/linux-enable-ir-emitter/.github/workflows/deps/build/opencv/src/opencv/modules/highgui/src/window_gtk.cpp:638: error: (-2:Unspecified error) Can't initialize GTK backend in function 'cvInitSystem'

Aborted                    sudo linux-enable-ir-emitter tweak
```

**Output of `v4l2-ctl --list-devices`**
```
Integrated_Webcam_HD: Integrate (usb-0000:00:14.0-5):
	/dev/video0
	/dev/video1
	/dev/media0
```

**Ouput of `for dev in /dev/video*; do echo $dev && v4l2-ctl -d $dev --list-formats-ext; done`**
```
/dev/video0
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'MJPG' (Motion-JPEG, compressed)
		Size: Discrete 1280x720
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 960x540
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 848x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
	[1]: 'YUYV' (YUYV 4:2:2)
		Size: Discrete 340x340
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 424x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x180
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 160x120
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 340x340
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
/dev/video1
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture
```

**Output of `ls -l /dev/v4l/by-path`**
```
lrwxrwxrwx - root 25 Aug 01:47  pci-0000:00:14.0-usb-0:5:1.0-video-index0 -> ../../video0
lrwxrwxrwx - root 25 Aug 01:47  pci-0000:00:14.0-usb-0:5:1.0-video-index1 -> ../../video1
lrwxrwxrwx - root 25 Aug 01:47  pci-0000:00:14.0-usbv2-0:5:1.0-video-index0 -> ../../video0
lrwxrwxrwx - root 25 Aug 01:47  pci-0000:00:14.0-usbv2-0:5:1.0-video-index1 -> ../../video1
```

**Give more information if you have**
I can't make the IR emitter work, since I started using arch about 6 months ago, I've never seen the red light appear by the camera side, but this works fine on windows (I'm dual-booting). I use "sudo -E linux-enable-ir-emitter configure", I have to add the -E flag so the pop-up appears otherwise this error appears 

```
[INFO] Stand in front of and close to the camera and make sure the room is well lit.
[INFO] Ensure to not use the camera during the execution.
[ WARN:0@0.001] global cap_v4l.cpp:913 open VIDEOIO(V4L2:/dev/video1): can't open camera by index
[ WARN:0@0.001] global cap.cpp:478 open VIDEOIO(V4L2): backend is generally available but can't be used to capture by index
[INFO] Configuring the camera /dev/video0.
Is the ir emitter flashing (not just turn on)? Yes/No? Authorization required, but no authorization protocol specified

terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.11.0) /home/runner/work/linux-enable-ir-emitter/linux-enable-ir-emitter/.github/workflows/deps/build/opencv/src/opencv/modules/highgui/src/window_gtk.cpp:638: error: (-2:Unspecified error) Can't initialize GTK backend in function 'cvInitSystem'

Aborted                    sudo linux-enable-ir-emitter configure
```

**Additional info**
 - Computer (or camera) model: Dell Inspiron 13 (7378), Bus 001 Device 002: ID 0bda:58c2 Realtek Semiconductor Corp. Integrated Webcam HD
 - Linux distribution: Arch Linux (Hyprland)
 - Version of linux-enable-ir-emitter: 6.1.1

EmixamPP, 2025-08-30:
It seems that you do not have an infrared camera. There is not GREY video format (command `for dev in /dev/video*; do echo $dev && v4l2-ctl -d $dev --list-formats-ext; done`)

izayazaya, 2025-09-03:
That's the way Arch or Linux is detecting it but I do in fact have an infrared camera, it works fine on Windows Hello (a red light to the right of the laptop camera blinks) and when I registered a face for the login, I could see the IR camera feed. I definitely have two cameras on my laptop, the normal one (white light to the left of the camera when it's active) and the infrared one (blinking red light to the right of the camera when it's active), which I've been completely unable to access in Linux.